espressif/esp-protocols #305: CMUX, OK response received on next AT-Command

inimin, 2023-06-10:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using **v1.0.0** of esp_modem over **esp-idf-v4.4.2**, I do implement **Quectel MC60** module class from generic module. The problem I faced on CMUX_MODE is that OK token is not received with the Command response. 

```
V (134404) command_lib: get_signal_quality
V (134404) command_lib: generic_get_string
V (134414) Send: 0x3ffbfee0   f9 09 ef 0f                                       |....|
V (134424) Send: 0x3ffc00e8   41 54 2b 43 53 51 0d                              |AT+CSQ.|
V (134424) Send: 0x3ffbfee4   3b f9                                             |;.|
V (134414) CMUX Received: 0x3ffbe864   f9 09 ef 03 41 32 f9 f9  09 ef 03 54 32 f9 f9 09  |....A2.....T2...|
V (134444) CMUX Received: 0x3ffbe874   ef 03 2b 32 f9 f9 09 ef  03 43 32 f9 f9 09 ef 03  |..+2.....C2.....|
V (134444) CMUX Received: 0x3ffbe884   53 32 f9 f9 09 ef 03 51  32 f9 f9 09 ef 03 0d 32  |S2.....Q2......2|
V (134464) CMUX Received: 0x3ffbe894   f9 f9 09 ef 1d 0d 0a 2b  43 53 51 3a 20 33 31 2c  |.......+CSQ: 31,|
D (134474) CMUX: Payload frame: dlci:02 type:ef payload:1 available:60
D (134474) CMUX: Payload frame: dlci:02 type:ef payload:1 available:53
D (134484) CMUX: Payload frame: dlci:02 type:ef payload:1 available:46
D (134484) CMUX: Payload frame: dlci:02 type:ef payload:1 available:39
D (134494) CMUX: Payload frame: dlci:02 type:ef payload:1 available:32
D (134504) CMUX: Payload frame: dlci:02 type:ef payload:1 available:25
D (134514) CMUX: Payload frame: dlci:02 type:ef payload:1 available:18
D (134514) CMUX: Payload frame: dlci:02 type:ef payload:14 available:11
V (134524) CMUX Received: 0x3ffbe8a4   30 0d 0a c4 f9                                    |0....|
D (134534) CMUX: Payload frame: dlci:02 type:ef payload:3 available:5
V (134534) command_lib: 0x3ffbe899   0d 0a 2b 43 53 51 3a 20  33 31 2c 30 0d 0a        |..+CSQ: 31,0..|
} (134554) command_lib: Token: {

V (134554) command_lib: Token: {+CSQ: 31,0}

E (134934) modem_console: Failed with TIMEOUT
Command returned non-zero error code: 0x1 (ERROR)

```

david-cermak, 2023-06-13:
@inimin Have you checked the TA response profile setting for your device? I can see that it supports V.25ter commands for adjusting TA response and presentation mode, such as `ATQ` and `ATV` which might cause the `OK` response to replaced by a number or skipped together (and maybe these settings are applicable per virtual terminal? as you seem to get the `OK` response on the `CMUX=xxx` command for entering the cmux mode)
You can use the modem-console example to explore these commands and modes.

inimin, 2023-06-13:
@david-cermak TA has `OK` in response message, but ESP doesn't received it on time. After I put some delay before `term->read` I got correct response message with `OK` in it.  
 
https://github.com/espressif/esp-protocols/blob/68392f0ba9d4fd9c9c7fa1f7d54d0b3de53a1990/components/esp_modem/src/esp_modem_cmux.cpp#L287

p.s I also not using hardware control flow, but I don't think this issue relate to it.

david-cermak, 2023-06-13:
would it help if you give it more time? so instead of calling:
```
get_signal_quality
```
could you please try
```
cmd AT+CSQ -t 5000
```
from the console?

inimin, 2023-06-14:
In the modem_console example, I executed the `AT+CSQ -t 5000` command, but it did not make any difference. Next, I executed the simple AT command and received the OK message twice.

First call for `AT+CSQ -t 5000` in modem_console example:

```
esp> cmd AT+CSQ -t 5000
I (40732) modem_console: Sending command AT+CSQ with timeout 5000
V (40742) Send: 0x3ffbfed0   f9 09 ef 0f                                       |....|
V (40742) Send: 0x3ffc00e0   41 54 2b 43 53 51 0d                              |AT+CSQ.|
V (40752) Send: 0x3ffbfed4   3b f9                                             |;.|
V (40742) CMUX Received: 0x3ffbe804   f9 09 ef 03 41 32 f9 f9  09 ef 03 54 32 f9 f9 09  |....A2.....T2...|
V (40762) CMUX Received: 0x3ffbe814   ef 03 2b 32 f9 f9 09 ef  03 43 32 f9 f9 09 ef 03  |..+2.....C2.....|
V (40782) CMUX Received: 0x3ffbe824   53 32 f9 f9 09 ef 03 51  32 f9 f9 09 ef 03 0d 32  |S2.....Q2......2|
V (40792) CMUX Received: 0x3ffbe834   f9 f9 09 ef 1d 0d 0a 2b  43 53 51 3a 20 32 39 2c  |.......+CSQ: 29,|
D (40792) CMUX: Payload frame: dlci:02 type:ef payload:1 available:60
D (40802) CMUX: Payload frame: dlci:02 type:ef payload:1 available:53
D (40802) CMUX: Payload frame: dlci:02 type:ef payload:1 available:46
D (40812) CMUX: Payload frame: dlci:02 type:ef payload:1 available:39
D (40822) CMUX: Payload frame: dlci:02 type:ef payload:1 available:32
D (40832) CMUX: Payload frame: dlci:02 type:ef payload:1 available:25
D (40832) CMUX: Payload frame: dlci:02 type:ef payload:1 available:18
D (40842) CMUX: Payload frame: dlci:02 type:ef payload:14 available:11
V (40842) CMUX Received: 0x3ffbe844   30 0d 0a c4 f9                                    |0....|
D (40852) CMUX: Payload frame: dlci:02 type:ef payload:3 available:5
I (40852) modem_console: 
+CSQ: 29,0

esp>

``` 

Then I ask for simple `AT` command:

```
esp> cmd at
I (82572) modem_console: Sending command at with timeout 1000
V (82582) Send: 0x3ffbfed0   f9 09 ef 07                                       |....|
V (82582) Send: 0x3ffc00e0   61 74 0d                                          |at.|
V (82582) CMUX Received: 0x3ffbe804   f9 09 ef 0d 0d 0a 4f 4b  0d 0a d8 f9 f9 09 ef 03  |......OK........|
V (82602) CMUX Received: 0x3ffbe814   61 32 f9 f9 09 ef 03 74  32 f9 f9 09 ef 03 0d 32  |a2.....t2......2|
V (82602) CMUX Received: 0x3ffbe824   f9 f9 09 ef 0d 0d 0a 4f  4b 0d 0a d8 f9           |.......OK....|
D (82612) CMUX: Payload frame: dlci:02 type:ef payload:6 available:41
I (82632) modem_console: 
OK

D (82632) CMUX: Payload frame: dlci:02 type:ef payload:1 available:29
V (82592) Send: 0x3ffbfed4   35 f9                                             |5.|
D (82632) CMUX: Payload frame: dlci:02 type:ef payload:1 available:22
D (82642) CMUX: Payload frame: dlci:02 type:ef payload:1 available:15
D (82652) CMUX: Payload frame: dlci:02 type:ef payload:6 available:8
esp> 
```

inimin, 2023-06-14:
I think the problem might be in concurrency control, I test this witht **1s** delay after `while(frame.len > 0)` at `on_cmux_data` function which is going to be called uart_termnial read operation.
https://github.com/espressif/esp-protocols/blob/68392f0ba9d4fd9c9c7fa1f7d54d0b3de53a1990/components/esp_modem/src/esp_modem_cmux.cpp#L295

 This make `CMUX::Init` to failed on `timeout++ > 100` check. which mean it doesn't wait for read operation to get completed.

david-cermak, 2023-06-14:
This is interesting! Could you please check if it fails also if you set
```
CONFIG_ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD=n
```
(disable in menuconfig)

inimin, 2023-06-17:
With `ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD` disabled in menuconfig:
```
esp> cmd AT+CSQ
I (76662) modem_console: Sending command AT+CSQ with timeout 1000
V (76662) Send: 0x3ffbfed0   f9 09 ef 0f                                       |....|
V (76662) Send: 0x3ffc00e0   41 54 2b 43 53 51 0d                              |AT+CSQ.|
V (76672) Send: 0x3ffbfed4   3b f9                                             |;.|
V (76672) CMUX Received: 0x3ffbe804   f9 09 ef 03 41 32 f9 f9  09 ef 03 54 32 f9 f9 09  |....A2.....T2...|
V (76692) CMUX Received: 0x3ffbe814   ef 03 2b 32 f9 f9 09 ef  03 43 32 f9 f9 09 ef 03  |..+2.....C2.....|
V (76702) CMUX Received: 0x3ffbe824   53 32 f9 f9 09 ef 03 51  32 f9 f9 09 ef 03 0d 32  |S2.....Q2......2|
V (76712) CMUX Received: 0x3ffbe834   f9 f9 09 ef 1d 0d 0a 2b  43 53 51 3a 20 32 39 2c  |.......+CSQ: 29,|
D (76712) CMUX: Payload frame: dlci:02 type:ef payload:1 available:60
D (76722) CMUX: Payload frame: dlci:02 type:ef payload:1 available:53
D (76732) CMUX: Payload frame: dlci:02 type:ef payload:1 available:46
D (76742) CMUX: Payload frame: dlci:02 type:ef payload:1 available:39
D (76742) CMUX: Payload frame: dlci:02 type:ef payload:1 available:32
D (76752) CMUX: Payload frame: dlci:02 type:ef payload:1 available:25
D (76752) CMUX: Payload frame: dlci:02 type:ef payload:1 available:18
D (76762) CMUX: Payload frame: dlci:02 type:ef payload:14 available:11
I (76762) modem_console: 
+CSQ: 29,
V (76782) CMUX Received: 0x3ffbe804   30 0d 0a c4 f9 f9 09 ef  0d 0d 0a 4f 4b 0d 0a d8  |0..........OK...|
V (76782) CMUX Received: 0x3ffbe814   f9                                                |.|
D (76792) CMUX: Payload frame: dlci:02 type:ef payload:3 available:17
D (76792) CMUX: Payload frame: dlci:02 type:ef payload:6 available:8
esp> 
```

inimin, 2023-06-18:
Seem like GSM module echo mode deactivated after calling ATD*99#, and my issue get resolved by calling ATE0 after network connection got established.

closed 2023-06-19 by inimin