espressif/esp-protocols #306: CMUX: Protocol mismatch: Missed trailing SOF, recovering...

diplfranzhoepfinger, 2023-06-12:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

enabled 4-Wire Mode like this:

diplfranzhoepfinger, 2023-06-12:
I will make you a "verbose" Log later.

diplfranzhoepfinger, 2023-06-12:
i added Verbose log above. 

@david-cermak

diplfranzhoepfinger, 2023-06-12:
Hi, 

have advanced the Sample a bit: 

diplfranzhoepfinger/esp-protocols-tree-master-components-esp\_modem-examples-simple\_cmux\_client@d8f88b0

as the !7672 does not do Autobaud, we can do it the "old" way, 

and i tested and working. 

log here: 

https://gist.github.com/diplfranzhoepfinger/94ac7f9e94dbac07d93e71971766379f

david-cermak, 2023-10-09:
@diplfranzhoepfinger Could we close this, too?

david-cermak, 2023-11-28:
Closing, feel free to reopen

closed 2023-11-28 by david-cermak