espressif/esp-protocols #307: CPU usage increase in time on esp_websocket_client

ozancakir, 2023-06-18:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Created web socket client as task, for every 10 sec I am sending _vTaskGetRunTimeStats_ results to my socket server with _esp_websocket_client_send_text_ and saving the log in a file in the server. After some time you can also see increase in CPU usage on both _esp_websocket_client_ and _wifi_ package:
- `websocket_task `starts at <1% and after some time it ends up ~%17
- similarly `wifi `starts at <1% and ends up ~%17
Logs are attached.

Using:
- idf release/v5.1
- esp32-s3 devkit c1

[app.log](https://github.com/espressif/esp-protocols/files/11780948/app.log)

david-cermak, 2023-06-19:
@ozancakir This is a general question, unrelated to a protocol library. Could you please close this issue and ask this question on the [forum](https://www.esp32.com/)?

ozancakir, 2023-06-19:
@david-cermak As far as I know esp_websocket_client is part of esp-protocols, and websocket_task which is the core task of esp_websocket_client protocol somehow using almost half of cpu after 12h. It looks like a bug to me.

igrr, 2023-06-19:
If you are seeing CPU usage go over 100%, you might be facing https://github.com/espressif/esp-idf/issues/9359

ozancakir, 2023-06-20:
Thanks @igrr, I will replace vTaskGetRunTimeStats with [real time stats example](https://github.com/espressif/esp-idf/tree/v5.0.1/examples/system/freertos/real_time_stats) and check for the case again. Will provide log too if issue is replicating

ozancakir, 2023-06-21:
[app2.log](https://github.com/espressif/esp-protocols/files/11815790/app2.log)

It looks like issue is related to native freertos vTaskGetRunTimeStats method, closing this one thanks for help @igrr

closed 2023-06-21 by ozancakir