espressif/esp-protocols #313: Cmake errors when using component asio and esp_mqtt_cxx

morgana2313, 2023-06-24:
### Answers checklist.

- [X] I have read the documentation for [esp-protocols components](https://espressif.github.io/esp-protocols/) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

<details><summary>1. asio: Include directory 'asio/asio/asio/include' is not a directory</summary>

When I add component `asio` to an idf-application (without using it) I get an error:

```
CMake Error at /Users/x/esp/esp-idf/tools/cmake/component.cmake:313 (message): 
Include directory '/Users/x/esp-protocols/components/asio/asio/asio/include' is not a directory.
```
This also happens when I build `asio/examples/udp_echo_server`:

```
-- Found Git: /usr/bin/git (found version "2.32.1 (Apple Git-133)") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/x/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/x/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/x/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.2.0)
-- Project sdkconfig file /Users/x/esp-protocols/components/asio/examples/udp_echo_server/sdkconfig
Loading defaults file /Users/x/esp-protocols/components/asio/examples/udp_echo_server/sdkconfig.defaults...
-- Compiler supported targets: xtensa-esp32s3-elf
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /Users/x/.espressif/python_env/idf5.2_py3.11_env/bin/python (found version "3.11.4") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "asio_udp_echo_server" version: modem-v1.0.1-dirty
-- Adding linker script /Users/x/esp-protocols/components/asio/examples/udp_echo_server/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/x/esp/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /Users/x/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/x/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/x/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/x/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/x/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/x/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
CMake Error at /Users/x/esp/esp-idf/tools/cmake/component.cmake:313 (message):
  Include directory
  '/Users/x/esp-protocols/components/asio/asio/asio/include' is not a
  directory.
Call Stack (most recent call first):
  /Users/x/esp/esp-idf/tools/cmake/component.cmake:481 (__component_add_include_dirs)
  /Users/x/esp-protocols/components/asio/CMakeLists.txt:26 (idf_component_register)

-- Configuring incomplete, errors occurred!

 *  The terminal process "/bin/bash '-c', 'cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /Users/x/esp-protocols/components/asio/examples/udp_echo_server/build -S /Users/x/esp-protocols/components/asio/examples/udp_echo_server'" terminated with exit code: 1. 
```

</details>

<details><summary>2. esp_mqtt_cxx: semaphore.h: No such file or directory</summary>
When I add component `esp_mqtt_cxx` to an idf-application (without using it) I get an error:

```
In file included from /Users/x/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/12.2.0/semaphore:35,
                 from /Users/x/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/12.2.0/stop_token:37,
                 from /Users/x/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/12.2.0/condition_variable:46,
                 from /Users/x/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/12.2.0/future:39,
                 from /Users/x/mp-v5-flexy/projects/component-test/managed_components/espressif__esp-idf-cxx/include/spi_host_cxx.hpp:16,
                 from /Users/x/mp-v5-flexy/projects/component-test/managed_components/espressif__esp-idf-cxx/spi_host_cxx.cpp:15:
/Users/x/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/12.2.0/bits/semaphore_base.h:45:11: fatal error: semaphore.h: No such file or directory
   45 | # include <semaphore.h> // sem_t, sem_init, sem_wait, sem_post etc.
      |           ^~~~~~~~~~~~~
compilation terminated.
```
The example application gives a different error:

```
Harmony:mqtt x$ idf.py menuconfig
/Users/x/esp/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: menuconfig
Running cmake in directory /Users/x/esp-protocols/examples/mqtt/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/x/.espressif/python_env/idf5.2_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/x/esp-protocols/examples/mqtt"...
CMake Error at CMakeLists.txt:7 (if):
  if given arguments:

    "STREQUAL" "linux"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
```

</details>

<details><summary>3. the link in this issue template to the documentation is wrong.</summary>It leads to a *GCC Code Coverage Report* page.</details>

I tested this on macos, using a fresh esp-idf and esp-protocols checkout.

david-cermak, 2023-06-26:
Thanks for reporting the problem with our issue template, will fix!

About other issues:
1) Please clone this repository recursively (asio is added as a submodule)

2a) There's another dependency to `esp-idf-cxx` repository, please update that version. This issue has been fixed in https://github.com/espressif/esp-idf-cxx/pull/21

2b) This example is currently being fixed in https://github.com/espressif/esp-protocols/pull/286 (while adding it to the ci). It also supports building and running on host, so it needs to know in advance which target we want to build.
As a workaround, you can set the target before building
```bash
idf.py --preview set-target linux
```
or 
```bash
idf.py set-target esp32xx
```

david-cermak, 2023-06-28:
> Sorry, I don't understand how to do this. Where should I put the esp-idf-cxx repo clone?

I'm sorry, the build is currently broken, since the dependency hasn't published the fixed version yet.
You can workaround this issue by cloning `esp-idf-cxx` and updating `idf_component.yml` to:
```patch
diff --git a/components/esp_mqtt_cxx/idf_component.yml b/components/esp_mqtt_cxx/idf_component.yml
index b5232c9..28ccf1c 100644
--- a/components/esp_mqtt_cxx/idf_component.yml
+++ b/components/esp_mqtt_cxx/idf_component.yml
@@ -2,7 +2,10 @@ version: "0.1.0"
 description: esp mqtt cxx
 url: https://github.com/espressif/esp-protocols/tree/master/components/esp_mqtt_cxx
 dependencies:
-  espressif/esp-idf-cxx: "^1.0.0-beta"
+  espressif/esp-idf-cxx:
+    version: "^1.0.0-beta"
+    override_path: "/home/david/repos/esp-idf-cxx"
+
```
will fix soon.

> but weird that I have to do this manually, the other examples use the IDF_TARGET from the visual code studio environment.

This is a limitation of our linux target support, we need to know if we're building for host before running CMake configuration step. This is not needed for other examples, as they only support building for specific ESP platform.

david-cermak, 2023-06-29:
All issues have been fixed:

* `esp_mqtt_cxx` builds clean with latest IDF: https://github.com/espressif/esp-protocols/actions/runs/5403161640/jobs/9829718640

* issue template updated

https://github.com/espressif/esp-protocols/blob/36258890495efadd758c44ea8696cf29d7e44b0a/.github/ISSUE_TEMPLATE/bug-report.yml?plain=1#L11

Thanks for reporting!

closed 2023-06-29 by david-cermak