espressif/esp-protocols #317: switch off Modem an on again

diplfranzhoepfinger, 2023-07-02:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Code: 
https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/CMUX_PWR_DOWN_UP

Log:
https://gist.github.com/diplfranzhoepfinger/744cb0e775afe25e9d8b5d70b4627d3d

I want to switch off the Modem, switch it on again, 
and then make again a CMUX Session with Data and Command Channel. 

but i get a URC of +PPPD: DISCONNECTED

how to solve this ?

diplfranzhoepfinger, 2023-07-02:
i thnik we will need a new ppp and/or netif Object, but i am far away of an Expert in this. 

@david-cermak

diplfranzhoepfinger, 2023-07-02:
i guess we need something like this: 

```c
    // UART DTE clean-up
    esp_modem_destroy(dce);
    esp_netif_destroy(esp_netif);
```

but this code is from pppos Sample, and 'esp\_modem\_destroy' is not available in C++

also which Code is required to re-establish the dce and netif ? 

this are only guesses, we need clear advice here.

diplfranzhoepfinger, 2023-07-03:
Hi, 

i tried hard to destroy DCE and Netif, 

but it leads to crashes: 

Log: 

https://gist.github.com/diplfranzhoepfinger/8d679dcb099aa86aca3d977f53543111  
`Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.`

Code:

https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/CMUX_PWR_DOWN_UP_with_new_netif

diplfranzhoepfinger, 2023-07-03:
made another Try: 

Log:   
https://gist.github.com/diplfranzhoepfinger/2fd9c7bacefa3159fa7712361cf03455  
Code:   
https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/CMUX_PWR_DOWN_UP_with_new_netif2

Not Successful.

```
assert failed: spinlock_acquire spinlock.h:122 (lock->count == 0)
```

diplfranzhoepfinger, 2023-07-03:
VERY Curious: 

after trying forth and back, 
we have now a running Configuration: 

Code: https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/leave_CMUX2
Log: https://gist.github.com/diplfranzhoepfinger/d5725e99bc821babbc180f2606c4a13a 

we have here a switch-off and a switch on of the Modem, 
start CMUX again, 
and get a IP Address. 

so now the Question shrinks down to the Question: 

why does 

https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/leave_CMUX2

work 

while 

https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/CMUX_PWR_DOWN_UP

does not work.

diplfranzhoepfinger, 2023-07-03:
Curious 2: 

Log: https://gist.github.com/diplfranzhoepfinger/4f945347e35d9684b53ea24eb5bca051
Code: https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/leave_CMUX3

still working.

diplfranzhoepfinger, 2023-07-03:
and 3

Log: https://gist.github.com/diplfranzhoepfinger/5aa5d6eee2c6a31bae07409e066a1db8
Code: https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/leave_CMUX4

not working.

diplfranzhoepfinger, 2023-07-03:
and next: 

Log:
https://gist.github.com/diplfranzhoepfinger/4ee1a7f61b107dc1ecd80c9a1d34a4b5
Code: [feature/leave_CMUX5](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/leave_CMUX5)

tokk over 3 Lines from the non-Working branch, but still working ? 

isn't this curious ?

david-cermak, 2023-10-11:
I guess you just have to send a `hanpup()` command (`ATH`) to complete the closure correctly.

david-cermak, 2023-11-28:
Closing, feel free to reopen

closed 2023-11-28 by david-cermak