espressif/esp-protocols #318: Need a Diagram for CMUX States

diplfranzhoepfinger, 2023-07-02:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to sketch together a mermaid Diagram about the CMUX States:

diplfranzhoepfinger, 2023-07-02:
@david-cermak can you check this 3 Diagramms, 

and comment them. 

my goal is to add also Descriptions of what this state means. 

like "2 Command Channels now" 

and so on ... 

after this is correct, we should take it to the Documentation.

david-cermak, 2023-07-19:
Thanks for sketching these diagrams! They look correct to me!

Besides adding them to the docs, we could also "add" them to unit tests, i.e. add new unit tests exercising these transitions.

diplfranzhoepfinger, 2023-07-19:
yes, agree !

diplfranzhoepfinger, 2023-07-21:
So, 

in addition to this we will need Diagrams what to do when. 

Example my Problem recently: 

```
E (00:01:16.622) cmux_example: Cannot get IP within specified timeout... exiting
```

how to overcome e.g. such a Problem. 

if it happens in Field, WHAT Steps to do to "later" try again getting a IP.