espressif/esp-protocols #319: websocket crash when using invalid uri

MigliusBudriunas, 2023-07-03:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Use invalid Uri ex: "ws://192.168.1.182:80800922".
After esp_websocket_client_init function fails in esp_websocket_client_set_uri function it goes to __websocket_init_fail and finally in destroy_and_free_resources it crashes, because client->transport_list is still NULL. 
It should be enough to reproduce problem with default example and changed Uri.

To solve the problem checking for NULL in destroy_and_free_resources function was sufficient.

gabsuren, 2023-07-24:
@MigliusBudriunas thank you for opening the issue.
In the `destroy_and_free_resources(...)` function, there is a NULL guard check for `transport_list`.
You can find the specific code section here:
    https://github.com/espressif/esp-protocols/blob/f6d5186e5b76820ba50401240f168e3990e91d18/components/esp_websocket_client/esp_websocket_client.c#L427C1-L429C6
 
Could you please provide an example code that allows us to reproduce the issue?

Thank you!

MigliusBudriunas, 2023-07-28:
In 1.0.0 version it was not fixed. In 1.0.1 version it is fixed now. 
I see all works now. Sorry for inconvenience, and thank you.

closed 2023-07-28 by gabsuren