espressif/esp-protocols #320: need a Forced Exit of CMUX Mode

diplfranzhoepfinger, 2023-07-03:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

need a Forced Exit of CMUX Mode

diplfranzhoepfinger, 2023-07-03:
so we hacked something together, 
but i fear this is not 100% the right way: 

I would more like to have a Mode like: `CMUX_MANUAL_EXIT_FORCED`,
so we call it only when we know that we want to "force" esp-modem to leave cmux without checking the circumstances. 

Log: https://gist.github.com/diplfranzhoepfinger/c7ab11ed17bde1920a7094c6288f1cc4

Code: https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/CMUX_FORCE_Vasil

diplfranzhoepfinger, 2023-07-03:
we hacked further: 

Log: [https://gist.github.com/diplfranzhoepfinger/a36fd06a8b3ed9398deb52cc9fb774b0](https://gist.github.com/diplfranzhoepfinger/a36fd06a8b3ed9398deb52cc9fb774b0)

Code: [https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/CMUX_FORCE_Vasil2](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/CMUX_FORCE_Vasil2)

as expected, the Code is crashing, BUT, at least, we can start Cmux again and again get a IP Address. 

so @david-cermak i think the Problem is solvable, but needs expertise of the Internals.

diplfranzhoepfinger, 2023-07-18:
@david-cermak any response ?

david-cermak, 2023-07-19:
Noted, I think it would be really useful to kind of reset to a default state on ESP32.

Could we use this "Unknown state" for that? For example if we know that we got somehow out of sync
1) we switch the modem to the "unknown" (just need to allow this transition)
2) from there, everything should be the same as after init

diplfranzhoepfinger, 2023-07-19:
cool idea. go forward with it.

VasilNikolovRilabs, 2023-10-17:
Same issue on our side, we need a solution for ESP modem to allow a safe start of CMUX from any situation.

HDLA-BG, 2023-10-17:
Same issue here!

DOevermann, 2023-10-17:
Same issue here!

david-cermak, 2023-11-03:
Created https://github.com/espressif/esp-protocols/pull/416 to support the forced transitions.

As a workaround, you can destruct modem's DCE (to enter the `UNDEF` mode) and then switch to:
1) `CMUX_MANUAL_ENTER`
2) `CMUX_MANUAL_EXIT`