espressif/esp-protocols #323: 'OK' is tangling in the Receive Buffer

diplfranzhoepfinger, 2023-07-06:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Code: https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/AT_CPSI

Log: https://gist.github.com/diplfranzhoepfinger/dde959578cb954f23908c07fb10b463a

we observe the following: 

the Command "AT+CPSI?" is sent out. 
the Response is coming inside the 500ms Timeout. 
but the OK is only received after we send out another "AT" with `dce->sync();`

this happens in CMUX Mode. 

is the receive Buffer not flushed while waiting for the OK ? Or what can be the Reason ?

diplfranzhoepfinger, 2023-07-06:
Now test to do this multiple times. 

Code: [https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/AT_CPSI3](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/AT_CPSI3)

Log:

diplfranzhoepfinger, 2023-07-18:
@david-cermak any news ? 

SIMCOM side or Espressif side ?

david-cermak, 2023-07-19:
Those are probably leftovers from the previous command. Does it cause any trouble?

We can flush the buffers before sending a new command.

david-cermak, 2023-07-19:
> yes, as we get "Inquiring UE system information ERROR or TIMEOUT" 

it was probably timeout, so the `OK` was actually coming, but didn't make it before timeout. And the `OK` response was buffered and appeared in the next command processing.

Clearing buffers before commands is probably the right thing to do, but wouldn't help much (just gives ~4 bytes to buffer size available)

diplfranzhoepfinger, 2023-07-19:
> > yes, as we get "Inquiring UE system information ERROR or TIMEOUT"
> 
> it was probably timeout, so the `OK` was actually coming, but didn't make it before timeout. And the `OK` response was buffered and appeared in the next command processing.

Thw Question is: 

in Command Mode we never see this Problem. 

in CMUX Mode we see this Problem even if we make a really long timeout to this Command, like 15s. 

and the OK comes once we issue a "AT" to the Modem. 

so we are unsure if this is a SIMCOM Modem or a esp-modem Problem.

david-cermak, 2023-07-19:
> in CMUX Mode we see this Problem even if we make a really long timeout to this Command, like 15s. 

Could you please connect another UART-USB converter to modem Tx line and check if the `OK` comes before or after those 15s?
It might be theoretically also possible that the modem flow control is preventing the SIMCOM device to continue? (ESP32 is not ready to receive? in that case #312 should fix it as well)

diplfranzhoepfinger, 2023-07-21:
Test here: 
https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-simple_cmux_client/tree/feature/AT_CPSI4

david-cermak, 2023-10-09:
If I understand correctly the change in #312 fixed this issue, correct?

If not please reopen

closed 2023-10-09 by david-cermak