espressif/esp-protocols #327: Commands are echoed even if we set Echo off by ATE0

diplfranzhoepfinger, 2023-07-21:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

see this Log: 
https://gist.github.com/diplfranzhoepfinger/7656fcdd4f1ac351ca72b3d092762a59

we clearly send "ATE0" to the Modem, which means "Echo mode off" 

But later in the log we see: 

```
V (00:00:15.761) Send: 0x3fcf4884   41 54 2b 43 4f 50 53 3f  0d                       |AT+COPS?.|
V (00:00:15.762) CMUX Received: 0x3fcf5c74   f9 09 ef 13 41 54 2b 43  4f 50 53 3f 0d 2e f9 f9  |....AT+COPS?....|
```

so it seems everything is still echoed

Can you explain this ?

diplfranzhoepfinger, 2023-07-21:
Test: 

https://gist.github.com/diplfranzhoepfinger/ee04afe741f0dddd316cbba36bdfec9a

david-cermak, 2023-10-09:
@diplfranzhoepfinger I don't see this behaviour on any device. Is it possible that the problem lies somewhere within the A7672 device?

david-cermak, 2023-10-31:
@diplfranzhoepfinger Any update here? Can we close this?

diplfranzhoepfinger, 2023-11-01:
I will check with the new USB Version.

david-cermak, 2023-11-28:
I assume that it's not reproducible anymore.
Please reopen otherwise.

closed 2023-11-28 by david-cermak