espressif/esp-protocols #328: How to disconnect and reconnect from the CMUX example code?

fmadhurakkariyan, 2023-07-25:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Suppose a condition where the internet connection gets dropped and  I want to try to disconnect and then try to reconnect from the CMUX sample code. How would I go about doing this?

david-cermak, 2023-07-27:
Probably the fastest option would be to exit and re-enter the CMUX mode, if your device supports it.

If not, you can use manual CMUX commands to setup the PPP mode again, but only on the virtual channel already created.
I'd suggest playing with the modes a bit using the modem-console example. You can use these commands:

https://github.com/espressif/esp-protocols/blob/5143f5ac01f27b669add28c80ac809b023d2cb1f/components/esp_modem/examples/modem_console/main/modem_console_main.cpp#L238C1-L247

If you get disconnected, some devices also exit the PPP mode automatically, if that's the case, you can:
* `set_mode CMUX3` (CMUX_MANUAL_SWAP) to be able to talk to the secondary terminal
* `cmd AT` to check if you're already in command mode
* `set_mode CMUX4` (CMUX_MANUAL_DATA) to re-enter the PPP mode

david-cermak, 2023-10-09:
Any update?

If not, I'm going to merge this test https://github.com/espressif/esp-protocols/pull/363
are refer to the manual CMUX mode processing in https://github.com/espressif/esp-protocols/blob/dee2df69ef4a31bfe179fae885db7ef3d10e7e3e/components/esp_modem/test/target_ota/main/simple_cmux_client_main.cpp#L154-L156 as a resolution

closed 2023-10-09 by david-cermak