espressif/esp-protocols #331: Don't get IP assign event when using other than standard baud rate (Quectel)

sporubcan, 2023-07-26:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I am trying to use esp_modem with Quectel EC21, which is very similar to officially supported BG96. 
I was successful at obtaining IP address at modem default baud rate (115200bps). In case I change baud rate to lower or higher baud rate, I dont get IP anymore. Could you help with this issue, please? 

My code:

```
void app_main(void)
{
    esp_log_level_set("*", ESP_LOG_VERBOSE);

    ESP_LOGI(TAG, "Startup..");
    ESP_LOGI(TAG, "Free memory: %d bytes", esp_get_free_heap_size());
    ESP_LOGI(TAG, "IDF version: %s", esp_get_idf_version());

    // Init and register system/core components */
    //
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed, NULL));

    event_group = xEventGroupCreate();

    // Reset modem
    //
    ESP_LOGI(TAG, "Power-on / reset modem START");

    gpio_set_direction(MODEM_PWKEY, GPIO_MODE_OUTPUT);
    gpio_set_direction(MODEM_RST, GPIO_MODE_OUTPUT);
    gpio_set_direction(MODEM_DTR, GPIO_MODE_OUTPUT);
    
    gpio_set_level(MODEM_RST, 0);
    gpio_set_level(MODEM_PWKEY, 0);
    gpio_set_level(MODEM_DTR, 0);

    initializeEC21();

    //vTaskDelay(10 / portTICK_PERIOD_MS);

    // Configure the DTE
    //
    esp_modem_dte_config_t dte_config = ESP_MODEM_DTE_DEFAULT_CONFIG();
    dte_config.uart_config.tx_io_num = MODEM_TXD;
    dte_config.uart_config.rx_io_num = MODEM_RXD;
    dte_config.uart_config.rx_buffer_size = 1024;
    dte_config.uart_config.tx_buffer_size = 512;
    dte_config.uart_config.cts_io_num = MODEM_CTS;
    dte_config.uart_config.rts_io_num = MODEM_RTS;
    dte_config.uart_config.event_queue_size = 30;
    dte_config.task_stack_size = 2048;
    dte_config.task_priority = 5;
    dte_config.dte_buffer_size = 1024 / 2;
    // dte_config.uart_config.source_clk = UART_SCLK_REF_TICK;
    dte_config.uart_config.baud_rate = 115200;
    dte_config.uart_config.data_bits = UART_DATA_8_BITS;
    dte_config.uart_config.parity = UART_PARITY_DISABLE;
    dte_config.uart_config.stop_bits = UART_STOP_BITS_1;
    dte_config.uart_config.flow_control = ESP_MODEM_FLOW_CONTROL_NONE;

    // Configure the DCE
[... 175 lines cut ...]
D (74646) esp-netif_lwip-ppp: Phase Disconnect
D (74646) esp-netif_lwip-ppp: Phase Dead
I (74646) esp-netif_lwip-ppp: Connection lost
D (74646) event: running post IP_EVENT:7 with handler 0x400d72b8 and context 0x3ffb8a20 on loop 0x3ffb7ad0
0x400d72b8: on_ip_event at /home/sporubcan/Nextcloud/___FIRMWARE_SOFTWARE/esp_modem_test/main/main.c:47

D (74656) esp_modem_test: IP event! 7
I (74656) esp_modem_test: Modem Disconnect from PPP Server
D (74666) event: running post IP_EVENT:7 with handler 0x400e74a8 and context 0x3ffbbe4c on loop 0x3ffb7ad0
0x400e74a8: esp_netif_action_disconnected at /home/sporubcan/esp/esp-idf/components/esp_netif/esp_netif_handlers.c:91

D (74676) esp_netif_handlers: esp_netif action disconnected with netif0x3ffb8a9c from event_id=7
D (74686) esp_netif_lwip: check: remote, if=0x3ffb8a9c fn=0x400e812c
0x400e812c: esp_netif_down_api at /home/sporubcan/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1363

D (74686) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffb8a9c
D (74696) esp_netif_lwip: check: local, if=0x3ffb8a9c fn=0x400e885c
0x400e885c: esp_netif_update_default_netif_lwip at /home/sporubcan/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:219

D (74706) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffb8a9c
V (74706) esp_netif_objects: esp_netif_next_unsafe 0x0
V (74716) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffb8a9c
V (74716) esp_netif_objects: esp_netif_next_unsafe 0x3ffb8a9c
D (74726) esp_netif_lwip: call api in lwip: ret=0x0, give sem
```

sporubcan, 2023-07-26:
Solved :) I noticed that **esp_modem_set_baud** doesn't change real ESP UART interface baud rate. When I did hard set Quectel's baud rate using **AT+IPR=9600&W** and reboot software with preset 9600bps baud rate in **dte_config.uart_config.baud_rate parameter**, it worked.

boraozgen, 2023-07-28:
I would like to achieve this without storing the baud rate in the modem. How is the ESP UART baud rate changed at runtime?

david-cermak, 2023-07-28:
You can use `uart_set_baudrate(dte_config.uart_config.port_num, ...);`

boraozgen, 2023-07-28:
Thanks for the quick reply, this means that the UART API must be accessed directly despite the abstraction over esp_modem_uart... I guess it would be better for this to be done in esp_modem_set_baud automatically.

david-cermak, 2023-08-28:
>  I guess it would be better for this to be done in esp_modem_set_baud automatically.

That's a good point, and I think it would make sense if the `esp_modem` completely **consumed** the UART driver.
In the current design it creates an UART terminal and passes it to the DTE, which then treats it as a generic terminal, so we'd have to either create a generic terminal API to update the baudrate; or keep a copy of the uart terminal and rely on a specific uart-terminal API. None of the options is perfect, but considering the pros and cons, I'd just like accessing UART API directly from the user code, since it's locked separately, safe and simple.

david-cermak, 2023-08-28:
Another alternative might be to detach a terminal from higher layers, perform the specific updates (set the baudrate), and re-attach the terminal back. Here'are the changes to support this for setting UART flow control. https://github.com/espressif/esp-protocols/compare/master...david-cermak:esp-protocols:feat/modem_dte_detach_term

(Note that setting baud rate or flow control is not a generic terminal property that should be propagated up to the DTE/DCE layers)

closed 2023-10-09 by david-cermak