espressif/esp-protocols #333: Race condition in esp_modem_dte

mariusz-r, 2023-07-31:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I seem to be hitting [this](https://github.com/espressif/esp-protocols/blob/master/components/esp_modem/src/esp_modem_dte.cpp#L62) exception:

I think that `set_read_cb` is being run twice before [this](https://github.com/espressif/esp-protocols/blob/master/components/esp_modem/src/esp_modem_dte.cpp#L61) check. It seems to happen when the timings are being "relaxed" by increased logging level, or some higher priority task running in the background. I've found [this](https://github.com/espressif/esp-protocols/commit/a02bf05eed7c96f703958a3a3f52c6234e386ffb) commit, but yeah, the race is still there. Having the `GOT_LINE` signal, but then also reading the `result` variable doesn't guarantee atomicity.
Let me know if you need more details.

david-cermak, 2023-08-14:
Thanks for the report! Could you please check if this PR https://github.com/espressif/esp-protocols/pull/312 fixes your issues?
This simply adds a mutex to protect the callback.

The thread safety was kind of handled (in a very fragile way) by cooperation between the DTE class and its terminal (so the attached terminal would signal when it's okay to clean the callback), but it got broken when adding USB terminal in 44bae24c78d15f8b0a028e3016d5043c1198fe01. I'm sorry, I posted the fix some time ago but haven't really gotten to merge it and resolve this issue sooner.

mariusz-r, 2023-08-17:
> Thanks for the report! Could you please check if this PR https://github.com/espressif/esp-protocols/pull/312 fixes your issues?

Yes, I don't see the race, but there seems to be one regression: `command_result::OK` is returned instead of  `command_result::TIMEOUT` in `generic_command`. `void DTE::set_command_callbacks` probably needs:

```
command_cb.result = command_result::TIMEOUT;
```

david-cermak, 2023-08-17:
Good catch, thanks!

Have set the default state into:

https://github.com/espressif/esp-protocols/blob/22da99a540c135fd2f6610884ce8916714d6b253/components/esp_modem/src/esp_modem_dte.cpp#L120C19-L133

mariusz-r, 2023-08-17:
Also, my confirmation above might be premature... I still seem to be hitting https://github.com/david-cermak/esp-protocols/blob/fix/minor_modem_refactor/components/esp_modem/src/esp_modem_dte.cpp#L321, but this time when logging level is `INFO`, or higher.

david-cermak, 2023-08-17:
You're correct, there might be also other issues with the PR. I'm sorry, it's been 2 months since I touched it and I need to spend some time with it.

Clearly this change https://github.com/espressif/esp-protocols/issues/333#issuecomment-1682088473 did help introducing a data race, since we need to set the `TIMEOUT` state while the callback is **locked**. Was able to reproduce the bug just running a host test, and our CI also indicated something wrong: https://github.com/espressif/esp-protocols/actions/runs/5890096726/job/15974984309

david-cermak, 2023-08-28:
I'm sorry it took so long, but I've finally merged https://github.com/espressif/esp-protocols/pull/312

You were right, there were multiple issues with that PR, not only the locking:
* was checking the `std::function<>` for `nullptr` after it's moving it!  https://github.com/espressif/esp-protocols/compare/22da99a540c135fd2f6610884ce8916714d6b253..db9f3f3bbc1d065f05acb095a361a6148f5464c2
* the PR also fixed network init, so the example got an IP *before* we registered the event handler https://github.com/espressif/esp-protocols/pull/312/commits/cb6e03ac628eb937befc9d31883c3e73a6a35de6#diff-e5cf7c1d6038b9cbf36cb92e72665046de2974bfffaf5c4fa5bcd30fc1d9688d
* and other minor issues

Now it should be fixed and merged to master.

I'll test it some more, and if everything works I'd create a new release.

david-cermak, 2023-10-09:
This issue has been resolved in cb6e03ac, after merging #312; closing now, only after proper testing and releasing of a new version.

closed 2023-10-09 by david-cermak