espressif/esp-protocols #335: Using this repository with vscode (esp-idf extension)

joelguittet, 2023-08-07:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### What component are you using? If you choose Other, provide details in More Information.

Other

### component version

websocket v1.0.1

### IDF version.

v5.0.2

### More Information.

Hello,

How to use this repository to have the components available when using the esp-idf extension in vscode (https://github.com/espressif/vscode-esp-idf-extension) ? Where to declare the path to the repository maybe ?

Can you document this in the README of this project  or point the relevant document ? I have not found it.

Thanks
Joel

gabsuren, 2023-08-07:
@joelguittet Please ask the question in the forum. Thanks!
https://www.esp32.com/viewforum.php?f=23

closed 2023-08-07 by gabsuren

joelguittet, 2023-08-08:
Hello @gabsuren 
Two times I m posting the question on the forum, it does not get accepted ! Despite it s polite, with no link etc etc.
What should I do ?! No way to get the information here ?
Thanks
Joel

joelguittet, 2023-08-09:
Ok, got it on the forum now.