espressif/esp-protocols #337: esp_event.h not found using the esp_websocket_client components

joelguittet, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

While using the esp_websocket_client component I get the following issue:

```
C:/path-to-my-project/managed_components/espressif__esp_websocket_client/include/esp_websocket_client.h:16:10: fatal error: esp_event.h: No such file or directory
```

This is very very similar to https://github.com/espressif/esp-protocols/issues/104 but not exactly the same.

When moving "esp_event" from PRIV_REQUIRES to REQUIRES in the CMakeLists.txt of the components, this solves my issue and I'm able to build my project. Can you confirm the right place of this ?

Regards
Joel

gabsuren, 2023-08-09:
@joelguittet thank you for your question. 
What version of `esp_websocket_client` and `esp-idf` do you use? The `PRIV_REQUIRES` should be work correctly, there are [examples](https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client/examples/target) of using it in that way
https://github.com/espressif/esp-protocols/blob/5143f5ac01f27b669add28c80ac809b023d2cb1f/components/esp_websocket_client/CMakeLists.txt#L20

Please check the latest version of [esp_websocket_client](https://github.com/espressif/esp-protocols/tree/5143f5ac01f27b669add28c80ac809b023d2cb1f/components/esp_websocket_client) and `esp-idf: version >=5.0`  let us know

Regards,
Suren

joelguittet, 2023-08-09:
Hello @gabsuren 

I'm using `esp_websocket_client` v1.0.1 (the latest) and IDF version 5.0.2.
I suspect the example is working because `websocket_example.c` itself include `esp_event.h` before including the websocket client header.
Note: the example include `esp_event.h` twice ... can be cleaned.

Joel

gabsuren, 2023-08-16:
@joelguittet thank you for your notes. 
I removed both includes of `esp_event.h` from `websocket_example.c` and it still was building successfully (as it was included in `esp_websocket_client.h`). 
https://github.com/espressif/esp-protocols/blob/42fe60828f63b38989fd1a44e731b2abeabe3461/components/esp_websocket_client/include/esp_websocket_client.h#L16

joelguittet, 2023-08-16:
Ok, thanks.
Probably because of the environment, Windows or something else.
A little bit anoying because I can t use the registry as is. I had to clone and copy the files locally to solve my issue, just to patch the cmakelist file.

david-cermak, 2023-08-16:
@joelguittet Could you please provide more information about your project? Ideally a minimum example, or maybe just your makefiles with component yaml files? Do you somehow reduce/modify the list of components?

@gabsuren I think the dependency should be public in this case, since we **declare** our own `WEBSOCKET_EVENTS` in the header file, so the websocket component have to make sure the `esp_event` dependency is transiently brought into the build for all components that include `esp_websocket_client`.

joelguittet, 2023-08-16:
Sure I can, but this is not very simple project to be honest.
Relevant source file is: https://github.com/joelguittet/mender-mcu-client/blob/feature/integrate-troubleshoot-addon/platform/net/esp-idf/src/mender-websocket.c
Relevant CMakeLists.txt is: https://github.com/joelguittet/mender-mcu-client/blob/feature/integrate-troubleshoot-addon/esp-idf/CMakeLists.txt (not there is a special include of the esp_websocket_client because it is optional in the project and it integrated depending of one CONFIG).

david-cermak, 2023-08-17:
Thanks for sharing!

> Relevant source file is: https://github.com/joelguittet/mender-mcu-client/blob/feature/integrate-troubleshoot-addon/platform/net/esp-idf/src/mender-websocket.c

So when compiling this file, you get an error of missing `esp_event.h` from `esp_websocket_client.h`, but this line yields no error?

https://github.com/joelguittet/mender-mcu-client/blob/54ac7cc06261404cabd4d3dff55f07a25451eb54/platform/net/esp-idf/src/mender-websocket.c#L28

> Relevant CMakeLists.txt is: https://github.com/joelguittet/mender-mcu-client/blob/feature/integrate-troubleshoot-addon/esp-idf/CMakeLists.txt 

This is a component makefile, I'm more interested in the project makefile (the one used when building your binary, in IDF it's typically the one that includes:

```cmake
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
```

Is it somewhere in you test directory? Do you use some mocks or building on host?

I think you can workaround the issue by adding a dependency to `esp_event` in your component:

https://github.com/joelguittet/mender-mcu-client/blob/54ac7cc06261404cabd4d3dff55f07a25451eb54/esp-idf/CMakeLists.txt#L74

(but to make the dependency optional you might need to use 
```cmake
if(CONFIG_WITH_WEBSOCKETS)
  idf_component_optional_requires(PRIVATE esp_event)
endif()
```
after `idf_component_register()` instead of adding the `esp_event` into the `${priv}` list)

joelguittet, 2023-08-17:
@david-cermak thanks a lot for this detailed answer !
I have tested the `idf_component_optional_requires` solution that I wasn't aware at all. This is doing the magic, no more issues to build using the websocket client from the registry.

I have added this:
```
if (CONFIG_MY_OPTION_WITH_WEBSOCKET)
    idf_component_optional_requires(PRIVATE espressif__esp_websocket_client esp_event)
endif()
```

Issue is solved :-)

closed 2023-08-17 by joelguittet