espressif/esp-protocols #339: Null pointer exception (LoadProhibited) on incoming UART data after failed CMUX initialization

victorrar, 2023-08-09:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Issue:
After unsuccessful CMUX initialization, incoming UART data will cause panic.

<details><summary>Details</summary>
<p>

E (00:02:45.702) uart: uart_read_bytes(1283): uart data null
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4000c250  PS      : 0x00060c30  A0      : 0x800fcb83  A1      : 0x3ffd0210  
A2      : 0x00000000  A3      : 0x0000000a  A4      : 0xffffffff  A5      : 0xffffffff  
A6      : 0x3f414474  A7      : 0x00000000  A8      : 0x800fb528  A9      : 0x3ffd01a0  
A10     : 0xffffffff  A11     : 0x00000000  A12     : 0x00000001  A13     : 0xffffffff  
A14     : 0x3ffd01e0  A15     : 0x0000000c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x4000c24d:0x3ffd0210 0x400fcb80:0x3ffd0230 0x400fb6e5:0x3ffd0260 0x400fb7f1:0x3ffd02a0

</p>
</details> 

I'm testing it on SIM7080G modem over UART, here is an example snippet.

```c++
auto dte = esp_modem::create_uart_dte(&storedConfig.dte_config);
auto dce = esp_modem::create_SIM7070_dce(&storedConfig.dce_config, dte, netifNB);
...
dce->set_mode(esp_modem::modem_mode::CMUX_MODE);

//<now I send some data or touch a floating RX pin with a finger>
```

This is caused in esp_modem_dte.cpp, because when CMUX is created, the terminal and buffer are moved to it and no longer valid.
But if `cmux_term->init()` fails, it does not restore the validity of the terminal/buffer.

```c++
bool DTE::setup_cmux()
{
    cmux_term = std::make_shared<CMux>(primary_term, std::move(buffer));
    if (cmux_term == nullptr) {
        return false;
    }
    if (!cmux_term->init()) {
        return false;
    }
    primary_term = std::make_unique<CMuxInstance>(cmux_term, 0);
    if (primary_term == nullptr) {
        return false;
    }
    secondary_term = std::make_unique<CMuxInstance>(cmux_term, 1);
    return true;
}

```

The UART callback directly gets the data pointer of the buffer (which was moved) and now .get() returns 0, causing a null pointer exception.

```c++
command_result DTE::command(const std::string &command, got_line_cb got_line, uint32_t time_ms, const char separator)
{
...
    primary_term->set_read_cb([this, got_line, separator](uint8_t *data, size_t len) {
        if (!data) {
            data = buffer.get();
            len = primary_term->read(data + buffer.consumed, buffer.size - buffer.consumed);
        } else {
            buffer.consumed = 0; // if the underlying terminal contains data, we cannot fragment
        }
        if (memchr(data + buffer.consumed, separator, len)) {
            result = got_line(data, buffer.consumed + len);
            if (result == command_result::OK || result == command_result::FAIL) {
                signal.set(GOT_LINE);
                return true;
            }
        }
        buffer.consumed += len;
        return false;
    });
...
```
The top branch is executed as `on_read` always called with nullptr for data
```c++
void UartTerminal::task()
{
...
            case UART_DATA:
                uart_get_buffered_data_len(uart.port, &len);
                if (len && on_read) {
                    on_read(nullptr, len);
                }
                break;
...
```

### Proposed solution
I suggest detaching the terminal and restoring primary_term/buffer on fail. 
I've tested it in my project, and it works correctly, not causing the error anymore.
I'm not sure about the implementation details, so I'm glad for this code sample to be reviewed.
Also, I can create a pull request if it would help.
```c++
bool DTE::setup_cmux()
{
    cmux_term = std::make_shared<CMux>(primary_term, std::move(buffer));
    if (cmux_term == nullptr) {
        return false;
    }
    if (!cmux_term->init()) {
        auto det = cmux_term->detach();
        primary_term = std::move(det.first);
        buffer = std::move(det.second);
        cmux_term = nullptr;
        return false;
    }
    primary_term = std::make_unique<CMuxInstance>(cmux_term, 0);
    if (primary_term == nullptr) {
        auto det = cmux_term->detach();
        primary_term = std::move(det.first);
        buffer = std::move(det.second);
        cmux_term = nullptr;
        return false;
    }
    secondary_term = std::make_unique<CMuxInstance>(cmux_term, 1);
    if(secondary_term == nullptr){
        auto det = cmux_term->detach();
        primary_term = std::move(det.first);
        buffer = std::move(det.second);
        cmux_term = nullptr;
    }
    return true;
}
```

### Affected code
https://github.com/espressif/esp-protocols/blob/5143f5ac01f27b669add28c80ac809b023d2cb1f/components/esp_modem/src/esp_modem_dte.cpp#L87C1-L102C2

david-cermak, 2023-08-14:
@victorrar Thanks for reporting this issue and proposing the solution!

The idea LGTM, feel free to open a PR.
just a nitpick that in your implementation, there's a bit of a repetitive section on detaching/cleanup that could be handled differently (e.g. renaming current `setup_cmux()` -> `try_setup_cmux()` and:

```cpp
bool DTE::setup_cmux()
{
    if (!try_setup_cmux()) {
        // detach + cleanup
        return false;
    }
    return true;
}
```
or using some RAII construct, but that could be discussed in the PR)

victorrar, 2023-08-18:
@david-cermak Thanks for your response; here it is https://github.com/espressif/esp-protocols/pull/343!

closed 2023-08-28 by david-cermak