espressif/esp-protocols #351: Modem Exits Data Mode After CONNECT

AdiXPS, 2023-09-05:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

First of all henstly the version is IDF 4.3. we cannot change it due to production retesting timelines. We notice that once we get an IP and we get CONNECT 1500000 the modem after couple of seconds send OK and exit the data mode (no hangup). Cannot find any trace in the code to trigger this? Modem is SIM7000G

Logs from modem responce:

`"5/9/2023 10:15:17 AM",+CGREG: 0,1

"5/9/2023 10:15:17 AM",OK

"5/9/2023 10:15:19 AM",OK

"5/9/2023 10:15:19 AM",CONNECT 150000000
"5/9/2023 10:15:19 AM",~}#!}!} } }9}"}&} } } } }#}%#}%}%}&(u}'}"}(}"fט~~}#!}"}!} }4}"}&} } } } }%}&%|£}'}"}(}"x~~}#!}!}!} }8}"}&} } } } }#}$#}%}&(u}'}"}(}"t~~}#!}+}"} }((u~~#  0~~! 
"5/9/2023 10:15:20 AM",
"5/9/2023 10:15:20 AM",
"5/9/2023 10:15:20 AM",
"5/9/2023 10:15:20 AM",
"5/9/2023 10:15:20 AM",
"5/9/2023 10:15:20 AM",A~~!  g~~! 
"5/9/2023 10:15:20 AM", -ז~~! »~~! 
"5/9/2023 10:15:20 AM",»Tכ7Tכ9ז2~~! 
"5/9/2023 10:15:20 AM",»Tכ7Tכ9זF~
"5/9/2023 10:15:21 AM",**OK**`

Then the PPP will not work due to this issue since the modem is out from data mode (we've probe it for AT command and its working as command mode) Sometimes, its will do as below (twice CONNECT) and even sending SIM7000G R1529 (same as ATI command!)?:

`"5/9/2023 11:12:26 AM",OK

"5/9/2023 11:12:27 AM",+CGREG: 0,1

"5/9/2023 11:12:27 AM",OK

"5/9/2023 11:12:28 AM",CONNECT 150000000
"5/9/2023 11:12:28 AM",~ }#└!}!} } }9}"}&} } } } }#}%┬#}%}%}&╙]}8╟}'}"}(}"▓5~~ }#└!}"}!} }4}"}&} } } } }%}&ДКмС}'}"}(}"╒N~~ }#└!}!}!} }8}"}&} } } } }#}$└#}%}&╙]}8╟}'}"}(}"а} ~~ }#└!}+}"} }(╙]}8╟д}:~~└#  ¤0~~А! Б
"5/9/2023 11:12:30 AM",
"5/9/2023 11:12:30 AM",Г
"5/9/2023 11:12:30 AM",В
"5/9/2023 11:12:30 AM",
"5/9/2023 11:12:30 AM",Д
"5/9/2023 11:12:30 AM",A~~А!  g├~~А! 
"5/9/2023 11:12:30 AM", - ц~~А! ╗Щ~~А! diАБTы7ГTы9цn~~А! diАБTы7ГTы9ц73~
"5/9/2023 11:12:31 AM",OK
"5/9/2023 11:12:31 AM",~!E  ,  @ -▌C▀u'diАXєЁЮ╧ZГ╟tЙT`·Ё5  P=│~
"5/9/2023 11:12:32 AM",SIM7000G R1529

"5/9/2023 11:12:32 AM",CONNECT 150000000
"5/9/2023 11:12:32 AM",~!E  ,  @ -▌C▀u'diАXєЁЮ╧ZГ╟tЙT`·Ё5  P=│~~!E  ,  @ -▌C▀u'diАXєЁЮ╧ZГ╟tЙT`·Ё5  P=│~~!E  (@@ -╘═C▀u'diАXєЁЮ╧ZД╟tМбPўг6О  э_~~!E  (@@ -╘╠C▀u'diАXєЁЮ╧ZД╟tП|Pўг3│  n6~~!E Ы@@ -╤XC▀u'diАXєЁЮ╧ZД╟tП|Pўг═;  HTTP/1.1 200 OK
"5/9/2023 11:12:36 AM",Access-Control-Allow-Origin: *
"5/9/2023 11:12:36 AM",X-Powered-By: Express`

AdiXPS, 2023-09-05:
Detailed log:
"5/9/2023 1:38:59 PM",V (12467) esp_netif_objects: esp_netif_next_unsafe 0x0[0m
"5/9/2023 1:38:59 PM",V (12477) esp_netif_objects: esp_netif_next_unsafe 0x0[0m
"5/9/2023 1:38:59 PM",D (12477) esp_netif_objects: esp_netif_add_to_list 0x3ffc64ec[0m
"5/9/2023 1:38:59 PM",D (12487) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)[0m
"5/9/2023 1:38:59 PM",D (12497) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffc681c[0m
"5/9/2023 1:38:59 PM",D (12507) esp-netif_lwip-ppp: Phase Dead[0m
"5/9/2023 1:38:59 PM",D (12507) esp-modem: Sending command:AT
"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",V (12527) esp-modem: uart_get_buffered_data_len()=9[0m
"5/9/2023 1:38:59 PM",V (12527) esp-modem: pattern-detection: 0x3ffb7c94   41 54 0d 0d 0a 4f 4b 0d  0a                       |AT...OK..|[0m
"5/9/2023 1:38:59 PM",D (12537) esp-modem: Handling line: >>AT

"5/9/2023 1:38:59 PM",<<[0m
"5/9/2023 1:38:59 PM",[0;32mI (12537) dce_service: esp_modem_dce_handle_response_default AT

"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",D (12547) esp-modem: Handling line: >>OK

"5/9/2023 1:38:59 PM",<<[0m
"5/9/2023 1:38:59 PM",[0;32mI (12547) dce_service: esp_modem_dce_handle_response_default OK
"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",D (12557) dce_service: sync ok[0m
"5/9/2023 1:38:59 PM",D (12557) esp-modem: Sending command:ATE0
"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",V (12587) esp-modem: uart_get_buffered_data_len()=11[0m
"5/9/2023 1:38:59 PM",V (12587) esp-modem: pattern-detection: 0x3ffb7c94   41 54 45 30 0d 0d 0a 4f  4b 0d 0a                 |ATE0...OK..|[0m
"5/9/2023 1:38:59 PM",D (12587) esp-modem: Handling line: >>ATE0

"5/9/2023 1:38:59 PM",<<[0m
"5/9/2023 1:38:59 PM",[0;32mI (12597) dce_service: esp_modem_dce_handle_ate0 ATE0

"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",[0;32mI (12597) dce_service: esp_modem_dce_handle_ate0 ATE0

"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",D (12607) esp-modem: Handling line: >>OK

"5/9/2023 1:38:59 PM",<<[0m
"5/9/2023 1:38:59 PM",[0;32mI (12607) dce_service: esp_modem_dce_handle_ate0 OK
"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",[0;32mI (12617) dce_service: disable echo ok[0m
"5/9/2023 1:38:59 PM",D (12617) esp-modem: Sending command:AT+CGDCONT=1,"IP","jawalnet.com.sa"
"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",V (12697) esp-modem: uart_get_buffered_data_len()=6[0m
"5/9/2023 1:38:59 PM",V (12697) esp-modem: pattern-detection: 0x3ffb7c94   0d 0a 4f 4b 0d 0a                                 |..OK..|[0m
"5/9/2023 1:38:59 PM",D (12707) esp-modem: Handling line: >>OK

"5/9/2023 1:38:59 PM",<<[0m
"5/9/2023 1:38:59 PM",[0;32mI (12707) dce_service: esp_modem_dce_handle_response_default OK
"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",D (12717) dce_service: define pdp context ok[0m
"5/9/2023 1:38:59 PM",D (12717) esp-modem: Sending command:AT+CGMM
"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",V (12767) esp-modem: uart_get_buffered_data_len()=25[0m
"5/9/2023 1:38:59 PM",V (12767) esp-modem: pattern-detection: 0x3ffb7c94   0d 0a 53 49 4d 43 4f 4d  5f 53 49 4d 37 30 30 30  |..SIMCOM_SIM7000|[0m
"5/9/2023 1:38:59 PM",V (12767) esp-modem: pattern-detection: 0x3ffb7ca4   47 0d 0a 0d 0a 4f 4b 0d  0a                       |G....OK..|[0m
"5/9/2023 1:38:59 PM",D (12777) esp-modem: Handling line: >>SIMCOM_SIM7000G

"5/9/2023 1:38:59 PM",<<[0m
"5/9/2023 1:38:59 PM",D (12787) esp-modem: Handling line: >>OK

"5/9/2023 1:38:59 PM",<<[0m
"5/9/2023 1:38:59 PM",D (12787) dce_service: get module name ok[0m
"5/9/2023 1:38:59 PM",D (12797) esp-modem: Sending command:AT+CGSN
"5/9/2023 1:38:59 PM",[0m
"5/9/2023 1:38:59 PM",V (12837) esp-modem: uart_get_buffered_data_len()=25[0m
"5/9/2023 1:38:59 PM",V (12837) esp-modem: pattern-detection: 0x3ffb7c94   0d 0a 38 36 39 39 35 31  30 33 36 38 35 30 31 36  |..86995103685016|[0m
"5/9/2023 1:38:59 PM",V (12837) esp-modem: pattern-detection: 0x3ffb7ca4   36 0d 0a 0d 0a 4f 4b 0d  0a                       |6....OK..|[0m
"5/9/2023 1:38:59 PM",D (12857) esp-modem: Handling line: >>869951036850166
[... 629 lines cut ...]
"5/9/2023 1:40:09 PM",D (83087) event: running post NETIF_PPP_STATUS:268 with handler 0x400e5ba4 and context 0x3ffc7c9c on loop 0x3ffb69c0[0m
"5/9/2023 1:40:09 PM",[0;32mI (83097) esp-netif_lwip-ppp: User interrupt[0m
"5/9/2023 1:40:09 PM",D (83107) event: running post NETIF_PPP_STATUS:256 with handler 0x400da268 and context 0x3ffb79e0 on loop 0x3ffb69c0[0m
"5/9/2023 1:40:09 PM",[0;32mI (83117) pppos_example: PPP state changed event 256[0m
"5/9/2023 1:40:09 PM",D (83127) event: running post NETIF_PPP_STATUS:256 with handler 0x400e5ba4 and context 0x3ffc7c9c on loop 0x3ffb69c0[0m
"5/9/2023 1:40:09 PM",D (83137) event: running post NETIF_PPP_STATUS:5 with handler 0x400da268 and context 0x3ffb79e0 on loop 0x3ffb69c0[0m
"5/9/2023 1:40:09 PM",[0;32mI (83147) pppos_example: PPP state changed event 5[0m
"5/9/2023 1:40:09 PM",[0;32mI (83157) pppos_example: User interrupted event from netif:0x3ffc64ec[0m
"5/9/2023 1:40:09 PM",D (83157) event: running post NETIF_PPP_STATUS:5 with handler 0x400e5ba4 and context 0x3ffc7c9c on loop 0x3ffb69c0[0m
"5/9/2023 1:40:09 PM",[0;32mI (83167) esp-modem-netif: PPP state changed event 5[0m
"5/9/2023 1:40:09 PM",[0;32mI (83177) pppos_example: modem_power entering..[0m
"5/9/2023 1:40:09 PM",[0;32mI (83187) pppos_example: modem_power turning off..[0m
[... cut ...]
"5/9/2023 1:40:17 PM",V (91107) esp_netif_objects: esp_netif_remove_from_list 0x3ffc64ec[0m
"5/9/2023 1:40:17 PM",D (91107) esp_netif_objects: esp_netif_remove_from_list netif successfully removed (total netifs: 0)[0m
"5/9/2023 1:40:17 PM",D (91107) pppos_example: netif destroyed[0m
"5/9/2023 1:40:17 PM",D (91107) event: deleted loop 0x3ffc54cc[0m
"5/9/2023 1:40:17 PM",V (91117) intr_alloc: esp_intr_free: Disabling int, killing handler[0m
"5/9/2023 1:40:17 PM",[0;32mI (91117) pppos_example: All deleted[0m

AdiXPS, 2023-09-07:
Would really get your support on this @david-cermak thanks in advance

david-cermak, 2023-09-07:
@AdiXPS Are you sure that you're not terminating the connection from your side? From your log:
```
"5/9/2023 1:40:09 PM",�[0;32mI (83097) esp-netif_lwip-ppp: User interrupt�[0m
```
this tells me that the ESP32 has requested exiting the data mode by sending this "User interrupt" to the PPP session.

Note, that if you're using C++ API and your local object gets out of scope, the destructors will be called, also closing any open connection.

david-cermak, 2023-09-07:
> On the modem side its sending OK then it exited the data mode,

Does it mean that your modem just sends `OK` abruptly, and exits the data mode? I'd just make sure that nothing comes from the ESP32 side before that (using a logic analyzer or just snooping on the Tx line with a UART-USB converter) to rule out potential bug in this library.
Otherwise, I could only suggest enabling the `LCP keepalive`, so the PPP mode on the ESP side would also exit (sooner), and you can retry...

AdiXPS, 2023-09-07:
> Does it mean that your modem just sends OK abruptly, and exits the data mode? 

Exactly.

>  I'd just make sure that nothing comes from the ESP32 side before that (using a logic analyzer or just snooping on the Tx line with a UART-USB converter) to rule out potential bug in this library.

Absolutely I've did that. I cannot see +++ from ESP, seems that its when the negotiation part take place after getting the IP the modem send OK then its exits the mode. But the ESP is not aware about it. Also there are two CONNECT happening sometimes? Not sure what is happening, this is from the raw data of the modem:

"5/9/2023 11:12:28 AM",**CONNECT 150000000**
"5/9/2023 11:12:28 AM",~ }#└!}!} } }9}"}&} } } } }#}%┬#}%}%}&╙]}8╟}'}"}(}"▓5~~ }#└!}"}!} }4}"}&} } } } }%}&ДКмС}'}"}(}"╒N~~ }#└!}!}!} }8}"}&} } } } }#}$└#}%}&╙]}8╟}'}"}(}"а} ~~ }#└!}+}"} }(╙]}8╟д}:└#�� � ¤0А!�� �Б�
"5/9/2023 11:12:30 AM",�
"5/9/2023 11:12:30 AM",Г�
"5/9/2023 11:12:30 AM",��В�
"5/9/2023 11:12:30 AM",�
"5/9/2023 11:12:30 AM",Д�
"5/9/2023 11:12:30 AM",��A�А!� �g├А!��
"5/9/2023 11:12:30 AM",�� -�� цА!�� �╗ЩА!�� ���di�АБ�Tы�7Г�Tы9цn�~~А!�� ���di�АБ�Tы�7Г�Tы9ц73~
"5/9/2023 11:12:31 AM",OK
"5/9/2023 11:12:31 AM",!E , @ -��▌C▀u'di�А�XєЁЮ╧ZГ╟tЙT`�·Ё�5 ���P=│
"5/9/2023 11:12:32 AM",SIM7000G R1529

"5/9/2023 11:12:32 AM",**CONNECT 150000000**
"5/9/2023 11:12:32 AM",~!E , @ -��▌C▀u'di�А�XєЁЮ╧ZГ╟tЙT�·Ё�5  ���P=│~~!E  ,  @ -��▌C▀u'di�А�XєЁЮ╧ZГ╟tЙT�·Ё�5 ���P=│~~!E (@�@ -�╘═C▀u'di�А�XєЁЮ╧ZД╟tМбP�ўг6О э_!E (@�@ -�╘╠C▀u'di�А�XєЁЮ╧ZД╟tП|P�ўг3│ n6!E �Ы@�@ -�╤XC▀u'di�А�XєЁЮ╧ZД╟tП|P�ўг═; HTTP/1.1 200 OK
"5/9/2023 11:12:36 AM",Access-Control-Allow-Origin: *
"5/9/2023 11:12:36 AM",X-Powered-By: Express`**

david-cermak, 2023-09-11:
@AdiXPS I'm afraid I cannot help much here. I could only recommend to check the power supply, and connection. The modem draws significantly higher current when connecting, so it might go though some internal resets or disconnections? Perhaps something like that is happening, if you're seeing the `CONNECT` message multiple times?

Anyway, it seems to be a problem of your device, not the library. I don't see such behaviour on SIM7000 modems. Do you even use  the `esp_modem` library, btw? From the logs I saw it appears like you're still using the `pppos_example` from IDF <= v4.4.

david-cermak, 2023-09-12:
Thanks for the information! Didn't really expect the DTR signal would be switching modes (though I knew some of the flow control signals were used for example to switch between low power modes). I'll describe this in the troubleshooting section of the docs.

> I think as part of enhancement, OK command should be accounted for in data mode as early detection

Good point, but that's not so easy to implement, as we should check for the `OK` reply only if the packet is dropped in lwip, which is happening in lower layers, and searching for the pattern in data path would impact performance.

> AT&D command should be defaulted to AT&D0 in the library.

This is a specific feature of your modem and doesn't belong to the library. Might be added to some post-init hooks implemented by users? But in real world application, users typically reconfigure their devices permanently, so they wouldn't need to reinit all the config options in startup code.

AdiXPS, 2023-09-16:
Thanks. it was just a suggestion. I'll close it.

closed 2023-09-16 by AdiXPS