espressif/esp-protocols #356: mqtt_cxx - Client::on_subscribed has empty topic

tajnymag, 2023-09-13:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

While using esp_mqtt(_cxx) component, I was surprised by lines of `Subscribed to` in the monitor.

`idf::mqtt:Client` has a [hardcoded `printf` call in its `on_subcribed`](https://github.com/espressif/esp-protocols/blob/13b29c22913c97d7e50b3b74956e284f83460c16/components/esp_mqtt_cxx/esp_mqtt_cxx.cpp#L218C70-L218C70) method. This method, however, seems to get an empty string for its given `event->topic`.

In `Client::on_data`, event has `topic` loaded properly and I can print it just fine with the exact same printf statement.

euripedesrocha, 2023-09-19:
Hi @tajnymag thanks for reporting. 
That printf is meaningless since the event dispatched from the mqtt client doesn't have the information. I created #358 to remove it. 

You can check about the events in [IDF documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html#events)

Regarding the information in the DATA event, this is part of your code?  I Searched in trhe examples and we don't have this.

tajnymag, 2023-09-19:
> Hi @tajnymag thanks for reporting.
> That printf is meaningless since the event dispatched from the mqtt client doesn't have the information. I created #358 to remove it.
>
> You can check about the [...]

Thank you for removing the line. 

No, it's part of `idf::mqtt::Client` ([](https://github.com/espressif/esp-protocols/blob/master/components/esp_mqtt_cxx/esp_mqtt_cxx.cpp#L232)). I had played with [this example](https://github.com/espressif/esp-protocols/blob/master/components/esp_mqtt_cxx/examples/tcp/main/mqtt_tcp_example.cpp) where an override for `on_data` is used to check received topic.

closed 2023-09-21 by euripedesrocha