espressif/esp-protocols #375: testing new Modem modem-v1.0.3 on a SIMCOM A7672E

diplfranzhoepfinger, 2023-10-12:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

testing new Modem modem-v1.0.3 on a SIMCOM A7672E

david-cermak, 2023-10-13:
@diplfranzhoepfinger Does it mean that it works with the inflatable buffer? Looking at the logs, your'e hitting only the exit CMUX problem, which is related to the known issue of SIMCOM A7672

I think you're actually seeing this problem:

https://github.com/espressif/esp-protocols/blob/93dd56750e8db96f7ce71efe2ebd025fe6f4f855/components/esp_modem/src/esp_modem_dte.cpp#L81-L82

I was thinking about adding a debug/warning log here, but this could also mean a standard timeout, so decided against it.

I know that I told you to set `ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD` to `false` in the past, because your device uses a longer CMUX payload and esp_modem wasn't ready to handle that.
It is also documented:

https://github.com/espressif/esp-protocols/blob/93dd56750e8db96f7ce71efe2ebd025fe6f4f855/components/esp_modem/Kconfig#L15-L17

that this mode is perfectly suited for DATA mode, but in the COMMAND mode, you might see get some timeout issues and this is probably the case. And it also explains why the logging verbosity makes such a difference, clearly sending long hexdump payloads to UART causes delays, so the data come in chunks and if the `OK` delimiter is not present in the first chunk we have no other option (in this configuration) than to give up and report failure.

Now the `esp_modem` was fixed to properly work in the default mode, as well. You can give it a try. (the data mode might be a bit slower, since we keep and copy buffers between CMUX and DTE layers). This config also handles corner cases, especially with your device `A7672`, so It would give a warning if the defragmenting bufffer size is not big enough:

https://github.com/espressif/esp-protocols/blob/93dd56750e8db96f7ce71efe2ebd025fe6f4f855/components/esp_modem/src/esp_modem_cmux.cpp#L315-L320

---

That said, I would recommend these configuration of `esp_modem` to work with A7672:
* `ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD=y`, `ESP_MODEM_USE_INFLATABLE_BUFFER_IF_NEEDED=n` -- default, optimized for "embedded" usecase, static buffer on CMUX layer, may need bigger buffer size, slower networking
* `ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD=n`, `ESP_MODEM_USE_INFLATABLE_BUFFER_IF_NEEDED=y` -- optimized for DATA throughput, buffer size could be small, but an allocation would take place if you get a long (or fragmented) command reply (which you do if you're using GNSS commands)

diplfranzhoepfinger, 2023-10-13:
ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD=n, ESP_MODEM_USE_INFLATABLE_BUFFER_IF_NEEDED=y

Set like you recommend 

It does now at least get a IP Address; which was a big issue before.
But how can we go further?
We want not only get a IP but also get a Connection which seems not to be that case now. 

Where does this "invalid header" warning come from?

david-cermak, 2023-10-13:
> But how can we go further?

This example terminates the connection in the end, so you either remove the exit or reconnect after it. Looking at your logs I can see that the connection and publishing to the MQTT broker succeeded -- in both verbose/info version.

> Where does this "invalid header" warning come from?

I think we've discussed that already. Your modem doesn't comply to some control sequences of the CMUX protocol, I've even added a note about it in the docs:

https://github.com/espressif/esp-protocols/blob/93dd56750e8db96f7ce71efe2ebd025fe6f4f855/docs/esp_modem/en/README.rst?plain=1#L148-L149

Previous version of `esp_modem` simply ignored it (the last command at the end didn't work), but I've recently added 8edbac697478bd7abef1b2f69e7e5c24b48e0369 to strictly check all violations, not only to emit warnings, but mainly to be able to recover the protocol. This proved very useful indeed, especially on UART if we miss an interrupt or buffer overflows. Very much applicable to devices like A7672, which define two byte payloads, so if such a glitch happen and is interpreted as a very long payload (like 32kB) we'd keep reading on and on, looking for the end of the frame (and CRC) which never comes.

diplfranzhoepfinger, 2023-10-14:
so i will integrate your patch and make new Tests. 

i do not think that anything runs in the current Version. The Device get an IP, but furthere communication does not happen. 

i will do more tests later today, and post them here. 

Thanks, 

Franz

diplfranzhoepfinger, 2023-10-17:
|                                    |                                                                              |  
|------------------------------------|------------------------------------------------------------------------------|  
| CONFIG\_LOG\_DEFAULT\_LEVEL\_INFO=y    | \[log\]()|  
| CONFIG\_LOG\_DEFAULT\_LEVEL\_VERBOSE=y | \[log\]()                                                                             |

david-cermak, 2023-10-17:
> seems also working, only Problem is the Fragmentation of the AT AT+CPSI? Answer.

It comes fragmented, but should be passed to the command library multiple times, but the new fragments should be appended to the previous so in the end, it should arrive as a whole. This is also happening for other commands, that's why we parse them like this:

https://github.com/espressif/esp-protocols/blob/3e8de3af3a8bda2c1b0b1eb3e20c277fe123c40a/components/esp_modem/src/esp_modem_command_library.cpp#L335-L342

(returning `command_result::TIMEOUT` to indicate that we need to read more, but after the next fragment arrives the buffer contains all accumulated data so far)

> can i increase the Default Buffer, so that is works in most cases without a Fragmentation ?

Yes, as indicated above: https://github.com/espressif/esp-protocols/issues/375#issuecomment-1760953943 you can use the most default configuration (`ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD=y`, `ESP_MODEM_USE_INFLATABLE_BUFFER_IF_NEEDED=n`) to have the CMUX layer defragment the data. You can start with a smaller buffer and the library would tell you if you have to increase it.

diplfranzhoepfinger, 2023-10-18:
> 
> (returning `command_result::TIMEOUT` to indicate that we need to read more, but after the next fragment arrives the buffer contains all accumulated data so far)

so, how to get the Data then ? can i grab it somehow ? 

> 
> > can i increase the Default Buffer, so that is works in most cases without a Fragmentation ?
> 
> Yes, as indicated above: [#375 (comment)](https://github.com/espressif/esp-protocols/issues/375#issuecomment-1760953943) you can use the most default configuration (`ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD=y`, `ESP_MODEM_USE_INFLATABLE_BUFFER_IF_NEEDED=n`) to have the CMUX layer defragment the data. You can start with a smaller buffer and the library would tell you if you have to increase it.

so you say maybe this config is the better one in some cases ? 
Data Througput is less then you say ? Even with a increased Buffer ?

diplfranzhoepfinger, 2023-10-20:
Now I understand. We must change the Parser.

franz-ms-muc, 2023-10-20:
thanks a lot, i will do a test.

david-cermak, 2023-10-20:
@diplfranzhoepfinger You can also try to cherry-pick https://github.com/espressif/esp-protocols/pull/391/commits/1db1e1508d63c9d03161e3b7785f6cc7a2ac81b8 to see if your parser works with `ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD=n`, `ESP_MODEM_USE_INFLATABLE_BUFFER_IF_NEEDED=n`.
Context: There's a regression in `v1.0.3` with handling USB DTE data and this fix actually mimics the previous behavior for that config.

david-cermak, 2023-10-20:
> but we use A7672 on 4-wire serial. 

This also affects UART in the configuration of `ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD`=n, `ESP_MODEM_USE_INFLATABLE_BUFFER_IF_NEEDED=n`

> do you think USB has some siginficant advantages ? OK, it saves 2 Lines. 

Wires and speed

> but i still need CMUX right ?

Not necessarily, A7670 supports two logical channels (endpoints) in USB, which could represent two physical channels (i.e. terminals) that you connect to the DTE. So you can have one channel for commands and another for data without multiplexing.

franz-ms-muc, 2023-10-20:
> > but i still need CMUX right ?
> 
> Not necessarily, A7670 supports two logical channels (endpoints) in USB, which could represent two physical channels (i.e. terminals) that you connect to the DTE. So you can have one channel for commands and another for data without multiplexing.

then i start ESP-Modem 2 times, once in DATA Mode, once in AT Mode ?

david-cermak, 2023-10-20:
Just check the examples.
This would create the corresponding DTE with these two terminals already included:

https://github.com/espressif/esp-protocols/blob/3e8de3af3a8bda2c1b0b1eb3e20c277fe123c40a/components/esp_modem/examples/pppos_client/main/pppos_client_main.c#L220-L222

The rest is the exactly same (one DTE, one DCE, one network interface).

---
PS: This constructs a DTE which simply takes two terminals, and sets the mode to `DUAL_MODE` so no further switching is necessary.

https://github.com/espressif/esp-protocols/blob/3e8de3af3a8bda2c1b0b1eb3e20c277fe123c40a/components/esp_modem/src/esp_modem_dte.cpp#L33-L36

franz-ms-muc, 2023-10-20:
that is cool 

i want to test it !

david-cermak, 2023-11-28:
It looks like the tests pass. Please reopen if not.

closed 2023-11-28 by david-cermak

diplfranzhoepfinger, 2023-11-28:
YES ! 

we use it, and beside some initial Problems with the VBUS Pin ( USB still work when VBUS not on 5V, but Fail after this when activating GNSS) 

it is much more cool than using CMUX. 

so if somebody read this: 

i would rate USB much better than CMUX.

david-cermak, 2023-11-28:
I'm planning on enabling `Discussions` in this repo, so people can share what they think about different components and modes.
It's probably a good place for you to post a feedback on CMUX vs USB

diplfranzhoepfinger, 2023-11-28:
yep. might be a good thing !