espressif/esp-protocols #384: Build issues with esp_modem on Mac M1 Pro

larsakeekstrand, 2023-10-15:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### What component are you using? If you choose Other, provide details in More Information.

esp_modem

### component version

modem-v1.0.3

### IDF version.

4.4.6

### More Information.

We are using esp_modem in our project and it builds fine on macOs Ventura x86 but not on macOS Ventura M1 Pro. It seems to be a C++ version issue. I have tried to force version by adding "set(CMAKE_CXX_STANDARD 17)" to the CMakeLists.txt in our project but it doesn't affect anything.

Any ideas?

Here is the first error given in the build-log:

FAILED: esp-idf/esp_modem/CMakeFiles/__idf_esp_modem.dir/src/esp_modem_uart.cpp.obj 
/Users/laek/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++  -I/Users/laek/source/easyserv/sg-sga/build/config -I/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/include -I/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/private_include -I/Users/laek/esp/esp-idf-4.4.6/components/newlib/platform_include -I/Users/laek/esp/esp-idf-4.4.6/components/freertos/include -I/Users/laek/esp/esp-idf-4.4.6/components/freertos/include/esp_additions/freertos -I/Users/laek/esp/esp-idf-4.4.6/components/freertos/port/xtensa/include -I/Users/laek/esp/esp-idf-4.4.6/components/freertos/include/esp_additions -I/Users/laek/esp/esp-idf-4.4.6/components/esp_hw_support/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_hw_support/include/soc -I/Users/laek/esp/esp-idf-4.4.6/components/esp_hw_support/include/soc/esp32s3 -I/Users/laek/esp/esp-idf-4.4.6/components/esp_hw_support/port/esp32s3/. -I/Users/laek/esp/esp-idf-4.4.6/components/esp_hw_support/port/esp32s3/private_include -I/Users/laek/esp/esp-idf-4.4.6/components/heap/include -I/Users/laek/esp/esp-idf-4.4.6/components/log/include -I/Users/laek/esp/esp-idf-4.4.6/components/lwip/include/apps -I/Users/laek/esp/esp-idf-4.4.6/components/lwip/include/apps/sntp -I/Users/laek/esp/esp-idf-4.4.6/components/lwip/lwip/src/include -I/Users/laek/esp/esp-idf-4.4.6/components/lwip/port/esp32/include -I/Users/laek/esp/esp-idf-4.4.6/components/lwip/port/esp32/include/arch -I/Users/laek/esp/esp-idf-4.4.6/components/soc/include -I/Users/laek/esp/esp-idf-4.4.6/components/soc/esp32s3/. -I/Users/laek/esp/esp-idf-4.4.6/components/soc/esp32s3/include -I/Users/laek/esp/esp-idf-4.4.6/components/hal/esp32s3/include -I/Users/laek/esp/esp-idf-4.4.6/components/hal/include -I/Users/laek/esp/esp-idf-4.4.6/components/hal/platform_port/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_rom/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_rom/include/esp32s3 -I/Users/laek/esp/esp-idf-4.4.6/components/esp_rom/esp32s3 -I/Users/laek/esp/esp-idf-4.4.6/components/esp_common/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_system/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_system/port/soc -I/Users/laek/esp/esp-idf-4.4.6/components/esp_system/port/public_compat -I/Users/laek/esp/esp-idf-4.4.6/components/xtensa/include -I/Users/laek/esp/esp-idf-4.4.6/components/xtensa/esp32s3/include -I/Users/laek/esp/esp-idf-4.4.6/components/driver/include -I/Users/laek/esp/esp-idf-4.4.6/components/driver/esp32s3/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_pm/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_ringbuf/include -I/Users/laek/esp/esp-idf-4.4.6/components/efuse/include -I/Users/laek/esp/esp-idf-4.4.6/components/efuse/esp32s3/include -I/Users/laek/esp/esp-idf-4.4.6/components/vfs/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_wifi/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_event/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_netif/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_eth/include -I/Users/laek/esp/esp-idf-4.4.6/components/tcpip_adapter/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_phy/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_phy/esp32s3/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_ipc/include -I/Users/laek/esp/esp-idf-4.4.6/components/app_trace/include -I/Users/laek/esp/esp-idf-4.4.6/components/esp_timer/include -mlongcalls  -std=gnu++17 -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fstack-protector -fmacro-prefix-map=/Users/laek/source/easyserv/sg-sga=. -fmacro-prefix-map=/Users/laek/esp/esp-idf-4.4.6=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.4.6\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/esp_modem/CMakeFiles/__idf_esp_modem.dir/src/esp_modem_uart.cpp.obj -MF esp-idf/esp_modem/CMakeFiles/__idf_esp_modem.dir/src/esp_modem_uart.cpp.obj.d -o esp-idf/esp_modem/CMakeFiles/__idf_esp_modem.dir/src/esp_modem_uart.cpp.obj -c /Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/src/esp_modem_uart.cpp
In file included from /Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/src/esp_modem_uart.cpp:14:
/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/src/esp_modem_uart.cpp: In function 'std::unique_ptr<esp_modem::Terminal> esp_modem::create_uart_terminal(const esp_modem_dte_config*)':
/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/src/esp_modem_uart.cpp:105:26: error: 'make_unique' is not a member of 'std'
         auto term = std::make_unique<UartTerminal>(config);
                          ^~~~~~~~~~~
/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/private_include/exception_stub.hpp:29:5: note: in definition of macro 'TRY_CATCH_RET_NULL'
     block
     ^~~~~
/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/src/esp_modem_uart.cpp:105:26: note: 'std::make_unique' is only available from C++14 onwards
         auto term = std::make_unique<UartTerminal>(config);
                          ^~~~~~~~~~~
/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/private_include/exception_stub.hpp:29:5: note: in definition of macro 'TRY_CATCH_RET_NULL'
     block
     ^~~~~
/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/src/esp_modem_uart.cpp:105:50: error: expected primary-expression before '>' token
         auto term = std::make_unique<UartTerminal>(config);
                                                  ^
/Users/laek/source/easyserv/sg-sga/components/esp-protocols/components/esp_modem/private_include/exception_stub.hpp:29:5: note: in definition of macro 'TRY_CATCH_RET_NULL'
     block
     ^~~~~

david-cermak, 2023-10-16:
Hi @larsakeekstrand 

How do you build your project and how do you include `esp_modem` to it? Do you use it as  a standard component (from component manager or this repo)? Or you just somehow include/add sources to your project?

Yes, this issue is related to C++ version, `make_unique<>` is available only from C++14, but from the command line above, I can see that you're building it with C++ 11 (`-std=gnu++11`).
IDF build system uses component style of building and each component could be build under defined standard. Here, for the `esp_modem` we set the standard to C++17, but you can still use C++11 if you consume it (as headers are C++11 compatible).

This file: `esp_modem_uart.cpp` belongs to the internals of the project, so it should be built under C++ 17, unless it's consumed in some other way to your project...
If you're including it "manually" you can try to set the C++ standard in a similar way as it's done here:

https://github.com/espressif/esp-protocols/blob/f591b7f4c142466ccaf7a36f78d25ddf89330e99/components/esp_modem/CMakeLists.txt#L39-L43

larsakeekstrand, 2023-10-16:
Hi @david-cermak 

and thanks for your suggestions. I tried to force CXX_STANDARD 17 as in your example above but it makes no difference. 
To rule out things specific to our project and how the component is included, I can also reproduce the issue using the esp_modem examples:

# idf.py create-project-from-example "espressif/esp_modem^1.0.3:modem_console"
# idf.py set-target esp32s3
# idf.py menuconfig
# idf.py build

... 
In file included from /Users/laek/source/easyserv/modem_console/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp:14:
/Users/laek/source/easyserv/modem_console/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp: In function 'std::unique_ptr<esp_modem::Terminal> esp_modem::create_uart_terminal(const esp_modem_dte_config*)':
/Users/laek/source/easyserv/modem_console/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp:103:26: error: 'make_unique' is not a member of 'std'
         auto term = std::make_unique<UartTerminal>(config);
                          ^~~~~~~~~~~
/Users/laek/source/easyserv/modem_console/managed_components/espressif__esp_modem/private_include/exception_stub.hpp:29:5: note: in definition of macro 'TRY_CATCH_RET_NULL'
     block
     ^~~~~
/Users/laek/source/easyserv/modem_console/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp:103:26: note: 'std::make_unique' is only available from C++14 onwards
         auto term = std::make_unique<UartTerminal>(config);
                          ^~~~~~~~~~~
/Users/laek/source/easyserv/modem_console/managed_components/espressif__esp_modem/private_include/exception_stub.hpp:29:5: note: in definition of macro 'TRY_CATCH_RET_NULL'
     block
     ^~~~~
/Users/laek/source/easyserv/modem_console/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp:103:50: error: expected primary-expression before '>' token
         auto term = std::make_unique<UartTerminal>(config);
                                                  ^
/Users/laek/source/easyserv/modem_console/managed_components/espressif__esp_modem/private_include/exception_stub.hpp:29:5: note: in definition of macro 'TRY_CATCH_RET_NULL'
     block
     ^~~~~

david-cermak, 2023-10-16:
I've just tried to run these commands on my Mac (not a M1 Pro, but that shouldn't be the issue) and it has built correctly:
```
idf.py create-project-from-example "espressif/esp_modem^1.0.3:modem_console"
cd modem_console
idf.py set-target esp32s3
idf.py build
```

So the only missing part is the configuration. Could you please share what you change in `idf.py menuconfig` or just post your `sdkconfig` ?

larsakeekstrand, 2023-10-16:
To test if there is a difference between esp-idf versions, I downloaded esp-idf v5.1.1 and ran the same tests as above and it works fine. So, the issue seems to be related to v4.4.x of esp-idf.
Unfortunately we cannot switch from the v4.4.x series to v5.x.x easily as this seems to contain breaking changes.

david-cermak, 2023-10-16:
Are you sure that you're building the freshly created project using the STR that you shared in https://github.com/espressif/esp-protocols/issues/384#issuecomment-1764337131 ?

I didn't see this `cd modem_console` in your commands. I don't think any of our example won't build on `v4.4`, since we run the build step in the CI, so the `modem_console` for example is being checked agains IDF down to `v4.2`:

https://github.com/espressif/esp-protocols/blob/3e8de3af3a8bda2c1b0b1eb3e20c277fe123c40a/.github/workflows/modem__build-host-tests.yml#L16-L17

Moreover, I don't think the M1 Pro machine would be the issue, since we clearly see the `-std=gnu++11` gets into your command line.

larsakeekstrand, 2023-10-16:
Yes, I'm building from fresh directories. 

To be 100% sure I did the same thing again with both v4.4.6 and v5.1.1:

**Using esp-idf v.4.4.6**
```
% idf.py --version
ESP-IDF v4.4.6
% ls -l
total 0
% idf.py create-project-from-example "espressif/esp_modem^1.0.3:modem_console"
% cd modem_console
% idf.py set-target esp32s3
% idf.py menuconfig
% idf.py build

...Failure (same as above) 
In file included from /Users/laek/esp/source/modem_console/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp:14:
/Users/laek/esp/source/modem_console/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp: In function 'std::unique_ptr<esp_modem::Terminal> esp_modem::create_uart_terminal(const esp_modem_dte_config*)':
/Users/laek/esp/source/modem_console/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp:103:26: error: 'make_unique' is not a member of 'std'
         auto term = std::make_unique<UartTerminal>(config);
```
**Using esp-idf v5.1.1**
```
% idf.py --version
ESP-IDF v5.1.1
% ls -l
total 0
% idf.py create-project-from-example "espressif/esp_modem^1.0.3:modem_console"
% cd modem_console
% idf.py set-target esp32s3
% idf.py menuconfig
% idf.py build

Success!
```

david-cermak, 2023-10-17:
Thanks for the report and sharing all the details. It really looks like there's a problem somewhere.

From the first log you shared, I can see two places where we define c++ standard: `-std=gnu++17` and `-std=gnu++11`.
I can actually reproduce the same on my linux machine, the only difference is that C++11 comes first.
So maybe the expansion takes place differently on Mac M1 Pro and gcc uses the last supplied flag. I'll try to get hold of some M1 machine to check if it's really the case. But the problem is already the fact that different standard appears twice on the same command line.

larsakeekstrand, 2023-10-17:
I added the fix into the CMakeLists.txt file and it seems to do the trick so everything builds fine now. Thanks @david-cermak for your fast help!

david-cermak, 2023-11-28:
Closing as resolved, feel free to reopen if any other issues.

ankit-bansal-oxit, 2024-07-12:
@david-cermak Exactly same thing happens with me on kubuntu 24.04. I need to use the `idf_build_set_property(CXX_COMPILE_OPTIONS "--std=gnu++17" APPEND)` to mitigate this