espressif/esp-protocols #404: testing new Modem modem-v1.0.3 on a SIMCOM A7672E with USB

diplfranzhoepfinger, 2023-10-25:
Just check the examples.
This would create the corresponding DTE with these two terminals already included:

https://github.com/espressif/esp-protocols/blob/3e8de3af3a8bda2c1b0b1eb3e20c277fe123c40a/components/esp_modem/examples/pppos_client/main/pppos_client_main.c#L220-L222

The rest is the exactly same (one DTE, one DCE, one network interface).

---
PS: This constructs a DTE which simply takes two terminals, and sets the mode to `DUAL_MODE` so no further switching is necessary.

https://github.com/espressif/esp-protocols/blob/3e8de3af3a8bda2c1b0b1eb3e20c277fe123c40a/components/esp_modem/src/esp_modem_dte.cpp#L33-L36

_Originally posted by @david-cermak in https://github.com/espressif/esp-protocols/issues/375#issuecomment-1773086446_

diplfranzhoepfinger, 2023-10-25:
@david-cermak 

here is my Ouput of 

  
[https://gist.github.com/diplfranzhoepfinger/8085a54f2320639e93eab6c43a846061](https://gist.github.com/diplfranzhoepfinger/8085a54f2320639e93eab6c43a846061)  
running this Code

[https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client/tree/main/usb_host_lib](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client/tree/main/usb_host_lib)

david-cermak, 2023-10-26:
@diplfranzhoepfinger Did you include the fix I mentioned in https://github.com/espressif/esp-protocols/issues/375#issuecomment-1772525793 ?

Just merged it yesterday; and added a new CI runner, which runs this example (S2+A7670E) https://github.com/espressif/esp-protocols/actions/runs/6628377154/job/18006092226#step:5:161 with this config:

https://github.com/espressif/esp-protocols/blob/2e6732882d813f7830fdc0604da49221b91a134d/components/esp_modem/examples/pppos_client/sdkconfig.ci.usb_a7670_s2#L3-L14

franz-ms-muc, 2023-10-26:
this looks cool, i will try.

diplfranzhoepfinger, 2023-10-26:
With your mentioned patch it works perfect in INFO Logging Mode: 

Gist with Logs: [https://gist.github.com/diplfranzhoepfinger/44aa433829e36bc7266f11f1e7fc5ab2](https://gist.github.com/diplfranzhoepfinger/44aa433829e36bc7266f11f1e7fc5ab2)

Repo with my Code: [https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client/tree/v0.2](https://github.com/diplfranzhoepfinger/esp-protocols-tree-master-components-esp_modem-examples-pppos_client/tree/v0.2)

david-cermak, 2023-11-28:
Good to hear that it works, closing then

closed 2023-11-28 by david-cermak