espressif/esp-protocols #406: esp_modem regression from 1.0.1 to 1.0.3: fails to acknowledge basic AT commands via USB

DaStoned, 2023-10-26:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm issuing basic AT queries to a SimCom SIM7070G module connected via USB - get module name, firmware revision, battery level etc - using the C++ API. They were working fine with esp_modem version 1.0.1 and started to fail rather spectacularly in 1.0.3.

```
// Run a modem query with a few retries
// Adapted from https://stackoverflow.com/a/76233658/9989141
template <typename Callable, typename ...ArgsT>
constexpr decltype(auto) retryQuery(size_t maxRetries, Callable&& func, ArgsT&&... args) {
    size_t retries = 0;
    auto ret = invoke(forward<Callable>(func), forward<ArgsT>(args)...);
    while (command_result::OK != ret && retries++ < maxRetries) {
        tag_debug("Cmd result: %s (%u) (retry %u/%u)", modem_err_to_name(ret), static_cast<unsigned int>(ret), retries, maxRetries);
        ret = invoke(forward<Callable>(func), forward<ArgsT>(args)...);
    }
    return ret;
}

...

    struct esp_modem_usb_term_config usb_config =
        ESP_MODEM_DEFAULT_USB_CONFIG(GSM_USB_VID, GSM_USB_PID, GSM_USB_IDX_CMD);
    usb_config.timeout_ms = GSM_USB_APPEAR_TIMEOUT_MS;
    esp_modem_dte_config_t dte_usb_config =
        ESP_MODEM_DTE_DEFAULT_USB_CONFIG(usb_config);
    dte_usb_config.dte_buffer_size = 1536; // Must fit PPPoS MTU (1500 B?)

    _uart_dte = create_usb_dte(&dte_usb_config);
    if (!_uart_dte) {
        tag_error("Fail create USB DTE");
        return false;
    }

    _uart_dte->set_error_cb(onUsbTerminalErr);
    _modem = create_SIM7070_dce(&_modem_config, _uart_dte, _esp_netif);
    if (!_modem) {
        tag_error("Fail create SIM7070 modem");
        return false;
    }

    string strOut;
    command_result cmdRes;

    // Module may take time starting from cold boot. Wait until responsive.
    cmdRes = retryQuery(100, [this](string& out)->command_result {return _modem->at("AT+CGMI", out, GSM_CMD_TIMEOUT_MS);}, strOut);
    if (command_result::OK == cmdRes) {
        tag_info("GSM manufacturer: [%s]", strOut.c_str());
    } else {
        tag_error("Fail query GSM manufacturer");
        return false; // Looks like GSM is not responding. Call it a failure.
    }

    cmdRes = retryQuery(10, [this](string& out)->command_result {return _modem->get_module_name(out);}, strOut);
    if (command_result::OK == cmdRes) {
        tag_info("GSM HW model: [%s]", strOut.c_str());
    } else {
        tag_error("Fail query GSM HW model: %s", modem_err_to_name(cmdRes));
    }

```

With 1.0.1 it works OK:

```
[... 85 lines cut ...]
D (22496) usb_terminal: 0x3fcb8244   41 54 2b 43 47 4d 4d 0d                           |AT+CGMM.|
D (22496) usb_terminal: 0x3fcb822c   0d 0a 53 49 4d 43 4f 4d  5f 53 49 4d 37 30 37 30  |..SIMCOM_SIM7070|
D (22506) usb_terminal: 0x3fcb823c   0d 0a 0d 0a 4f 4b 0d 0a                           |....OK..|
D (22506) gsm_modem: Cmd result: ERROR (1) (retry 7/10)
D (23036) command_lib: generic_get_string
D (23036) command_lib: generic_get_string
D (23036) usb_terminal: 0x3fcc72ac   41 54 2b 43 47 4d 4d 0d                           |AT+CGMM.|
D (23036) usb_terminal: 0x3fcb8244   41 54 2b 43 47 4d 4d 0d                           |AT+CGMM.|
D (23046) usb_terminal: 0x3fcb822c   0d 0a 53 49 4d 43 4f 4d  5f 53 49 4d 37 30 37 30  |..SIMCOM_SIM7070|
D (23056) usb_terminal: 0x3fcb823c   0d 0a 0d 0a 4f 4b 0d 0a                           |....OK..|
} (23066) command_lib: Token: {

D (23066) command_lib: Token: {SIMCOM_SIM7070}

} (23076) command_lib: Token: {

D (23076) command_lib: Token: {OK}

I (23086) gsm_modem: GSM HW model: [SIMCOM_SIM7070]
```

The SimCom SIM7070G is connected via USB. The component configuration for esp_modem is default. I don't use CMUX (it was completely broken over USB, but that's another story altogether). 

My debugger indicates that we never get around to parsing the modem response with `generic_get_string`. Instead this line activates, indicating a buffer over/underflow issue. In the same time I've configured the DTE with a buffer of 1536 bytes, so there's something else going on here.
https://github.com/espressif/esp-protocols/blob/2e6732882d813f7830fdc0604da49221b91a134d/components/esp_modem/src/esp_modem_dte.cpp#L114C30-L114C30

david-cermak, 2023-10-26:
Sorry for the regression. It's just been fixed on master in https://github.com/espressif/esp-protocols/commit/1db1e1508d63c9d03161e3b7785f6cc7a2ac81b8

I'll merge few minor fixes and create `v1.0.4` soon (also added a USB runner to our CI to prevent it from happening in future)

DaStoned, 2023-10-26:
Ah, thank you. That's good news!

DaStoned, 2023-11-01:
I tried with 1.0.4 and cannot reproduce the regression. So I confirm the fix. Thank you.

closed 2023-11-01 by DaStoned