espressif/esp-protocols #426: Esp Modem panic_abort on unsolicited data from SIM card

antfarming, 2023-11-13:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using a roaming SIM card that will perform an OTA occasionally. During this process you can expect unsolicited data from the module, due to the sim tool kit. I cannot stop the sim from doing this, or anticipate when it will.

The modem library does not seem to like this, and I am not sure how to dig further into the issue. I have tried to provide as much info as possible below. As you can see, I request the CGREG status, and before it returns a valid response, sim tool kit data is coming through first.

```
[V][esp_modem_command_library.cpp:522] get_network_attachment_cgreg()::(79409):command_lib: get_network_attachment_cgreg
[V][esp_modem_command_library.cpp:56] generic_get_string()::(79409):command_lib: generic_get_string
[V][esp_modem_command_library.cpp:66] operator()()::(79429):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79429):command_lib: Token: {+MSTK: 11, "D02A810301400382028182050035010339020200470A036F74610573757065723C030100353E052108080808"}
[V][esp_modem_command_library.cpp:66] operator()()::(79449):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79459):command_lib: Token: {+CGEV: ME PDN DEACT 9}
[V][esp_modem_command_library.cpp:66] operator()()::(79469):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79469):command_lib: Token: {+MSTK: 11, "D02A810301400382028182050035010339020200470A036F74610573757065723C030100353E052108080808"}
[V][esp_modem_command_library.cpp:66] operator()()::(79489):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79499):command_lib: Token: {+CGEV: ME PDN DEACT 9}
[V][esp_modem_command_library.cpp:66] operator()()::(79509):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79519):command_lib: Token: {+MSTK: 11, "D02A810301400382028182050035010339020200470A036F74610573757065723C030100353E052108080404"}
[V][esp_modem_command_library.cpp:66] operator()()::(79539):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79539):command_lib: Token: {+CGEV: ME PDN DEACT 9}
[V][esp_modem_command_library.cpp:66] operator()()::(79549):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79559):command_lib: Token: {+MSTK: 11, "D02A810301400382028182050035010339020200470A036F74610573757065723C030100353E052108080404"}
[V][esp_modem_command_library.cpp:66] operator()()::(79579):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79589):command_lib: Token: {+CGEV: ME PDN DEACT 9}
[V][esp_modem_command_library.cpp:66] operator()()::(79599):command_lib: Token: {AT+CGREG?}
[V][esp_modem_command_library.cpp:66] operator()()::(79609):command_lib: Token: {+CGREG: 0,0}
[V][esp_modem_command_library.cpp:66] operator()()::(79619):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79629):command_lib: Token: {OK}
[V][esp_modem_command_library.cpp:66] operator()()::(79629):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(79639):command_lib: Token: {+MSTK: 11, "D081A98103011300820281830500060891349609406105F00B819141000437060000048802700000830D000000004432360000000000000109082923018425260381020A98880307000092909862051F3F01EDE2110C00070400001F426000000F0000000000000000000000000000060200050702260208030001010A0EFFFFFFFFFFFFFFFFFFFFFFFFFFFF24098326230515521620F343098326230565289775F00B0800FFFFFFFFFFFFFF"}

abort() was called at PC 0x4018145f on core 0
0x4018145f: __cxxabiv1::__terminate(void (*)()) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48

Backtrace: 0x40081c56:0x3ffd5290 0x4008aacd:0x3ffd52b0 0x4009245e:0x3ffd52d0 0x4018145f:0x3ffd5340 0x40181494:0x3ffd5360 0x4018156f:0x3ffd5380 0x400ef38d:0x3ffd53a0 0x400ef407:0x3ffd5400 0x400ef4b1:0x3ffd5440 0x400ee9c9:0x3ffd54a0 0x400f0c51:0x3ffd54d0 0x400f1c6a:0x3ffd5510 0x400ecb15:0x3ffd5580 0x401c3c82:0x3ffd55a0 0x400e19c7:0x3ffd55d0 0x400e2279:0x3ffd5620 0x4008d571:0x3ffd5f20
0x40081c56: panic_abort at /Users/anthony/esp/esp-idf/components/esp_system/panic.c:452
0x4008aacd: esp_system_abort at /Users/anthony/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x4009245e: abort at /Users/anthony/esp/esp-idf/components/newlib/abort.c:38
0x4018145f: __cxxabiv1::__terminate(void (*)()) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
0x40181494: std::terminate() at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
0x4018156f: __cxa_throw at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:98
0x400ef38d: esp_modem::throw_if_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_exception.hpp:86
0x400ef407: esp_modem::DTE::command_cb::set(std::function<esp_modem::command_result (unsigned char*, unsigned int)>, char) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_dte.hpp:215 (discriminator 2)
0x400ef4b1: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned long, char) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_dte.cpp:139
0x400ee9c9: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned long) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_dte.cpp:149
0x400f0c51: esp_modem::dce_commands::generic_get_string(esp_modem::CommandableIf*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string_view<char, std::char_traits<char> >&, unsigned long) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_command_library.cpp:57
0x400f1c6a: esp_modem::dce_commands::get_network_attachment_cgreg(esp_modem::CommandableIf*, int&) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_command_library.cpp:524
0x400ecb15: esp_modem::GenericModule::get_network_attachment_cgreg(int&) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_modules.cpp:38
0x401c3c82: esp_modem::command_result esp_modem::DCE::get_network_attachment_cgreg<int&>(int&) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_dce.hpp:114
 (inlined by) esp_modem_get_network_attachment_cgreg at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_c_api.cpp:353
0x400e19c7: waitForNetworkAttachment at /Users/anthony/repos/olarm_esp32_communicator_fw/components/NetworkingClient/Cell.c:1016 (discriminator 15)
0x400e2279: taskCellRun at /Users/anthony/repos/olarm_esp32_communicator_fw/components/NetworkingClient/Cell.c:288
0x4008d571: vPortTaskWrapper at /Users/anthony/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 968df8f3779f71c5

Rebooting...
```

This function is from my own CellularLibrary.c file.

```
static esp_err_t waitForNetworkAttachment(esp_modem_dce_t *dce, uint32_t timeoutMs) 
[... 42 lines cut ...]
    if (std::from_chars(out.data() + pos, out.data() + out.size(), state).ec == std::errc::invalid_argument) {
        return command_result::FAIL;
    }

    return command_result::OK;
}
```

These are from my sdkconfig.defaults file:

```
# Override some defaults to enable PPP
CONFIG_PPP_SUPPORT=y
CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_LWIP_PPP_NOTIFY_PHASE_SUPPORT=y
CONFIG_LWIP_PPP_PAP_SUPPORT=y
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=4096
CONFIG_LWIP_PPP_ENABLE_IPV6=n
CONFIG_ESP_MODEM=y

# Enable C++ exceptions and set emergency pool size for exception objects
# From james: Very important this is enabled, as esp_modem can get stuck without it
CONFIG_COMPILER_CXX_EXCEPTIONS=y
CONFIG_COMPILER_CXX_EXCEPTIONS_EMG_POOL_SIZE=1024
```

david-cermak, 2023-11-14:
Hi @antfarming 

Which version of this library are you using?

Did you see the crashes only during OTA or also in other scenarios? Do you send commands from multiple threads?
Would the issue also appear just by calling the `waitForNetworkAttachment()` as you described above?

antfarming, 2023-11-28:
Hi, sorry for delay.

I have a single thread using esp_modem. 
I also saw it for get csq.

I can provide more debug soon as I am now working on it again. However I think I may know what the issue is. 

In the function below, if any unexpected serial data appears before "+CGATT: ", then the way pos is determined will be incorrect. As it would be offset by the amount of unexpected serial data prior to the expected response. For example, PB DONE can sometimes occur if you start talking to the module early on after boot. So in the below scenario pos would actually be starting at letter 'E'.

AT+CGREG?

PB DONE

+CGREG: 0,5

OK

```
command_result get_network_attachment_state(CommandableIf *t, int &state)
{
    ESP_LOGV(TAG, "%s", __func__ );
    std::string_view out;
    auto ret = generic_get_string(t, "AT+CGATT?\r", out);
    if (ret != command_result::OK) {
        return ret;
    }
    constexpr std::string_view pattern = "+CGATT: ";
    constexpr int pos = pattern.size();
    if (out.find(pattern) == std::string::npos) {
        return command_result::FAIL;
    }

    if (std::from_chars(out.data() + pos, out.data() + out.size(), state).ec == std::errc::invalid_argument) {
        return command_result::FAIL;
    }

    return command_result::OK;
}
```

pos seems to be incosistantly determined in different functions. In the below, it looks like it would cater for unexpected data prior to the expected response.

```

command_result get_operator_name(CommandableIf *t, std::string &operator_name, int &act)
{
    ESP_LOGV(TAG, "%s", __func__ );
    std::string_view out;
    auto ret = generic_get_string(t, "AT+COPS?\r", out, 75000);
    if (ret != command_result::OK) {
        return ret;
    }
    auto pos = out.find("+COPS");
    auto property = 0;
    while (pos != std::string::npos) {
        // Looking for: +COPS: <mode>[, <format>[, <oper>[, <act>]]]
        if (property++ == 2) {  // operator name is after second comma (as a 3rd property of COPS string)
            operator_name = out.substr(++pos);
            auto additional_comma = operator_name.find(',');    // check for the optional ACT
            if (additional_comma != std::string::npos && std::from_chars(operator_name.data() + additional_comma + 1, operator_name.data() + operator_name.length(), act).ec != std::errc::invalid_argument) {
                operator_name = operator_name.substr(0, additional_comma);
            }
            // and strip quotes if present
            auto quote1 = operator_name.find('"');
            auto quote2 = operator_name.rfind('"');
            if (quote1 != std::string::npos && quote2 != std::string::npos) {
                operator_name = operator_name.substr(quote1 + 1, quote2 - 1);
            }
            return command_result::OK;
        }
        pos = out.find(',', ++pos);
    }
    return command_result::FAIL;
}
```

Do you think this could be the issue or should I carry on debugging?

david-cermak, 2023-11-28:
> In the function below, if any unexpected serial data appears before "+CGATT: ", then the way pos is determined will be incorrect.

I don't think so, as the `out` variable would contain also those unexpected data, so it would count the num of char from the `PB DONE` URC, which should be correct.

@antfarming Which version of `esp_modem` are you using? Or ff you're working on a forked repo, or made certain changes, could you please post it here?

I've seen the crash you reported on some earlier versions, but it has been fixed and even unit-tested quite intensively injecting an expected response (also played with the test case to inject some garbage...)

https://github.com/espressif/esp-protocols/blob/9758379921efce816bf46da48ab58bfa786520ca/components/esp_modem/test/host_test/main/test_modem.cpp#L15-L36

jamesrob4, 2023-11-29:
I am using 1.0.2. 
I haven't made any changes, but did add one or two custom commands.

I am trying to reproduce the error by modifiying the pppos client example, no luck so far. I will update asap on this.

antfarming, 2023-11-29:
This is the one of the extra functions I added that caused a similar error. 
What I don't understand, is that second occurrence of AT+CEGREG? after OK. 

But as I said I am trying to reproduce the error using as close to the example code as possible.

```
[V][esp_modem_command_library.cpp:56] generic_get_string()::(100776):command_lib: generic_get_string
[V][esp_modem_command_library.cpp:66] operator()()::(100796):command_lib: Token: {AT+CGREG?}
[V][esp_modem_command_library.cpp:66] operator()()::(100806):command_lib: Token: {+CGREG: 0,5}
[V][esp_modem_command_library.cpp:66] operator()()::(100816):command_lib: Token: {
[V][esp_modem_command_library.cpp:66] operator()()::(100826):command_lib: Token: {OK}
[V][esp_modem_command_library.cpp:66] operator()()::(100836):command_lib: Token: {AT+CGREG?}
[V][esp_modem_command_library.cpp:66] operator()()::(100846):command_lib: Token: {PB DONE}

abort() was called at PC 0x40187217 on core 0
0x40187217: __cxxabiv1::__terminate(void (*)()) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48

Backtrace: 0x40081caa:0x3ffd8210 0x4008b0b9:0x3ffd8230 0x40092c86:0x3ffd8250 0x40187217:0x3ffd82c0 0x4018724c:0x3ffd82e0 0x40187327:0x3ffd8300 0x400f2ca5:0x3ffd8320 0x400f2d1f:0x3ffd8380 0x400f2dc9:0x3ffd83c0 0x400f22e1:0x3ffd8420 0x400f4581:0x3ffd8450 0x400f55c2:0x3ffd8490 0x400f03f1:0x3ffd8500 0x401c9dc2:0x3ffd8520 0x400e3ed3:0x3ffd8550 0x400e4795:0x3ffd85a0 0x4008dc39:0x3ffd8e90
0x40081caa: panic_abort at /Users/james/esp/esp-idf/components/esp_system/panic.c:452
0x4008b0b9: esp_system_abort at /Users/james/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x40092c86: abort at /Users/james/esp/esp-idf/components/newlib/abort.c:38
0x40187217: __cxxabiv1::__terminate(void (*)()) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
0x4018724c: std::terminate() at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
0x40187327: __cxa_throw at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:98
0x400f2ca5: esp_modem::throw_if_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int) at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_exception.hpp:86
0x400f2d1f: esp_modem::DTE::command_cb::set(std::function<esp_modem::command_result (unsigned char*, unsigned int)>, char) at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_dte.hpp:215 (discriminator 2)
0x400f2dc9: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned long, char) at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_dte.cpp:139
0x400f22e1: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned long) at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_dte.cpp:149
0x400f4581: esp_modem::dce_commands::generic_get_string(esp_modem::CommandableIf*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string_view<char, std::char_traits<char> >&, unsigned long) at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_command_library.cpp:57
0x400f55c2: esp_modem::dce_commands::get_network_attachment_cgreg(esp_modem::CommandableIf*, int&) at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_command_library.cpp:524
0x400f03f1: esp_modem::GenericModule::get_network_attachment_cgreg(int&) at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_modules.cpp:38
0x401c9dc2: esp_modem::command_result esp_modem::DCE::get_network_attachment_cgreg<int&>(int&) at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_dce.hpp:114
 (inlined by) esp_modem_get_network_attachment_cgreg at /Users/james/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_c_api.cpp:353
0x400e3ed3: waitForNetworkAttachment at /Users/james/repos/olarm_esp32_communicator_fw/components/NetworkingClient/Cell.c:1088 (discriminator 15)
0x400e4795: taskCellRun at /Users/james/repos/olarm_esp32_communicator_fw/components/NetworkingClient/Cell.c:311
0x4008dc39: vPortTaskWrapper at /Users/james/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

command_result get_network_attachment_cereg(CommandableIf *t, int &state)
{
    ESP_LOGV(TAG, "%s", __func__ );
    std::string_view out;
    auto ret = generic_get_string(t, "AT+CEREG?\r", out);
    if (ret != command_result::OK) {
        return ret;
    }
    constexpr std::string_view pattern = "+CEREG: 0,";
    constexpr int pos = pattern.size();
    if (out.find(pattern) == std::string::npos) {
        return command_result::FAIL;
    }

    if (std::from_chars(out.data() + pos, out.data() + out.size(), state).ec == std::errc::invalid_argument) {
        return command_result::FAIL;
    }

    return command_result::OK;
}
```

david-cermak, 2023-11-29:
> What I don't understand, is that second occurrence of AT+CGREG? after OK.

Does it mean that you're sending the `cgerg` command only once?
Could you try to add another debug log in the `generic_get_string()` to see the entire data coming in?
```patch
diff --git a/components/esp_modem/src/esp_modem_command_library.cpp b/components/esp_modem/src/esp_modem_command_library.cpp
index 3947d70..4364614 100644
--- a/components/esp_modem/src/esp_modem_command_library.cpp
+++ b/components/esp_modem/src/esp_modem_command_library.cpp
@@ -57,6 +57,7 @@ static command_result generic_get_string(CommandableIf *t, const std::string &co
     return t->command(command, [&](uint8_t *data, size_t len) {
         size_t pos = 0;
         std::string_view response((char *)data, len);
+        ESP_LOGD(TAG, "Response: {%.*s}\n", static_cast<int>(response.size()), response.data());
         while ((pos = response.find('\n')) != std::string::npos) {
             std::string_view token = response.substr(0, pos);
             for (auto it = token.end() - 1; it > token.begin(); it--) // strip trailing CR or LF
```

> I am trying to reproduce the error using the pppos client example.

That would be really helpful!

antfarming, 2023-11-29:
I have not been able to replicate in pppos client example yet. But on my other project I put in your debug statement and got the following output (the crash is very occasional).

I'm only calling get_network_attachment_cgreg once. 

```
[V][esp_modem_command_library.cpp:523] get_network_attachment_cgreg()::(27066):command_lib: get_network_attachment_cgreg
[V][esp_modem_command_library.cpp:56] generic_get_string()::(27066):command_lib: generic_get_string
[D][esp_modem_command_library.cpp:60] operator()()::(27086):command_lib: Response: {AT+CGREG?
+CGREG: 0,0

OK
}

[V][esp_modem_command_library.cpp:67] operator()()::(27086):command_lib: Token: {AT+CGREG?}

[V][esp_modem_command_library.cpp:67] operator()()::(27096):command_lib: Token: {+CGREG: 0,0}

[V][esp_modem_command_library.cpp:67] operator()()::(27106):command_lib: Token: {

[V][esp_modem_command_library.cpp:67] operator()()::(27116):command_lib: Token: {OK}

[D][esp_modem_command_library.cpp:60] operator()()::(27126):command_lib: Response: {AT+CGREG?
PB DONE
}

[V][esp_modem_command_library.cpp:67] operator()()::(27136):command_lib: Token: {AT+CGREG?}

[V][esp_modem_command_library.cpp:67] operator()()::(27146):command_lib: Token: {PB DONE}

abort() was called at PC 0x40183a1b on core 0
0x40183a1b: __cxxabiv1::__terminate(void (*)()) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48

Backtrace: 0x40081c72:0x3ffd5300 0x4008aae9:0x3ffd5320 0x40092556:0x3ffd5340 0x40183a1b:0x3ffd53b0 0x40183a50:0x3ffd53d0 0x40183b2b:0x3ffd53f0 0x400f1491:0x3ffd5410 0x400f150b:0x3ffd5470 0x400f15b5:0x3ffd54b0 0x400f0ae1:0x3ffd5510 0x400f2d6d:0x3ffd5540 0x400f3dda:0x3ffd5580 0x400eebf1:0x3ffd55f0 0x401c634e:0x3ffd5610 0x400e26af:0x3ffd5640 0x400e2801:0x3ffd5690 0x400e2e8b:0x3ffd56c0 0x4008d669:0x3ffd5fd0
0x40081c72: panic_abort at /Users/anthony/esp/esp-idf/components/esp_system/panic.c:452
0x4008aae9: esp_system_abort at /Users/anthony/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x40092556: abort at /Users/anthony/esp/esp-idf/components/newlib/abort.c:38
0x40183a1b: __cxxabiv1::__terminate(void (*)()) at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
0x40183a50: std::terminate() at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
0x40183b2b: __cxa_throw at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:98
0x400f1491: esp_modem::throw_if_error(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, int) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_exception.hpp:86
0x400f150b: esp_modem::DTE::command_cb::set(std::function<esp_modem::command_result (unsigned char*, unsigned int)>, char) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_dte.hpp:215 (discriminator 2)
0x400f15b5: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned long, char) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_dte.cpp:139
0x400f0ae1: esp_modem::DTE::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<esp_modem::command_result (unsigned char*, unsigned int)>, unsigned long) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_dte.cpp:149
0x400f2d6d: esp_modem::dce_commands::generic_get_string(esp_modem::CommandableIf*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string_view<char, std::char_traits<char> >&, unsigned long) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_command_library.cpp:57
0x400f3dda: esp_modem::dce_commands::get_network_attachment_cgreg(esp_modem::CommandableIf*, int&) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_command_library.cpp:525
0x400eebf1: esp_modem::GenericModule::get_network_attachment_cgreg(int&) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_modules.cpp:38
0x401c634e: esp_modem::command_result esp_modem::DCE::get_network_attachment_cgreg<int&>(int&) at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/include/cxx_include/esp_modem_dce.hpp:114
 (inlined by) esp_modem_get_network_attachment_cgreg at /Users/anthony/repos/olarm_esp32_communicator_fw/components/esp_modem/src/esp_modem_c_api.cpp:353
0x400e26af: waitForNetworkAttachment at /Users/anthony/repos/olarm_esp32_communicator_fw/components/NetworkingClient/Cell.c:1126 (discriminator 15)
0x400e2801: modemStart at /Users/anthony/repos/olarm_esp32_communicator_fw/components/NetworkingClient/Cell.c:987
0x400e2e8b: taskCellRun at /Users/anthony/repos/olarm_esp32_communicator_fw/components/NetworkingClient/Cell.c:264
0x4008d669: vPortTaskWrapper at /Users/anthony/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

david-cermak, 2023-11-29:
> I have not been able to replicate in pppos client example yet

One reason for that could be a memory corruption, you can try to set heap debugging to comprehensive to check if the crash/behaviour changes in some way: `CONFIG_HEAP_POISONING_COMPREHENSIVE=y`
```
(Top) → Component config → Heap memory debugging
```

From the log you shared I could only tell that the data arrived in two chunks. Assuming that in the first round we'll get the expected command response with `OK` and then just the `PB DONE` so it really looks like it's writing somewhere to the midsection of the buffer?

david-cermak, 2023-11-29:
But when thinking about it, we shouldn't allow further processing after getting `command_result::OK` or `command_result::FAIL`.

This should fix the issue:
```patch
diff --git a/components/esp_modem/src/esp_modem_dte.cpp b/components/esp_modem/src/esp_modem_dte.cpp
index d402166..db6a4a1 100644
--- a/components/esp_modem/src/esp_modem_dte.cpp
+++ b/components/esp_modem/src/esp_modem_dte.cpp
@@ -347,7 +347,7 @@ void DTE::on_read(got_line_cb on_read_cb)
 bool DTE::command_cb::process_line(uint8_t *data, size_t consumed, size_t len)
 {
-    if (memchr(data + consumed, separator, len)) {
+    if (memchr(data + consumed, separator, len) && result == command_result::TIMEOUT) {
         result = got_line(data, consumed + len);
         if (result == command_result::OK || result == command_result::FAIL) {
             signal.set(GOT_LINE);
```

will think about it more, but it really looks like an oversight in the modem codebase.

david-cermak, 2023-12-01:
Thanks for sharing the logs, it helped a lot! The problem was, that the unexpected data after the correct reply must preempt the calling thread (which never happened in my tests)

>  I am curious if they should be updated to include a case for this fix?

Good point, I've added #447 (should fail without the fix)

antfarming, 2023-12-03:
Have been running with no errors for 3 days. Thank you for the help! Am closing the ticket

david-cermak, 2023-12-05:
Thanks for your feedback and the extensive testing!

Let me just reopen this, so we can formally close the issue once its fix lands on master.

closed 2023-12-06 by david-cermak