espressif/esp-protocols #432: Provide an example for adding a new Module/Device.

a2800276, 2023-11-15:
### Is your feature request related to a problem?

We are attempting to create a custom device to use a Quectel BC66 module which requires some 'custom' commands to initialize. The esp_modem documentation clearly states:

_Adding a new device is a must-have requirement for the esp-modem component. Different modules support different commands, or some commands might have a different implementation. Adding a new device means to provide a new implementation as a class derived from GenericModule, where we could add new commands or modify the existing ones._

While the implementation seems to be quite sophisticated, the architecture is seriously frustrating our attempts to find any approach to actually extend `GenericModule`. We need to end up with a modem, which can only be instantiated through the DCE Factory, which in turn is a template class that requires parameterization using a ModemType enum class. This enum class contains a number of hardcoded modules which I can't extend to contain a new ModemType without modifying the actual library code. 

This problem is handled in the 'simple_cmux' example for the 7070 module to use gnss by instantiating a custom subtype of Factory  to specifically instantiate 7070_gnss instances. Having to create a custom Factory subtype for each new type of Module seems to pervert the idea of Factories, though. So I imagine this is not the intended mechanism to integrate new module types. It seems like I would also need to add `command_result <g|s>et_something_my_module(CommandableIf *t)` functions to  be processed by the DCE macros? 

I haven't been able to find any possibility to add a custom module in C.

As the library is so sophisticated in its use of architectural patterns and mechanisms, and adding a module seems like a fairly widespread use case, it would be really helpful to provide a minimal example of extending the library to support a custom module type. Quite honestly, I can't really remember ever seeing code making so much use of such a wide variety of C and CPP features that add to indirection. 

### Describe the solution you'd like.

A simple example of how to extend the library by a custom module that requires a different AT command to fullfill one of the interfaces.

### Describe alternatives you've considered.

Seriously refactor the esp_modem component to make the code a bit more accessible and rely less on hard-coded constants, dynamic macros, template metaprogramming, etc.

david-cermak, 2023-11-15:
Hi @a2800276 

I agree that adding a custom module (esp when using C-API) isn't very user friendly. The example is added here https://github.com/espressif/esp-protocols/pull/413

a2800276, 2023-11-15:
Thanks very much for the quick reply... So just to confirm: this is currently not supported until the referenced pull request is merged?
     -tim

david-cermak, 2023-11-15:
>  this is currently not supported until the referenced pull request is merged?

You can use the same principles, as the PR contains minimal changes to the modem layers. But If you want to use C-API it's better to wait for a new version. I'll merge it soon.

david-cermak, 2023-11-30:
Fixed in 0254d50128b91825d9d3578c37fdf7d7bb88de04

closed 2023-11-30 by david-cermak