espressif/esp-protocols #448: esp-modem: handle SIM card is deactivated

marchingband, 2023-12-04:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Often times users do not pay their cellular bill on time, and so we get `Modem Disconnect from PPP Server`. If ex. the bill is then paid, the modem does not automatically reconnect, and requires a reset.

If there is a good way to handle this case, can some example code possibly be provided?

How can I setup a loop to retry the connection to the PPP server?

david-cermak, 2023-12-05:
We already have some examples that retry connecting periodically with some timeout/backoff-time, e.g.

https://github.com/espressif/esp-protocols/blob/1bf6843b3ee6d3954be010aafe3c8338dc9078db/examples/esp_netif/multiple_netifs/main/ppp_connect_esp_modem.c#L55-L93

https://github.com/espressif/esp-protocols/blob/1bf6843b3ee6d3954be010aafe3c8338dc9078db/components/esp_modem/examples/ap_to_pppos/main/ap_to_pppos.c#L154-L158

---
In your case, however, it would be useful to check for the reason of the disconnection (I think BG96 provides some netwrok act command to check the reason), as it makes no sense to retry the connection every 5 seconds if the reason is an unpaid cellular bill.

marchingband, 2023-12-05:
Thank you @david-cermak 
I had seen those examples, and created a task that combines ap_to_pppos and pppos_client, I'll paste the code below

```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_netif.h"
#include "esp_netif_ppp.h"
#include "esp_modem_api.h"
#include "esp_log.h"
#include "sdkconfig.h"
#include "mqtt_client.h"
#include "driver/gpio.h"
#include "esp_mac.h"

#include "modem_config.h"

static const char* TAG = "modem";

static EventGroupHandle_t event_group = NULL;
static const int CONNECT_BIT = BIT0;
static const int DISCONNECT_BIT = BIT2;

static esp_modem_dce_t *dce;
static esp_netif_t *esp_netif;

static void pwk_gpio_init(void){
    gpio_config_t io_conf = {};
    io_conf.intr_type = GPIO_INTR_DISABLE;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pin_bit_mask = ( 1ULL << BG96_PWR_PIN );
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);
    gpio_set_level(BG96_PWR_PIN, 0);
}

static void bg96_set_pwr(bool on){
    gpio_set_level(BG96_PWR_PIN, on);
}

static bool modem_start_network()
{
    return esp_modem_set_mode(dce, ESP_MODEM_MODE_DATA) == ESP_OK;
}

static bool modem_stop_network()
{
    return esp_modem_set_mode(dce, ESP_MODEM_MODE_COMMAND);
}

static bool modem_check_sync()
{
    return esp_modem_sync(dce) == ESP_OK;
}

static void modem_reset()
{
    esp_modem_reset(dce);
}

static bool modem_check_signal()
{
    int rssi, ber;
    if (esp_modem_get_signal_quality(dce, &rssi, &ber) == ESP_OK) {
        return rssi != 99 && rssi > 5;
    }
    return false;
}
[... 139 lines cut ...]

void modem_init( void ){
    xTaskCreate(modem_task, "modem_task", 4096, NULL, 1, NULL);
}
```

The initial connection stage works well.
I use my SIM CARD provider dashboard to deactivate the SIM.
The ESP sees the disconnect and loops through toggling between DATA and COMMAND modes.
I then reactivate the SIM and wait for it to reconnect, but it never does, logging in a loop:

```
modem: Modem does not respond, maybe in DATA mode? ...exiting network mode
esp-netif_lwip-ppp: esp_netif_start_ppp: PPP connection cannot be started
modem: PPP state changed event 512
esp-netif_lwip-ppp: User interrupt
modem: PPP state changed event 5
modem: User interrupted event from netif:0x3fcc82a8
esp_modem_netif: PPP state changed event 5
```

If after 30 minutes, I reset the ESP32, then the reconnection works. My tests indicate that it takes about 10 minutes for the SIM to turn back on.

My guess was that there is more work to be done, to ask the modem to refresh its connection and get an IP.
Is there something I am missing?

marchingband, 2023-12-05:
@david-cermak thank you
I added the hangup to the loop, but unfortunately no change in the behaviour, it does not get an IP address, until I give up and reset the MCU. Here is where I placed the hangup:
```
static void start_network(void)
{
    // EventBits_t bits = 0;
    while ((xEventGroupGetBits(event_group) & CONNECT_BIT) == 0) {
        if (!modem_check_sync()) {
            ESP_LOGE(TAG, "Modem does not respond, maybe in DATA mode? ...exiting network mode");
            modem_stop_network();
            if (!modem_check_sync()) {
                ESP_LOGE(TAG, "Modem does not respond to AT ...restarting");
                modem_reset();
                ESP_LOGI(TAG, "Restarted");
            }
            int ret = esp_modem_at(dce, "ATH", NULL, 2000);
            ESP_LOGI(TAG, "hangup %s", ret == ESP_OK ? "Success" : "Failure");
            continue;
        }
        if (!modem_check_signal()) {
            ESP_LOGI(TAG, "Poor signal ...will check after 5s");
            vTaskDelay(pdMS_TO_TICKS(5000));
            continue;
        }
        if (!modem_start_network()) {
            ESP_LOGE(TAG, "Modem could not enter network mode ...will retry after 10s");
            vTaskDelay(pdMS_TO_TICKS(10000));
            continue;
        }
        EventBits_t bits = xEventGroupWaitBits(event_group, (DISCONNECT_BIT | CONNECT_BIT), pdTRUE, pdFALSE, pdMS_TO_TICKS(30000));
        if (bits & DISCONNECT_BIT) {
            ESP_LOGE(TAG, "Modem got disconnected ...retrying");
            modem_stop_network();
        } else if(bits & CONNECT_BIT){
            ESP_LOGE(TAG, "Modem got connected");
            break;
        }
    }
}
```
Could there be anything else I am missing?

david-cermak, 2023-12-06:
This is weird, could you please share the full log?

Does this command succeed?
```cpp
esp_modem_set_mode(dce, ESP_MODEM_MODE_DATA) == ESP_OK
```

david-cermak, 2023-12-08:
Interesting that I don't see this message in your log anymore:
```
esp-netif_lwip-ppp: esp_netif_start_ppp: PPP connection cannot be started
```
(That's why I asked if the `set_mode(...)` command succeeded or not, as it seemed like in it did not from the previous log, but it does in the recent log)

And by pressing reset, you reset only the ESP32, not the modem? Would it help if you destroy the `dce` and the `esp_netif` and create it again?

I'm sorry I cannot see what's wrong. I think I would have to try to simulate this issue and look into it.
Could you please also check if there's a difference in the output of this command:
```
 AT+CGREG 
```
when you disable or re-enable your SIM card?

marchingband, 2023-12-09:
@david-cermak thank you.

> Interesting that I don't see this message in your log anymore:

The first log was captured from a very different process, which I had created before discovering the ap_to_pppos example, and following that reconnect procedure.

> And by pressing reset, you reset only the ESP32, not the modem?

correct, I just press reset on the ESP32. The initialization process does then toggle the PWK pin, which I believe triggers a reset in the modem. It looks like this:

```
    pwk_gpio_init();

    // cycle pwk pin
    bg96_set_pwr(true);
    vTaskDelay(pdMS_TO_TICKS(1000));
    bg96_set_pwr(false);
    vTaskDelay(pdMS_TO_TICKS(1000));
    bg96_set_pwr(true);
    vTaskDelay(pdMS_TO_TICKS(10000)); // wait for boot
```

> Would it help if you destroy the dce and the esp_netif and create it again?

I have not yet tried this. If you would like me to test that approach I can give it a shot.

> Could you please also check if there's a difference in the output of this command:
>  AT+CGREG 
> when you disable or re-enable your SIM card?

I ran this test, and the function always returns `0` (`ESP_OK`), using this code:
```
    int ret = esp_modem_at(dce, "AT+CGREG", NULL, 2000);
    ESP_LOGI(TAG, "\n\nAT+CGREG returned %d", ret);
```
Changing the SIM activation state makes no difference.

Incidentally, examining the manual fro BG96, I see that this `AT+CGREG` command return a lot of data. Is there a convenience function in this library to access all the data returned from an AT command? I can't seem to find one.

>  I think I would have to try to simulate this issue and look into it.

Thanks so much for your attention on this, I am happy to run whatever test that might help, should you decide to test this yourself.

marchingband, 2023-12-09:
I upped the timeout to 90 seconds, and the reconnection routine succeeded.
Again thanks so much for your support. Running the CGREG command queued me to realize it was making progress in the connection, and I was resetting it too soon.
30 seconds seems like an eternity ... so I am not sure what is up there.
Feel free to close the issue, or let me know if there is more I can do.

david-cermak, 2024-03-20:
Closing as I don't have any better opinion.
My interpretation is that when your SIM card gets deactivated it simply *takes more time* to reflect on PPP layers than *normal* disconnections. If you'd like to investigate this further I'd suggest enabling PPP phase notifications to see if you there's anything indicating network status terminated and retrying (those retries would IMO prolong the reconnection phase).

closed 2024-03-20 by david-cermak