espressif/esp-protocols #457: Question about Data Mode vs. AT Mode

diplfranzhoepfinger, 2023-12-12:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Can we request the Status of a Port ? 

if it is in Data Mode or AT Mode ? 

especially when a Port in Data Mode swiches to AT Mode initiated by the Modem, 
OR the Modem is restarted or something else happens, 
then can we get notified somehow ? 

thanks, 
Franz

david-cermak, 2023-12-21:
Yes, this is planned for [`modem-v1.2`](https://github.com/espressif/esp-protocols/milestone/2) as mode autodetect feature.
I'll add this to the milestone.

david-cermak, 2024-11-07:
The feature has been merged in https://github.com/espressif/esp-protocols/pull/612

will be published in v1.2.0 per https://github.com/espressif/esp-protocols/pull/688

closed 2024-11-07 by david-cermak