espressif/esp-protocols #458: websocket not able to receive all the chunk of the data and getting parts when receiving

Pavan-keus, 2023-12-14:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]
in the above image u can see that i receive 1436 bytes and after 7 bytes but it should be 1443 why it got fragmented ? any idea and my configuration for websocket is 
 k_keusWebsocket_webSocketConfig_g.host = CONFIG_KEUS_WEB_SOCKET_HOST;
    k_keusWebsocket_webSocketConfig_g.port = CONFIG_KEUS_WEB_SOCKET_PORT;
    k_keusWebsocket_webSocketConfig_g.buffer_size = 8192;
    k_keusWebsocket_webSocketConfig_g.task_prio = 10;
    ESP_LOGI(TAG, "Connecting to Websocket ...");

    K_keusWebsocket_webSocketClient_g = esp_websocket_client_init(&k_keusWebsocket_webSocketConfig_g);
 esp_websocket_client_start(K_keusWebsocket_webSocketClient_g);

gabsuren, 2024-01-17:
@Pavan-keus the issue has been fixed in the latest `master` of [esp-protocols](https://github.com/espressif/esp-protocols). 
Could you please retry and see if you face it again?

Thanks you
Suren

closed 2024-02-08 by gabsuren

joelguittet, 2024-08-16:
Hello @gabsuren @Pavan-keus 

I have the exact same issue. Using the latest release websocket 1.2.3.

@gabsuren I browsed the history of websocket code and I don't see any recent (2023-2024) commits related to a fix for this problem. Can you elaborate on this "the issue has been fixed in the latest master" please ??

@Pavan-keus on your side is it fixed ? Just by updating the websocket client or maybe you have done other modifications on your application ?

Thanks for the feedbacks!
Joel

david-cermak, 2024-08-16:
Hi @joelguittet 

Do you receive your data in chunks? Did you check the values of `payload_len` and `payload_offset` variables of `esp_websocket_event_data_t` on getting an `WEBSOCKET_EVENT_DATA` event?

---

> Can you elaborate on this "the issue has been fixed in the latest master" please ??

I think that Suren mistaken this issue with another one (related to fragmentation of a connect message). the websocket client doesn't decompose fragmented data, so if they arrive in pieces they are passed like that in multiple events.

joelguittet, 2024-08-16:
@david-cermak good point! I just checked the payload length and offset values adding a log. This is exactly my issue. I need to concatenate the data before sending this to the upper layer.
Thanks a lot