espressif/esp-protocols #464: Define a subdomain using mDNS

yanmartins, 2023-12-21:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I would like to create a link like `parent.child.local`. It is possible?

I had tried something similar:
```c
ESP_ERROR_CHECK(mdns_init());
ESP_ERROR_CHECK(mdns_hostname_set("parent"));
ESP_ERROR_CHECK(mdns_instance_name_set("child"));
```

If this is possible, is there any example?

david-cermak, 2023-12-22:
No this won't work, you can only create *subtypes* for services, here's an example:

https://github.com/espressif/esp-protocols/blob/a6845f4cddac36cf5ce2a68e077231fede359800/components/mdns/examples/query_advertise/main/mdns_example_main.c#L57

yanmartins, 2023-12-22:
> No this won't work, you can only create *subtypes* for services, here's an example:
> 
> https://github.com/espressif/esp-protocols/blob/a6845f4cddac36cf5ce2a68e077231fede359800/components/mdns/examples/query_advertise/main/mdns_example_main.c#L57

Thank you!

yanmartins, 2023-12-22:
@david-cermak 

Another question: this library doesn't allow me to have a hostname for each interface, right?

For example:
- On WiFi STA: `example-one.local`
- On WiFi AP: `example-two.local`
- On Ethernet: `example-three.local`

david-cermak, 2024-01-03:
No, you cannot advertise different names on different interfaces. But. you can delegate replies and create aliases.. and you can advertize only on specific interfaces.

so it's possible to have a self-hostname as `example.local` (resolving to say `192.168.4.1` (AP), `192.168.0.111` (STA) and `192.168.0.222` (Ethernet)) and at the same time delegate
* `example-one.local` as `192.168.0.111`
* `example-two.local` as `192.168.4.1`
* `example-three.local` as `192.168.0.222`

(but all names would be available on all the interfaces you'll advertize them on)

closed 2024-01-11 by yanmartins