espressif/esp-protocols #468: problems migrating ESP-Modem from 2020-version to 1.0.5

JB-DX, 2024-01-02:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to migrate ESP-Modem from the old version of 2020 to 1.0.5 
The new version lacks features that the old one had.

1. the old version provided a modem_event_handler with ESP_MODEM_EVENT_UNKNOWN that delivered 
`"RING"` or `"+CLIP: ..."` when in command-mode and being called by a telephone.
In the new version I found no read-callback or similar that could be used.

2. `esp_modem_at()` returns only the last line of commands like `at+cpbf`  (read SIM-contact entries) that returns several lines 
i.e.  `at+cpbf="user"`  returns 3 lines if  SIM-contacts contain  user, user1, user2
In the old version I could create a custom handler for that.

Is there any solution for the above 2 issues?

david-cermak, 2024-01-03:
I'm sorry, I don't really have a good suggestion for C right now (other than create a custom DCE as in the console example and wrap it in C-API)

> dce->set_on_read() obviously does not compile.

It does compile in that example, but as said above -- uses some C++ magic to conveniently override `set_on_read()` in

https://github.com/espressif/esp-protocols/blob/7d0eb5c48ca0aa1495bd41baea84b68510c15828/components/esp_modem/examples/modem_console/main/my_module_dce.hpp#L67-L81

>For multiline reponses esp_modem_at() either needs the additional parameter accept_multi_line separate function esp_modem_at_multiline() so esp_modem_at() can be left unchanged.

Yes, I'd probably add a new function, as you proposed, which could be used more universally. I would also remove the trailing `\r` here, so the new function could be used to send any kind of characters, e.g. the `+++`:

https://github.com/espressif/esp-protocols/blob/7d0eb5c48ca0aa1495bd41baea84b68510c15828/components/esp_modem/src/esp_modem_command_library.cpp#L393-L394

JB-DX, 2024-01-08:
I managed to add a esp_modem_set_read_cb() in C and that works.
When will v1.2 available?

Another problem:
How can I reconnect to PPP ?  I'm using SIM800. 
After an established connection I call `esp_modem_set_mode(dce, ESP_MODEM_MODE_COMMAND)`
and later on  (it can be hours later) 
`esp_modem_set_mode(dce, ESP_MODEM_MODE_DATA);`

But that results just in a "User interrupted event" and it doesn't get connected.

The SIM800 sends:
```
CONNECT
~ÿ}#À!}!}!} }2}"}&} }*} } }#}$À#}'}"}(}"Uƒ~~ÿ}#À!}#}$} }*}"}&} }*} } t"~~ÿ}#À!}!}"} }.}"}&} }*} } }'}"}(}"ÐÓ~~ÿ}#À!}"}%} }4}"}&} }*} } }%}&áÏFˆ}'}"}(}"}=}(~~€!À¨þþHÌ~~€!-àB~~ÿÀ!…r~
NO CARRIER

```

A similar question right at the first connection-try after reset:
If it can't get a PPP connection because the provider has no capacity, I get  `IP_EVENT_PPP_LOST_IP`
How can I start a retry? Setting `ESP_MODEM_MODE_DATA` doesn't work.
Do I have to destroy and re-create the DCE each time? That sounds a bit awkward.

JB-DX, 2024-01-11:
Now I got it running. The reason was the missing APN

david-cermak, 2024-01-11:
Thanks for the update!

> When will v1.2 available?

I'm currently getting ready for `v1.1`, should be available next week or so,.
Version `v1.2` would add a few features, so I think it should be ready sometime in Feb/March. As for the URC handler, its not a big change and I think I could get it working after the 1.1 release.

david-cermak, 2024-03-18:
Closing, as I think all questions have been answered. Please reopen or create new issues if it's not the case.

closed 2024-03-18 by david-cermak

JB-DX, 2024-04-03:
I assumed that the URC handler would have been implemented by now, but I can't find it.

in esp_modem_c_api.cpp I added this:
```
extern "C" esp_err_t esp_modem_set_read_cb(esp_modem_dce_t *dce_wrap, esp_modem_terminal_read_cbt read_cb)
{
    if (dce_wrap == nullptr || dce_wrap->dce == nullptr || dce_wrap->dte == nullptr) {
        return ESP_ERR_INVALID_ARG;
    }
    if (read_cb) {
        dce_wrap->dte->on_read([read_cb](uint8_t *line, size_t leng) {
           read_cb(line, leng);
            return command_result::TIMEOUT;  
        });
    } else {
        dce_wrap->dte->on_read(nullptr);
    }
    return ESP_OK;
}

```
and in esp_modem_c_api_types.h add this:
`esp_err_t esp_modem_set_read_cb(esp_modem_dce_t *dce_wrap, esp_modem_terminal_read_cbt read_cb);
`