espressif/esp-protocols #480: ESP_MODEM_MODE_COMMAND fires IP_EVENT_PPP_LOST_IP only after 120 seconds

JB-DX, 2024-01-11:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-Modem 1.0.5 using example: `pppos_client`

`esp_modem_set_mode(ESP_MODEM_MODE_COMMAND)` does not fire  `IP_EVENT_PPP_LOST_IP` immediately.
Instead this event fires only after 120 seconds and then again after another 120 seconds.

Can this fixed?  
Why does it fire twice even when there existed no more IP-connection?

david-cermak, 2024-01-11:
> esp_modem_set_mode(ESP_MODEM_MODE_COMMAND) does not fire IP_EVENT_PPP_LOST_IP immediately.

This event comes from IDF lwip wrapper, which simply starts a timer and checks if the address is `IP_ANY`, it is configurable in `CONFIG_ESP_NETIF_IP_LOST_TIMER_INTERVAL`. You can also turn this off by setting this config to 0.

> Why does it fire twice even when there existed no more IP-connection?

The same event is also posted from the PPP layers, when the link gets closed.

 > Can this fixed?

The PPP netif is a bit different than other interfaces, so I'd probably suggest disbling the IP_LOST_TIMER and subscribe to PPP_LINK status events. I provides more granular status information, so you'd get to know exactly when the PPP link terminates (than it retries, reconnects, etc...) so the LOST_IP comes a bit later.
Also, if there's no network traffic, the interface wouldn't detect a link issue (bad connection, etc), so if you want to check for  link status, you'd need to use a LCP keepalive mechanism (could be enabled in Kconfig)

JB-DX, 2024-01-12:
Thanks for your hints.
I will not  IP_LOST_TIMER fron now on.

As PPP-events I only see NETIF_PPP_ERRORUSER
shouldn't be link-status-events also arrive here?

david-cermak, 2024-03-11:
@JB-DX Sorry I missed the last question.

> As PPP-events I only see NETIF_PPP_ERRORUSER

You probably just need to enable `CONFIG_LWIP_PPP_NOTIFY_PHASE_SUPPORT=y`

david-cermak, 2024-03-19:
it seems like all questions've been answered, please reopen if not.

closed 2024-03-19 by david-cermak