espressif/esp-protocols #482: OTA fails due to many "HW FIFO Overflows"

JB-DX, 2024-01-15:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using v1.0.5 and everything works fine so far.
But not OTA. 
It throws many `uart_terminal: HW FIFO Overflow` and then finally gives up.

I had a similar problem with the old version of 2020, and fixed it by increasing the UART buffer sizes.

So I tried it by increasing it from 400 to 2048, but without any improvements.

```
#define UART_BUF_SIZE   2048  

dte_config.uart_config.rx_buffer_size = 2*UART_BUF_SIZE; 
dte_config.uart_config.tx_buffer_size = UART_BUF_SIZE; 
dte_config.dte_buffer_size = UART_BUF_SIZE; 
dte_config.uart_config.event_queue_size = 30; 

```
A bit strange because SIM800 only uses G2 and its speed is quite low, so why is there a fifo overflow at all?

Any ideas how to fix it?

david-cermak, 2024-01-15:
Have you placed your UART ISR to IRAM?
```
CONFIG_UART_ISR_IN_IRAM=y
```

I thought we had this documented somewhere, but couldn't find it. Added https://github.com/espressif/esp-protocols/pull/484

There's also an example of using two phase OTA (using it as a test here):

https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/test/target_ota#readme

JB-DX, 2024-01-15:
Thank you.
It wasn't in IRAM,
Now it works excellent.
Case closed.

closed 2024-01-15 by JB-DX