espressif/esp-protocols #485: set APN after esp_modem_new_dev() is called?

JB-DX, 2024-01-15:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is there a way to define an APN after `esp_modem_new_dev()` is called?

In our case we read the APN off the SIM-card. This requires a valid dce handle, which we only get from `esp_modem_new_dev()`. So we end up in a catch-22 situation.
Currently we create a dce just to read the SIM-card, then deinit() and init() the dce again.
Not that optimal, but doable.
Is there a better approach?

```
 esp_modem_dce_config_t dce_config = ESP_MODEM_DCE_DEFAULT_CONFIG(myAPN);  
 ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event, NULL));
 ESP_ERROR_CHECK(esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &onPPP_Event, NULL));
....
dce= esp_modem_new_dev(ESP_MODEM_DCE_SIM800, &dte_config, &dce_config, gsmNetif);
```

david-cermak, 2024-01-16:
You can set APN via the pdp context as shown in

https://github.com/espressif/esp-protocols/blob/585e4b30b28e21ad3203519318f8252cef56be9f/components/esp_modem/examples/modem_console/main/modem_console_main.cpp#L393-L394

(I'll add corresponding C API if needed)

JB-DX, 2024-01-16:
Yes please add the C API,  since I only use C

closed 2024-01-22 by david-cermak