espressif/esp-protocols #505: Integrate UART_WAKEUP as part of the esp modem

srikanthpalvai, 2024-02-06:
### Is your feature request related to a problem?

 Integration of UART_WAKEUP in esp_modem component

### Describe the solution you'd like.

Dear ESP-IDF Development Team,

I hope this message finds you well. I'm reaching out to request a feature enhancement for the esp_modem component in the ESP-IDF repository.

Requirement:
I'm currently working on a project where I need to implement a functionality wherein upon receiving a new SMS on the SIM module, a specific message is sent to the UART in the form of +CMTI:"sm",0. This message serves as a trigger to wake up the ESP32 from light sleep mode. However, I've encountered an issue where the UART_WAKEUP feature is not integrated into the esp_modem component, hindering the smooth implementation of this functionality.

Observation:
I have attempted to integrate UART_WAKEUP and set a threshold of 3. However, the behavior is inconsistent. Despite sending SMS continuously, the wakeup is triggered intermittently. I've tested this scenario using the UART0 and UART1 pins on the ESP32S3.
https://github.com/espressif/esp-idf/blob/v5.1.2/examples/system/light_sleep/main/uart_wakeup.c#L84

But same worked in the lightsleep example. Please find the the screenshot.
Note : With ESP32 It's not working 
UART Pins ESP32 UART 1, RX_IO_NUM:9, TX_IO_NUM:10

[image]
[image]

srikanthpalvai, 2024-02-06:
[image]

srikanthpalvai, 2024-02-07:
@david-cermak 
Please have a look on it

david-cermak, 2024-02-09:
Hi @srikanthpalvai 

I'll look into it, but probably won't integrate the functionality into the `esp_modem`. I'll check if we can add this to one of our examples, just to demonstrate how the UART wakeup event could be used with this library.

srikanthpalvai, 2024-02-09:
From an end-user perspective, Integrating UART wakeup into esp_modem would enhance convenience for developers by reducing the need for additional patchwork. This integration could streamline processes and improve the overall usability of the library.

srikanthpalvai, 2024-02-28:
@david-cermak Kindky reminder ...

david-cermak, 2024-02-28:
Hi @srikanthpalvai 

I'm sorry, no update yet. This hasn't been planned yet, since we'll only consider it after `v1.2` is available (with already planned features).

I'd like to keep this open for some time to see if there are any related requests a ideas for other people.

srikanthpalvai, 2024-07-27:
Hi @david-cermak 
could you please add this feature into bucket.