espressif/esp-protocols #507: Compatibility of Simple CMUX Client Library with CAVLI C16QS Modem

Noushadalik, 2024-02-08:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
The purpose of this issue is to assess the compatibility of the Simple CMUX Client Library with the CAVLI C16QS Modem. Additionally, an application note for CMUX integration on the CAVLI C16QS Modem (based on 3GPP 27.010) has been provided for reference

[C16QS CMUX Application Note1.0.pdf](https://github.com/espressif/esp-protocols/files/14204704/C16QS.CMUX.Application.Note1.0.pdf)

Noushadalik, 2024-02-09:
HI,
Any update?

david-cermak, 2024-02-09:
Hi @Noushadalik 

Have you tried this library with your modem?
Looking at the Application Note, it looks like it should support the basic functionality of using DLC1 and DLC2 as either AT or data channel.
Unfortunately, I do not have this device with me to test it.

If you experience a specific problem, please report it here, with verbose logging enabled. (we were able to fix many issues in the past just by analyzing generic issue reports)

Noushadalik, 2024-02-12:
HELLO @david-cermak 

I changed the AT command to enable CMUX (from AT+CMUX=0 to AT+CMUX=1) as per the `CAVLI` C16QS modem CMUX application note.
[image]

But I can't enter CMUX mode. Detailed verbose logs are attached below

`I (1680) cmux_example: GSM POWER Off !!
I (5690) cmux_example: GSM POWER ON !!
I (7190) cmux_example: not set_flow_control, because 2-wire mode active.
V (7190) command_lib: set_cmux
V (7190) command_lib: generic_command_common
V (7190) command_lib: generic_command
D (7190) command_lib: generic_command command AT+CMUX=1
 
D (7200) command_lib: Response: ^boot.rom'v
 
V (7300) CMUX Received: 0x3ffbf90c   f9 03 73 01 d7 f9                                 |..s...|
D (7300) CMUX: Payload frame: dlci:00 type:73 payload:0 available:2
V (7310) CMUX Received: 0x3ffbf90c   f9 07 3f 01 de f9                                 |..?...|
D (7310) CMUX: Payload frame: dlci:01 type:3f payload:0 available:2
V (7320) CMUX Received: 0x3ffbf90c   f9 07 ef 27 0d 0a 52 44  59 0d 0a 0d 0a 2b 41 54  |...'..RDY....+AT|
V (7330) CMUX Received: 0x3ffbf91c   52 45 41 44 59 0d 0a eb  f9                       |READY....|
D (7340) CMUX: Payload frame: dlci:01 type:ef payload:19 available:21
E (8350) cmux_example: Failed to configure multiplexed command mode... exiting
V (8350) intr_alloc: esp_intr_free: Disabling int, killing handler
D (8350) event: deleted loop 0x3ffba760`

Noushadalik, 2024-02-13:
Hi @david-cermak 
Any update?

Noushadalik, 2024-02-14:
hello @david-cermak 
Any update Today?

david-cermak, 2024-02-14:
Thanks for testing this and sorry for the delay.

From the log, I can see that the device successfully switched to CMUX protocol, but failed later, probably trying to get one channel into data mode. This could be also related to the fact, that you've set the CMUX mode directly after powering the device up.

Could you please try to run the [console example](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console) (with the modified command library) and try to run these commands?

https://github.com/espressif/esp-protocols/blob/38a3631a27f71f25109f1d6e28d3c2ba309f292d/components/esp_modem/examples/modem_console/main/modem_console_main.cpp#L240-L249

Especially `CMUX1` (which would only switch to cmux protocol without any additional channel processing)
`CMUX2` would probably fail (cmux exit is usually very device specific) and share the verbose logs again. 

The expected behavior would be
1) `set_mode CMUX1` -> manually switched to the CMUX mode
2) `cmd AT` -> expect `OK` on DLCI1
3) `set_mode CMUX3` -> swap the channels
4) `cmd AT` -> expect `OK` on DLC2
5) `set_mode CMUX4` -> expect using the DLCI2 as the network channel with PPP protocol (might fail, but would reveal the reason in the logs)

Noushadalik, 2024-03-16:
hi @david-cermak 
Sorry for the delay.
We tested the Modem console example with our CAVLI16QS modem. We can't set the cmux mode by using the "set_mode CMUX1" command.
The responses are attached below.

[image]

Noushadalik, 2024-03-18:
hi @david-cermak 
Please let me know if any updates are available.

david-cermak, 2024-03-18:
Hi @Noushadalik 

What I meant was using the console application to *interact* with the device (not updating the app to send these commands):
1) Update the command library to use `AT+CMUX=1` (as you already did)
2) Use the console application **as is** and wait for the device to boot up
3) Send commands and share replies (as mentioned above in https://github.com/espressif/esp-protocols/issues/507#issuecomment-1944414357)

The interaction should look something like:

[image]

PS: Not sure if the VS code console allows you to type in commands, but you can use any other window console...

PPS: I've already order an evaluation board with this modem, so I'll take a look into it

Noushadalik, 2024-03-18:
Hi,
@david-cermak

As per your instructions, we tested the Modem console on Putty and waited for the device to boot up. The detailed logs are attached below.

Steps to reproduce:
1) Download [console example](https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/examples/modem_console) code 
2) Update the command library to use AT+CMUX=1
3) Build the code and flash to ESP
4) Open Putty and check the following commands
set_mode CMUX1 -
cmd AT -
set_mode CMUX3 -
cmd AT -
set_mode CMUX4 -
`
I (27) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (27) boot: compile time 12:43:40
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=211e4h (135652)                                                                              map
I (158) esp_image: segment 1: paddr=0003120c vaddr=3ff80063 size=00008h (     8)                                                                              load
I (159) esp_image: segment 2: paddr=0003121c vaddr=3ffb0000 size=02510h (  9488)                                                                              load
I (168) esp_image: segment 3: paddr=00033734 vaddr=40080000 size=0c8e4h ( 51428)                                                                              load
I (194) esp_image: segment 4: paddr=00040020 vaddr=400d0020 size=7b55ch (505180)                                                                              map
I (377) esp_image: segment 5: paddr=000bb584 vaddr=4008c8e4 size=01c2ch (  7212)                                                                              load
I (380) esp_image: segment 6: paddr=000bd1b8 vaddr=400c0000 size=00064h (   100)                                                                              load
I (383) esp_image: segment 7: paddr=000bd224 vaddr=50000000 size=00028h (    40)                                                                              load
I (398) boot: Loaded app from partition at offset 0x10000
I (398) boot: Disabling RNG early entropy source...
I (414) cpu_start: Pro cpu up.
D (414) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (414) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
D (421) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (428) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (435) cpu_start: Starting app cpu, entry point is 0x40081484
I (0) cpu_start: App cpu up.
D (449) clk: RTC_SLOW_CLK calibration value: 3149018
I (461) cpu_start: Pro cpu start user code
I (461) cpu_start: cpu freq: 160000000
I (461) cpu_start: Application information:
I (466) cpu_start: Project name:     modem-console
I (471) cpu_start: App version:      v4.4.3-dirty
I (477) cpu_start: Compile time:     Mar 18 2024 12:42:52
I (483) cpu_start: ELF file SHA256:  97a3d23437bfc22e...
I (489) cpu_start: ESP-IDF:          v4.4.3-dirty
V (494) memory_layout: reserved range is 0x3f4211cc - 0x3f421204
D (500) memory_layout: Checking 7 reserved memory ranges:
D (506) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (512) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb3fd0
D (519) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (525) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (531) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (538) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (544) memory_layout: Reserved memory range 0x40080000 - 0x4008e510
D (551) memory_layout: Building list of available memory regions:
V (557) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
V (563) memory_layout: Start of region 0x3ffae000 - 0x3ffb0000 overlaps reserved                                                                              0x3ffae000 - 0x3ffae6e0
D (573) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
[... 196 lines cut ...]
Command returned non-zero error code: 0x1 (ERROR)
esp> cmd AT
I (96995) modem_console: Sending command AT with timeout 1000
E (97995) modem_console: Failed with TIMEOUT
Command returned non-zero error code: 0x1 (ERROR)
esp> set_mode CMUX3
I (123995) modem_console: Switching to CMUX3 name...
I (124005) modem_console: OK
esp> cmd AT
I (131275) modem_console: Sending command AT with timeout 1000
E (132275) modem_console: Failed with TIMEOUT
Command returned non-zero error code: 0x1 (ERROR)
esp> set_mode CMUX4
I (141195) modem_console: Switching to CMUX4 name...
V (141195) command_lib: set_echo
V (141195) command_lib: generic_command_common
V (141195) command_lib: generic_command
D (141205) command_lib: generic_command command ATE0

E (141705) modem_console: Failed to set the desired mode
Command returned non-zero error code: 0x1 (ERROR)
esp>
D (716215) console.repl: empty line
esp>
`

david-cermak, 2024-03-18:
Thanks for testing this, could you please try to
* revert the command lib change, so that we use the original `AT+CMUX=0` ?
* try to set `CONFIG_ESP_MODEM_CMUX_DELAY_AFTER_DLCI_SETUP` to `500` or `1000` to see if anything changed?

Noushadalik, 2024-03-19:
Hi @david-cermak 
1)  when AT+CMUX=0

[image]

2)  when CONFIG_ESP_MODEM_CMUX_DELAY_AFTER_DLCI_SETUP = 500 (AT+CMUX=1)

[image]

3)  when CONFIG_ESP_MODEM_CMUX_DELAY_AFTER_DLCI_SETUP = 1000 (AT+CMUX=1)

[image]

david-cermak, 2024-03-19:
Hi @Noushadalik 

Thanks for testing this! Could you please remove (or comment out) these few lines?

https://github.com/espressif/esp-protocols/blob/cffe26818b432c428b8c1160bd8358b985d9d27c/components/esp_modem/src/esp_modem_cmux.cpp#L241-L245

and test again (with `CONFIG_ESP_MODEM_CMUX_DELAY_AFTER_DLCI_SETUP=500` and `AT+CMUX=1`)

Noushadalik, 2024-03-19:
Hi @david-cermak 
Here is the test outcome

[image]

david-cermak, 2024-03-19:
Nice, we're one step closer.
Could you please apply this patch and try again?
```patch
diff --git a/components/esp_modem/src/esp_modem_cmux.cpp b/components/esp_modem/src/esp_modem_cmux.cpp
index b1bfc78..34a87c5 100644
--- a/components/esp_modem/src/esp_modem_cmux.cpp
+++ b/components/esp_modem/src/esp_modem_cmux.cpp
@@ -137,7 +137,7 @@ bool CMux::data_available(uint8_t *data, size_t len)
         } else {
             return false;
         }
-    } else if (data == nullptr && type == (FT_UA | PF) && len == 0) { // notify the initial SABM command
+    } else if (data == nullptr && (type == (FT_UA | PF) || type == 0x3f) && len == 0) { // notify the initial SABM command
         Scoped<Lock> l(lock);
         sabm_ack = dlci;
     } else if (data == nullptr && dlci > 0) {
```

(and could you please share your results as text only? it's easier for me to go through, copy-paste and is just smaller)

Noushadalik, 2024-03-19:
Hi@david-cermak 

Here is the test outcome

[cmux_text.txt](https://github.com/espressif/esp-protocols/files/14648619/cmux_text.txt)

david-cermak, 2024-03-19:
Oh, very nice, looks like success already (with a few warnings which i'll fix real quick)
just give me few day to process the changes and come up with a PR, then I'll ask you to test it again.
Thanks for your help!

Noushadalik, 2024-03-19:
Thank you for the update! 
Take your time to fix the warnings and prepare the PR. Let me know when you're ready, and I'll test it again. 
Your efforts are appreciated!

david-cermak, 2024-03-26:
Hi @Noushadalik 

Sorry it takes so long, but it seems like the changes we added to make it working on the CAVLI device are not compliant with cmux/gsm0710 standard.
The app note, you linked above, says the CMUX mode complies to `3GPP TS 27.010`, but this spec clearly states (on the SABM response, paragraph `5.4.1`) that we should get an `UA` frame upon success (or `DM` frame if the counterpart isn't able to establish the channel). But we're getting `0x3F`, which is neither of them. And the same frame type is also mentioned in their app note, chapter `3.1`, step `3`:
```
F9 07 3F 01 DE F9
```
(so it's not only an issue of your device or a wrong firmware).

I'm not sure what we should do in this case. I'll add this info to the list of "known issues" for a start, it already mentions some limitations, deviations and patches.

https://github.com/espressif/esp-protocols/blob/14763888ed5501414771e788e8d50cab0788cf93/docs/esp_modem/en/README.rst?plain=1#L212-L236

Noushadalik, 2024-05-03:
Hi @david-cermak 
can I use the esp_modem component as a component of the esp idf framework on PlatformIO?

Noushadalik, 2024-05-06:
Hi @david-cermak 
Any update today?

david-cermak, 2024-05-06:
I'm not very familiar with PlatformIO, but it should be possible based on their docs: https://docs.platformio.org/en/latest/frameworks/espidf.html#esp-idf-components

closed 2024-11-11 by david-cermak