espressif/esp-protocols #508: IPV6 for PDP Context

yanmartins, 2024-02-08:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I use the esp-modem library to create a class for a generic DCE modem.

The function present in `esp_modem_command_library` uses the global PDP context.

```c++
command_result set_pdp_context(CommandableIf *t, PdpContext &pdp, uint32_t timeout_ms)
{
     ESP_LOGV(TAG, "%s", __func__ );
     std::string pdp_command = "AT+CGDCONT=" + std::to_string(pdp.context_id) +
                               ",\"" + pdp.protocol_type + "\",\"" + pdp.apn + "\"\r";
     return generic_command_common(t, pdp_command, timeout_ms);
}
```

My generic class also has an implementation of the `CGDCONT` command, but at some point, the above function is invoked and uses the global pdp struct from the ESP library.

Is there any way for me to set this PDP value? Because the `protocol_type` value is fixed to `"IP"` within the struct. And I want to change it to support IPV6.

Also, is there any other configuration I should do to accept IPV6?

Thanks

david-cermak, 2024-02-09:
Hi @yanmartins 

Yes, this API uses pre-configured PDP context. To update it, you can call `configure_pdp_context()` instead:
```cpp
            auto new_pdp = std::make_unique<PdpContext>("APN");
            new_pdp->protocol_type = "IP6";
            dce->get_module()->configure_pdp_context(std::move(new_pdp));
```
Unfortunately, we don't have an explicit constructor with the protocol, so you'll to update it after creating.

To enable IPv6 protocol in PPP session, you'd have enable: `CONFIG_LWIP_PPP_ENABLE_IPV6` (off by default).

david-cermak, 2024-03-11:
@yanmartins I think this can be closed (please reopen if any other question or request)

closed 2024-03-11 by david-cermak