espressif/esp-protocols #512: [BUG]

filzek, 2024-02-23:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### What component are you using? If you choose Other, provide details in More Information.

esp_websocket_client

### component version

1.2.2

### IDF version.

SDK V5.2 Release

### More Information.

WEBSOCKET do not connect to any websocket and give the error as below, no matter use WS or WSS the problem is the same

I (380) cpu_start: Multicore app
I (389) cpu_start: Pro cpu start user code
I (389) cpu_start: cpu freq: 160000000 Hz
I (390) cpu_start: Application information:
I (393) cpu_start: Project name:     websocket_example
I (398) cpu_start: App version:      ad31043-dirty
I (404) cpu_start: Compile time:     Feb 23 2024 13:53:08
I (410) cpu_start: ELF file SHA256:  49b9d2b32...
I (415) cpu_start: ESP-IDF:          v5.2-166-ge0944287f3-dirty
I (422) cpu_start: Min chip rev:     v0.0
I (427) cpu_start: Max chip rev:     v3.99
I (431) cpu_start: Chip rev:         v3.0
I (436) heap_init: Initializing. RAM available for dynamic allocation:
I (444) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (449) heap_init: At 3FFB84B0 len 00027B50 (158 KiB): DRAM
I (456) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (462) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (468) heap_init: At 40096C70 len 00009390 (36 KiB): IRAM
I (476) spi_flash: detected chip: gd
I (479) spi_flash: flash io: dio
W (483) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (491) nvs_sec_provider: NVS Encryption - Registering Flash encryption-based scheme...
I (501) main_task: Started on CPU0
I (511) main_task: Calling app_main()
I (511) websocket: [APP] Startup..
I (511) websocket: [APP] Free memory: 274564 bytes
I (511) websocket: [APP] IDF version: v5.2-166-ge0944287f3-dirty
I (531) nvs: NVS partition "nvs" is encrypted.
I (541) example_connect: Start example_connect.
I (551) wifi:wifi driver task: 3ffc0bec, prio:23, stack:6656, core=0
I (551) wifi:wifi firmware version: cc1dd81
I (551) wifi:wifi certification version: v7.0
I (551) wifi:config NVS flash: enabled
I (551) wifi:config nano formating: disabled
I (561) wifi:Init data frame dynamic rx buffer num: 32
I (561) wifi:Init static rx mgmt buffer num: 5
I (571) wifi:Init management short buffer num: 32
I (571) wifi:Init dynamic tx buffer num: 32
I (571) wifi:Init static rx buffer size: 1600
I (581) wifi:Init static rx buffer num: 10
I (581) wifi:Init dynamic rx buffer num: 32
I (591) wifi_init: rx ba win: 6
I (591) wifi_init: tcpip mbox: 32
I (591) wifi_init: udp mbox: 6
I (601) wifi_init: tcp mbox: 6
I (601) wifi_init: tcp tx win: 5760
I (611) wifi_init: tcp rx win: 5760
I (611) wifi_init: tcp mss: 1440
[... 140 lines cut ...]
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: BeQJhnRD1SdNZSdIulKZaw==

E (77761) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=0, errno=119
I (77771) websocket: WEBSOCKET_EVENT_ERROR
E (77781) websocket: Last error reported from tls stack: 0x3ffc9a10
E (77791) websocket: Last error captured as transport's socket errno: 0x68
I (77791) websocket_client: Reconnect after 10000 ms
I (77801) websocket: WEBSOCKET_EVENT_DISCONNECTED
E (87881) transport_base: poll_read select error 104, errno = Connection reset by peer, fd = 54
E (87881) transport_ws: Error read response for Upgrade header GET / HTTP/1.1
Connection: Upgrade
Host: echo.websocket.org:80
User-Agent: ESP32 Websocket Client
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: YmeJ9nJx2mTys7IhtC50Yw==

E (87901) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=0, errno=119
I (87921) websocket: WEBSOCKET_EVENT_ERROR
E (87921) websocket: Last error reported from tls stack: 0x3ffc9a10
E (87931) websocket: Last error captured as transport's socket errno: 0x68
I (87941) websocket_client: Reconnect after 10000 ms
I (87941) websocket: WEBSOCKET_EVENT_DISCONNECTED
I BOD: Brownout detector was triggered

david-cermak, 2024-02-28:
Hi @filzek 

I think they just discontinues the `echo.websocket.org` endpoint over plain text websockets.
you can switch to `wss://echo.websocket.org`

david-cermak, 2024-03-18:
Closing as the problem is not on the client side (feel free to reopen and comment further)...

closed 2024-03-18 by david-cermak