espressif/esp-protocols #539: little help in getting esp_modem and LiliGo-TSIM7080-S3 to work

dannybackx, 2024-03-25:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have two problems really. The examples with my board are all Arduino-based, I'm used to working with esp-idf.
Also the examples all have some code repeatedly triggering the power to the modem, like this : 
```
    while (!modem.testAT(1000)) {
[...]
            // Pull down PWRKEY for more than 1 second according to manual requirements
            digitalWrite(BOARD_MODEM_PWR_PIN, LOW);
            delay(100);
            digitalWrite(BOARD_MODEM_PWR_PIN, HIGH);
            delay(1000);
            digitalWrite(BOARD_MODEM_PWR_PIN, LOW);
```
When I try that, the modem is off, so I turned the HIGH and LOW around. Not sure why.

Anyway, none of the esp_modem examples do such thing.
So should I do this prior to enabling esp_modem ?

If yes, it looks like I need the uart module to do this (send AT commands and detect "OK" coming back to me). Right ? And should I close down the uart module after that and before starting esp_modem ?

Thanks, I appear to need a little help getting over this initial hurdle.

dannybackx, 2024-03-25:
Maybe some evidence of where I am ..
```
I (00:02:16.807) AXP2101: Implemented using built-in read and write methods (Use higher version >= 5.0 API)
I (00:02:16.818) AXP2101: Init PMU SUCCESS!
I (00:02:16.827) AXP2101: DCDC=======================================================================
I (00:02:16.833) AXP2101: DC1  : +   Voltage:3300 mV
I (00:02:16.838) AXP2101: DC3  : +   Voltage:3000 mV (modem)
I (00:02:16.845) AXP2101: ALDO=======================================================================
I (00:02:16.855) AXP2101: BLDO=======================================================================
I (00:02:16.864) AXP2101: BLDO1: +   Voltage:3300 mV (level converter)
I (00:02:16.872) AXP2101: CPUSLDO====================================================================
I (00:02:16.881) AXP2101: DLDO=======================================================================
I (00:02:16.892) AXP2101: ===========================================================================
I (00:02:16.907) gpio: GPIO[41]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
V (1054) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1064) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1074) intr_alloc: Connected src 29 to int 17 (cpu 0)
I (00:02:16.931) pppos_example: Toggling ..
I (00:02:18.133) pppos_example: Done
I (00:02:18.153) pppos_example: Toggling ..
I (00:02:19.353) pppos_example: Done
I (00:02:19.373) pppos_example: Recv str: AT

D (00:02:19.373) esp_netif_lwip: check: remote, if=0x3fca3280 fn=0x420202c0
0x420202c0: esp_netif_new_api at /home/danny/src/github/esp32/esp-idf-v5.2.1/components/esp_netif/lwip/esp_netif_lwip.c:690

V (00:02:19.374) esp_netif_objects: esp_netif_add_to_list_unsafe 0x3fca7b74
D (00:02:19.381) esp_netif_objects: esp_netif_add_to_list_unsafe netif added successfully (total netifs: 1)
D (00:02:19.391) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3fca7da4
D (00:02:19.399) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (00:02:19.406) pppos_example: Initializing esp_modem for the SIM7070 module...
I (00:02:19.415) uart: queue free spaces: 30
V (3564) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (3574) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (3584) intr_alloc: Connected src 28 to int 18 (cpu 0)
V (00:02:19.439) command_lib: set_flow_control
V (00:02:19.443) command_lib: generic_command_common
V (00:02:19.448) command_lib: generic_command
D (00:02:19.452) command_lib: generic_command command AT+IFC=2,2

E (00:02:19.953) pppos_example: Failed to set the set_flow_control mode
E (00:02:19.954) pppos_example: Error 263 at line 238, ESP_ERR_TIMEOUT
V (00:02:19.957) command_lib: read_pin
V (00:02:19.961) command_lib: generic_get_string
V (00:02:20.463) command_lib: get_signal_quality
V (00:02:20.463) command_lib: generic_get_string
E (00:02:20.963) pppos_example: esp_modem_get_signal_quality failed with 263 ESP_ERR_TIMEOUT
I (00:02:20.964) main_task: Returned from app_main()
```

david-cermak, 2024-03-26:
> When I try that, the modem is off, so I turned the HIGH and LOW around. Not sure why.

Probably a reverted login in the power-up sequence of your device or simply a transistor on your EVB inverting the signal?

> Anyway, none of the esp_modem examples do such thing.
> So should I do this prior to enabling esp_modem ?

You don't have to, unless you know which state your device really is upon starting your program. (In the real world application, you'd probably need to do something similar, as you'd probably only power the modem up only if needed or vice-versa -- the modem would stay up in some low power mode and would wake your ESP device, but that's very application specific)

> If yes, it looks like I need the uart module to do this (send AT commands and detect "OK" coming back to me). Right ?

If you're using `esp_modem` you don't have to run these commands on uart separately, you can send the sync command from esp_modem, like `dce->sync()` or `esp_modem_sync()`.

> E (00:02:19.953) pppos_example: Failed to set the set_flow_control mode
> E (00:02:19.954) pppos_example: Error 263 at line 238, ESP_ERR_TIMEOUT

This indicates that you cannot setup the desired control flow. Are you using modem with UART flow control modem signal?
There could also be an issue with wiring or you might need a delay after the initial power-up sequence.
Or you can run a while loop until the sync succeeds, like:
```cpp
void wait_for_sync(esp_modem_dce_t *dce, uint8_t count)
{
    for (int i = 0; i < count; i++) {
        if (esp_modem_sync(dce) == ESP_OK) {
            return ESP_OK;
        }
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
    return ESP_FAIL
}
```

dannybackx, 2024-03-30:
Thanks, you got me a step further :-)

closed 2024-03-30 by dannybackx