espressif/esp-protocols #551: esp-netif_lwip-ppp: Connection lost, Modem Disconnect from PPP Server

afayzullaev, 2024-04-15:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm getting esp-netif_lwip-ppp: Connection lost, Modem Disconnect from PPP Server error on A9G module. It Responses to AT commands, but when I switch it to data mode it can't get IP. Please give me advice on how to solve this problem.
Thank you!

david-cermak, 2024-04-16:
It could be either poor signal, or wrong configuration, per your operator's settings. If the connection never succeeded, I'd guess maybe an incorrect APN?

afayzullaev, 2024-04-20:
> It could be either poor signal, or wrong configuration, per your operator's settings. If the connection never succeeded, I'd guess maybe an incorrect APN?

With AT commands it works properly. I've got IP address with AT+CIFSR command. Below the code which works:

afayzullaev, 2024-04-20:
With AT commands it works properly. I've got IP address with AT+CIFSR command and also I succesfully sent sms with at commands. Below the code which works:
```
    esp_modem_PdpContext_t pdp_context = {
        .apn = APN,
        .context_id = 1,
        .protocol_type = "IP"
    };
    esp_modem_set_pdp_context(dce,&pdp_context);
    vTaskDelay(pdMS_TO_TICKS(1000));

    esp_modem_set_network_attachment_state(dce,1);
    vTaskDelay(pdMS_TO_TICKS(1000));
    

    esp_modem_at(dce,"AT+CGACT=1,1", at_data, 2000);
    ESP_LOGI(TAG, "AT+CGACT %s", at_data);
    memset(at_data, 0, sizeof(at_data));
    vTaskDelay(pdMS_TO_TICKS(1000));

    esp_modem_at(dce,"AT+CIFSR", at_data, 2000);
    ESP_LOGI(TAG, "AT+CIFSR %s", at_data);
    memset(at_data, 0, sizeof(at_data));
    vTaskDelay(pdMS_TO_TICKS(1000));
```
But with this code it can't be set to data mode. If I remove it the modem switches to data mode succesfully, but I got this log:

> V (37142) command_lib: set_echo
> V (37142) command_lib: generic_command_common
> V (37142) command_lib: generic_command
> D (37142) command_lib: generic_command command ATE0
> 
> D (37172) command_lib: Response: 
> OK
> 
> 
> V (37172) command_lib: set_pdp_context
> V (37172) command_lib: generic_command_common
> V (37182) command_lib: generic_command
> D (37182) command_lib: generic_command command AT+CGDCONT=1,"IP","internet.beeline.uz"
> 
> D (37242) command_lib: Response: 
> OK
> 
> 
> V (37242) command_lib: set_data_mode
> V (37242) command_lib: generic_command
> D (37242) command_lib: generic_command command ATD*99***1#
> 
> D (37272) command_lib: Response: 
> CONNECT
> 
> 
> D (37272) esp_netif_handlers: esp_netif action has started with netif0x3ffc036c from event_id=0
> D (37282) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3ffc067c
> D (37292) esp_netif_lwip: check: remote, if=0x3ffc036c fn=0x40102ca8
> 0x40102ca8: esp_netif_update_default_netif_lwip at F:/espressif/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:327
> 
> 
> D (37292) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffc036c
> V (37302) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffc036c
> D (37302) esp_netif_lwip: call api in lwip: ret=0x0, give sem
> I (37312) INTERNET: DATA MODE is set
> I (37312) INTERNET: Waiting for IP address
> I (43342) esp-netif_lwip-ppp: Connection lost
> D (43342) event: running post IP_EVENT:7 with handler 0x400d82b8 and context 0x3ffc0318 on loop 0x3ffbf3cc
> 0x400d82b8: on_ip_event at F:/Projects/esp32/radarsensor_a9g/main/libs/src/internet.c:73
> 
> D (43342) INTERNET: IP event! 7
> I (43342) INTERNET: Modem Disconnect from PPP Server
> D (43352) event: running post IP_EVENT:7 with handler 0x401016b4 and context 0x3ffc5cf8 on loop 0x3ffbf3cc
> 0x401016b4: esp_netif_action_disconnected at F:/espressif/esp-idf/components/esp_netif/esp_netif_handlers.c:83
> 
> D (43362) esp_netif_handlers: esp_netif action disconnected with netif0x3ffc036c from event_id=7
> D (43372) esp_netif_lwip: check: remote, if=0x3ffc036c fn=0x4010263c
> 0x4010263c: esp_netif_down_api at F:/espressif/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1638
> 
> 
> D (43372) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffc036c
> D (43382) esp_netif_lwip: esp_netif_internal_dhcpc_cb lwip-netif:0x3ffc03f0
> D (43392) esp_netif_lwip: check: local, if=0x3ffc036c fn=0x40102ca8
> 0x40102ca8: esp_netif_update_default_netif_lwip at F:/espressif/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:327
> 
> 
> D (43392) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffc036c
> V (43402) esp_netif_objects: esp_netif_next_unsafe 0x0
> V (43402) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffc036c
> V (43412) esp_netif_objects: esp_netif_next_unsafe 0x3ffc036c
> D (43422) esp_netif_lwip: call api in lwip: ret=0x0, give sem

david-cermak, 2024-04-23:
Maybe the switching between profiles via `AT+CGACT=1,1` is causing issues, as just before establishing the PPP mode, we set the configured PDP context again. Could you please check if that the case by supplying the desired PDP via an at command and overriding the default one?

afayzullaev, 2024-04-30:
I've left the code below:
```
esp_modem_PdpContext_t pdp_context = {
        .apn = APN,
        .context_id = 1,
        .protocol_type = "IP"
    };
    esp_modem_set_pdp_context(dce,&pdp_context);
    vTaskDelay(pdMS_TO_TICKS(1000));

    esp_modem_set_network_attachment_state(dce,1);
    vTaskDelay(pdMS_TO_TICKS(1000));
    

    esp_modem_at(dce,"AT+CGACT=1,1", at_data, 2000);
    ESP_LOGI(TAG, "AT+CGACT %s", at_data);
    memset(at_data, 0, sizeof(at_data));
    vTaskDelay(pdMS_TO_TICKS(1000));

    esp_modem_at(dce,"AT+CIFSR", at_data, 2000);
    ESP_LOGI(TAG, "AT+CIFSR %s", at_data);
    memset(at_data, 0, sizeof(at_data));
    vTaskDelay(pdMS_TO_TICKS(1000)); 
   
    err = esp_modem_set_mode(dce, ESP_MODEM_MODE_DATA);
```
And overrided this one(just commented set_pdp_context function):
```
    /**
     * @brief This is a mandatory method for ModuleIf class, which sets up the device
     * to be able to connect to the network. This typically consists of setting basic
     * communication parameters and setting the PDP (defining logical access point
     * to cellular network)
     */
    bool setup_data_mode() override
    {
        if (set_echo(false) != command_result::OK) {
            return false;
        }
        // if (set_pdp_context(*pdp) != command_result::OK) {
        //     return false;
        // }
        return true;
    }
```
I'm getting the following logs:

> V (25453) command_lib: set_pdp_context
> V (25453) command_lib: generic_command_common
> V (25453) command_lib: generic_command
> D (25453) command_lib: generic_command command AT+CGDCONT=1,"IP","internet.beeline.uz"
> 
> D (25563) command_lib: Response: 
> OK
> 
> 
> V (26563) command_lib: set_network_attachment_state
> V (26563) command_lib: generic_command_common
> V (26563) command_lib: generic_command
> D (26563) command_lib: generic_command command AT+CGATT=1
> 
> D (26593) command_lib: Response: 
> +CGATT:1
> 
> OK
> 
> 
> V (27593) command_lib: at
> V (27593) command_lib: generic_get_string
> I (29593) INTERNET: AT+CGACT 
> V (30593) command_lib: at
> V (30593) command_lib: generic_get_string
> } (31493) command_lib: Token: {
> 
> V (31493) command_lib: Token: {OK}
> 
> I (31493) INTERNET: AT+CIFSR
> V (32493) command_lib: set_echo
> V (32493) command_lib: generic_command_common
> V (32493) command_lib: generic_command
> D (32493) command_lib: generic_command command ATE0
> 
> D (32513) command_lib: Response: 
> 10.240.156.162
> 
> OK
> 
> 
> V (32513) command_lib: set_data_mode
> V (32513) command_lib: generic_command
> D (32513) command_lib: generic_command command ATD*99***1#
> 
> D (32523) command_lib: Response:
> OK
> 
> 
> D (32543) command_lib: Response: 
> OK
> 
> CONNECT
> 
> 
> D (32543) esp_netif_handlers: esp_netif action has started with netif0x3ffbfaa8 from event_id=0
> D (32543) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3ffbfdb8
> D (32553) esp_netif_lwip: check: remote, if=0x3ffbfaa8 fn=0x401019a0
> 0x401019a0: esp_netif_update_default_netif_lwip at F:/espressif/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:327
> 
> 
> D (32563) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbfaa8
> V (32573) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbfaa8
> D (32573) esp_netif_lwip: call api in lwip: ret=0x0, give sem
> I (32583) INTERNET: DATA MODE is set
> I (32583) INTERNET: Waiting for IP address
> I (38613) esp-netif_lwip-ppp: Connection lost
> D (38613) event: running post IP_EVENT:7 with handler 0x400d806c and context 0x3ffbfa54 on loop 0x3ffbeb08
> 0x400d806c: on_ip_event at F:/Projects/esp32/radarsensor_a9g/main/libs/src/internet.c:73
> 
> D (38613) INTERNET: IP event! 7
> I (38613) INTERNET: Modem Disconnect from PPP Server
> D (38623) event: running post IP_EVENT:7 with handler 0x401003ac and context 0x3ffc5434 on loop 0x3ffbeb08
> 0x401003ac: esp_netif_action_disconnected at F:/espressif/esp-idf/components/esp_netif/esp_netif_handlers.c:83
> 
> D (38633) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbfaa8 from event_id=7
> D (38643) esp_netif_lwip: check: remote, if=0x3ffbfaa8 fn=0x40101334
> 0x40101334: esp_netif_down_api at F:/espressif/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1638
> 
> 
> D (38643) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbfaa8
> D (38653) esp_netif_lwip: esp_netif_internal_dhcpc_cb lwip-netif:0x3ffbfb2c
> D (38663) esp_netif_lwip: check: local, if=0x3ffbfaa8 fn=0x401019a0
> 0x401019a0: esp_netif_update_default_netif_lwip at F:/espressif/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:327

afayzullaev, 2024-04-30:
> the
I've removed AT+CGACT=1,1, but nothing is changed