espressif/esp-protocols #552: Choosing The Right Mode For SIMCOM or Any Modem... (AT or DialUP)

parthbhat13, 2024-04-17:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all, and specifically @david-cermak 

### Just a bit of background
apologies in advance as i do not know, what i am going to produce below in terms of issue or suggestion would fall under this title or not. bare with me as i would be summarizing my issue and work of past 3 years together in some limited words. 

###The Problem Statement 
Basically since past 3 years ive been working and developing some UPI (Unified Payments Interface) system based payment devices for the vending machine, there has been a lot of upgrades and different versions have been developed and now there another final version which i am supposed to be launching. so far ive completed around 1000pcs in the market and we are trying to deploy even more. 
the stack which i am using is as follows 
- ESP32
- GSM(A7672S) 
dialUp connection or PPPos over uart bus. NOTE: I've not used the CTS and RTS lines. 
now in terms of the software I've got two different things working , MQTT is meant for sending and receiving payment transactions and state of the machine. plus we have used HTTP to make sure there is a heartbeat and also for verification of the device. 
during the runtime there is HTTP and MQTT both are technically running 24X7 and every 60Sec there is a HTTP request sent to server more like a patch request and MQTT is just ideal in KEEP-ALIVE of 120Sec. that is 2 mins. 

summary of a problem statement i need a real time communication with a server over GSM.. simple. 

### The Problem I'm facing and reason for this post ?
This will definitely be very long so bare with me please. 
after deploying multiple products in the market we have started receiving multiple complains from the field. mainly where the network drops, or the ping request fails or some other minor things which had been hard for me to figure out since i did not use my enough brains to have the OTA update function and remote LOG monitoring function. 

Taking matters in my own hands i started talking with some of the other people in the market, including my close mentors in india, 
- Mr.Franz from germany 
- Mr. Mahesh AsiaPacific Technical Head of Simcom
- Mr. Rajesh AsiaPacific Director of Simcom
- Mr.Ravi Pujjar Whos a youtuber and have made many great gsm projects. 

as well since it was hard to get in touch with mr.David Cenmark i was lucky enough to watch your youtube video where you mentioned few things which lead me to more confusions. 

to summarize the decisions from all the mentors mentioned above, all i could say is the ones from india asked to go with AT commands and use the internal GSM Http and MQTT stack to talk with the server. where in Mr.Franz and Mr.David asked to go for PPPOS dial Up connection. 

now with the updated and latest product, i need to not only have the payment transactions working, but also OTA update, real time logs and much more other things. where in i spent 2 months to write my own AT commands based library for the GSM which can be used by any modem (that was the plan), i came with some results .

- HTTP GET request to the server with custom AT library takes 3Seconds (Thats too much)
- The dialUp connection and esp32 internal http library takes 1.5Seconds. 

Ofcourse the right answer is to go with the esp-modem library but again cant forget the issues which ive had from the field where they complained of network loss, connection drop, http failed to ping and on and on. 

out of my curiosity, i even tested the Arduino and TinyGsm library, and it seems they are doing all AT commands and its all over the TCP stack of the gsm modem. to my surprise arduino with a vanilla HTTP request over tcp. takes 840mS for response lol. idk what have they used in the tinyGsm library for it to be so fast. all of my stack is developed on esp-idf V4.4.6 and C . 

### Solutions I Can Think Of..
eventually, i am genuinely stuck between what to do? and below are my options 
1. add CTS and RTS lines, Use the PPPOS dialUp connection, and use only single protocol MQTT/HTTP not both 
2. add custom code on top of esp-modem library to handle the GSM HTTP/MQTT Stack AT commands.
3. Try and find ways to implement the TinyGsm based communication DTE driver for the GSM for faster Result in reading and writing to and from uart.

with the above solutions, i can add the cts and rts to my existing payment device since i have spare pins, but my other products which involves Tea Coffee Vending Machine and Even Vending Machine controller, has no ESP32 pins left whatsoever. so idk how would that be of any help. yes i can shift to esp32S3 and use the USB. but now there no time for redeveloping the whole pcbs and sample it and test all the other loads of protocols which are running on the board.. 

so what shall be done and are there any inputs ?

### My Custom AT library At parser code.. 

1. Function To Send AT Command
```
gsm_err_t DTECommand(gsmConfig_t *config, char *at)
{
    char sendBuf[strlen(at) + 3];

    vTaskDelay(1 / portTICK_PERIOD_MS);
    uart_flush(config->gsmUartConfig.uartNum);

    memset(sendBuf, 0, sizeof(sendBuf));
    memcpy(sendBuf, at, strlen(at));
[... 160 lines cut ...]
            ret = GSM_OK;
            break;
        }
        else if(noAsciiFlag == 1)
        {
           
            memset(buffer, '\0', bufferSize);
            ret = GSM_ERROR;
            break;
        }
    }

    if (config->debugDTE)
    {
        ESP_LOGI(TAG, "Response: %s", buffer);
        ESP_LOG_BUFFER_HEX(TAG, buffer, strlen(buffer) + 1);
    }

    return ret;
}
```

i am not at all a good programmer so im sorry if its too much of stupid things i must've done. 

thank you very much if you have read all of the above. thanks a lot.

david-cermak, 2024-04-18:
Hi @parthbhat13 

I'm afraid I cannot give a general suggestion about the "right" mode, and I'd just repeat myself on what I've already said in that devcon talk: "depends on your application".
You've been talking about your customer's complaints on unstable connectivity, and also a bit about your application needs -- occasional HTTP request and MQTT  communication. So I'd assume the number one priority of your project is the **fast and fail save mechanism and recovery** on signal loss while the communication is not so frequent and kind of slow. In this case, I'd agree that using the AT commands only is probably a better choice (just mind the disadvantages, like security and portability)

Unfortunately we don't have any example which deals with connectivity losses and recoveries, as it's usually device and application dependent. I'll think about it -- my general suggestion in this scenario would be the CMUX manual mode or the USB dual channel mode.
Here's a simple example which demonstrates these retries and recoveries in the PPPoS mode:

https://github.com/espressif/esp-protocols/blob/bd6e12050928464b4357015c7c8c1b215432d580/examples/esp_netif/multiple_netifs/main/ppp_connect_esp_modem.c#L54-L93

you can also check this file, which does the same thing directly, without `esp_modem`

https://github.com/espressif/esp-protocols/blob/bd6e12050928464b4357015c7c8c1b215432d580/examples/esp_netif/multiple_netifs/main/ppp_connect_simple.c#L71-L95

parthbhat13, 2024-04-18:
Hi @david-cermak,
Thanks a lot for your reply, I'd like to add few more to what you've mentioned. 

1. Fail safe is important yes! 
While you mentioned having fail safe as important aspect, at the same time, it's not needed to have two communication protocols hence it was decided to choose only one protocol either Mqtt or http. 

2. You mentioned to use cmux over usb.
The biggest concern as of now which you can already see I've mentioned, I cannot change the hardware now, neither can I add esp32 S2 or S3. So USB goes out of question. 
In the Devcon you had even mentioned about using the RTS and CTS lines ? Do you think it's important for me for the Mqtt or HTTP And ota updates ? 

3. Yes going vanilla AT command sounds good to me to that now you have mentioned. But what DTE driver code would you recommend ? To use the esp-modems C function which allows me to send at commands ? Or make my own? Which I already have made and shown the code above which doesn't really work that fluent at times. ? 

Thanks in advance !

david-cermak, 2024-04-19:
1) Add fail safe: Yes, it's easier to cover that in the AT mode only, since you'd just get an `ERROR` or `DISCONNECTED` reply to your commands, while in the data mode you'd have to take care of different phases of PPP state machine and some devices just exit this data mode.

2) You can also use CMUX over UART, no need for USB. As for the UART flowcontrol signals, there're useful and prevent from buffer overflows, but you can easily tune up your application without them -- set the appropriate baud rate, move the UART ISR to RAM, increase Rx buffer sizes -- and according to my tests, the OTA should work seamlessly.

3) The problem is that you would really have to implement everything on your own, including OTA, and communication.
You can  check this file https://github.com/espressif/esp-protocols/blob/master/examples/esp_netif/multiple_netifs/main/ppp_connect_simple.c which uses UART driver directly, but with queues and interrupts, so should be faster than your code.
You can also use `esp_modem` with AT commands only, maybe just the DTE class and the command library. I'd suggest using C++ in this case, as it's easily extensible.

parthbhat13, 2024-04-19:
thanks for the response mr.david.

so let me summarize the things i shall test.
1. No need to change the hardware to use the CTS and RTS lines 
2. Increase the uart buffer (i already did that i guess) and move uart isr to ram 
3. i can choose to go with esp_modem dte class for my communication or i will have to find ways to port the TinyGsm library way of communication. 
4. i wonder what to do for OTA? use the TCP/IP custom code or get the Dialup?

what i can see from above list of things which i would do is that, using C++ is what i was avoiding at time. but i wonder how much would it change? since the esp's uart driver is written in C? only some string manipulation would end up being helpful if im using c++. 

let me get back to you with all the results. as i can see i will have to do multiple tests. i even had decided to work with CMUX mode where we can simply use the Data as well as AT commands to check for the network stability. 

honestly i cant yet come to any conclusion on what method to use. so if you have any more ideas please do share. thanks in advance.

david-cermak, 2024-04-19:
just a few comments:
* It's much easier to use the dialup connection for OTA, as everything is ready for you to use (vs. implementing OTA using AT commands)
* suggested C++ only if you decide to use `esp_modem`'s DTE layers, as they're written in C++ so you can leverage some of the generic methods and templates (in case of directly interfacing the uart, there's no need nor benefit in using either C or C++).
* note about connectivity: I'd also suggest looking into the actual issues with unstable connections and your customers problems which you mentioned earlier to understand why they were happening, i.e. to identify the reasons and root causes of these issues. You might as well re-work and re-implement everything using another mode, but end up receiving the same complaints again, since the GSM/LTE/NB connectivity in the field is simply unstable by nature...

parthbhat13, 2024-04-20:
Thanks for the response, I'm gonna take 4-5 different production boards and test all the different scenarios. 
I recently was with a professor of mine he was working on php and a commercial project which used a7672S and stm32, I was amazed to see how quick the http post request response was being received. Many pointed out to me that apart from the network drop in my product, there might be the server issues as well. So will even test it out and get back to you with results so it helps others as well. 

One small help I'd ask for, can you point me to the specific function in esp-modem which waits for the AT response ? 

I'd like to wait for gsm to get stable once I toggle the power pin, and wait to receive its own Startup at messages like 

- PB Done 
- EPS DONE 
- SIM READY 

And only then I'd send the At commands to the gsm. I saw there's a function to send at command and there's a timeout mentioned, but I'd wanna simply wait for a valid data to be received. 

Thanks in advance.

david-cermak, 2024-04-23:
You can either send a null command or use `dce->command()` API or even supply your own callback in `got_line` argument.

parthbhat13, 2024-04-23:
hi, 
thanks for the response, since i am not fluent with C++ there are issues i am facing trying to make things working. i was even stuck with initializing the modem where i did not wanna pass the apn and init the 
```
esp_modem_new_dev
```

perhaps so far this is what i have done and ported some of the espmodem function to work with mine, below are the code snippets. 

```
esp_err_t initGsmUartAndPower(gsmConfig_t *config)
{
    esp_err_t ret;

    // setup the esp modem stack and data 
    ESP_LOGI(TAG, "Setting Up GSM Modem");
    esp_modem_dte_config_t dteConfig = ESP_MODEM_DTE_DEFAULT_CONFIG();

    // Setup Uart Specific Configuration
    dteConfig.uart_config.tx_io_num = config->gsmUartConfig.txPin;
    dteConfig.uart_config.rx_io_num = config->gsmUartConfig.rxPin;
    dteConfig.uart_config.rts_io_num = UART_PIN_NO_CHANGE;
    dteConfig.uart_config.cts_io_num = UART_PIN_NO_CHANGE;

    dteConfig.uart_config.rx_buffer_size = config->gsmUartConfig.rxBufferSize;
    dteConfig.uart_config.tx_buffer_size = config->gsmUartConfig.txBufferSize;
    dteConfig.uart_config.event_queue_size = config->gsmModemTaskConfig.eventQueueSize;
    dteConfig.task_stack_size = config->gsmModemTaskConfig.taskSize;
    dteConfig.task_priority = config->gsmModemTaskConfig.taskPriority;
    dteConfig.dte_buffer_size = config->gsmUartConfig.dteBufferSize;

    // configure the dce 
    // here we can later call the command to set the APN in the library 
    // esp_err_t esp_modem_set_apn(esp_modem_dce_t *dce, const char *apn);
    dceConfig = ESP_MODEM_DCE_DEFAULT_CONFIG("airtelgprs.com");

    // Configure the PPP netif
    esp_netif_config_t netifPPPConfig = ESP_NETIF_DEFAULT_PPP();

    
    // Configure Netif Object
    if (espModemNetif == NULL)
    {
        espModemNetif = esp_netif_new(&netifPPPConfig);
        assert(espModemNetif);
    }

    if (dceConfig == NULL)
    {
        dce = esp_modem_new_dev(ESP_MODEM_DCE_SIM7600, &dteConfig, &dceConfig, espModemNetif);
        assert(dce);
    }

    // setup Power
    ret = gpio_set_direction(config->gsmUartConfig.powerPin, GPIO_MODE_OUTPUT);
    ret = gpio_set_pull_mode(config->gsmUartConfig.powerPin, GPIO_PULLUP_DISABLE);
    ret = gpio_set_pull_mode(config->gsmUartConfig.powerPin, GPIO_PULLDOWN_DISABLE);
    gpio_set_level(config->gsmUartConfig.powerPin, 0);

    return ret;
}
```

and even the wrapper function which i have called in all the other files, this is what i have done 

```
gsm_err_t DTESendCommandGetResponse(gsmConfig_t *config, char *atCommand, char *atResponse, char *atResponse2, int timeout)
{
    // TODO: well try to keep on resending the data maybe?
    gsm_err_t gsmErr;
    esp_err_t espErr;
    
    char bufData[config->gsmUartConfig.rxBufferSize];
    memset(bufData, '\0', sizeof(bufData));
    
    espErr = esp_modem_at(dce, atCommand, &bufData[0], timeout);

    if (espErr == ESP_OK)
    {

        // we can log the data here
        if (config->debugDTE)
        {
            ESP_LOGI(TAG, "Response: %s", bufData);
            ESP_LOG_BUFFER_HEX(TAG, bufData, strlen(bufData) + 1);
        }
        

        if (atResponse != NULL && bufData != NULL)
        {
            
            if (strstr(bufData, atResponse) != NULL)
            {
                if (atResponse2 != NULL)
                {
                    if (strstr(bufData, atResponse2) == 0)
                        return GSM_OK;
                    else
                        return GSM_ERROR;
                }
                return GSM_OK;
            }
            //TODO: Check if this needs to be fixed
            // this is basically to check if we got an error from the other function 
            // for example during pdp context if we got a known valid response or ERROR, then we should return erro r
            else if (atResponse2 != NULL && bufData != NULL)
            {
                if (strstr(bufData, atResponse2) != NULL)
                {
                    // again ideally we would reach to the else loop but i am confused with this part.
                    return GSM_ERROR;
                }
            }
            else
            {
                // ideally we should send this back to the URC
                if (strstr(bufData, "ERROR") != NULL)
                {
                    if (config->debugDTE)
                        ESP_LOGE(TAG, "Got Error URC Should Be called");
                    return GSM_ERROR;
                }
            }
        }
        else
        {
            ESP_LOGE(TAG, "Data did not receive correctly");
            return GSM_ERROR;
        }
    }
    else if (espErr == ESP_FAIL)
    {
        return GSM_ERROR;
    }
    else if (espErr == ESP_ERR_TIMEOUT)
    {
        return GSM_TIMEOUT;
    }
    else 
    {
        return GSM_ERROR;
    }

    return GSM_ERROR;
}
```

apologies if my code and concepts are too noob. trying to get my head around things after spending 10months on this project.

parthbhat13, 2024-04-30:
> You can either send a null command or use `dce->command()` API or even supply your own callback in `got_line` argument.

Hi David,

Can you please point me to more right direction on how can I possibly implement it? 
I'm too much confused with how can I use those above mentioned functions.

parthbhat13, 2024-04-30:
Hi, 
Quick update, One of the problem which was mentioned in terms of the HTTP request time used to be 3Seconds, Figured out that if we use AT commands and use the HTTP Stack of the gsm and make HTTPS request it is around 3~4Seconds. 
If we use the HTTP url instead of the HTTPS, then we end up much more faster and around 300mS of the time. 
Same with the DialUp with HTTPS its 1~2Seconds and with HTTP its 100~200mS. so one thing is clear over here that it is better to use HTTP rather than HTTPS. 

what i am planning now is to add a feature and modem of A7672S like how Mr.Franz have added for the gnss, and push this on the esp_modem. i went through all the old issues trying to figure out how we can add custom modem and add more features like using the internal HTTP stack of the gsm using AT commands. i could not find a good documentation for the same, would simply keep going through the issues and see how others have added features. if there is any other better documentation which you can walk me through it will be really helpful. 

will keep updating over here so others might as well get better idea on how and what stack to use when, specially in the indian market. 

thanks again.

parthbhat13, 2024-06-10:
hi @david-cermak ,

sorry for the late update, i was trying out multiple solutions to my problem. 
i decided to go with the AT commands rather than dial-up for the http requests to the server but it seems due to the server being secure and the TLS certs it takes too much time or sometimes misses the response from the server. i did try to connect with the Simcom people and they gave me response to use the SSLG at commands which i was not aware of and was not sure how to use.

i decided to go with dialup and cmux mode so i can keep checking the RSSI on runtime. i did apply  the 
below option to disable 
```
ESP_MODEM_CMUX_DEFRAGMENT_PAYLOAD
```

but it seems that the modem is not exiting the CMUX mode correctly. 
moreover i have even added the uart in the IRAM if that helps but still no use. i ran the modem console example which i have noticed you asking to use often in all the other issues ive read. below is the log of the same if you can take a closer look. 

i am using the SIMCOM-A7672S, i could not even see the option in the modem list to explicitly choose SIMCOM-A7672S, all i could see was SIM7600 so, maybe that can be the issue? i dont know. 
eventually i even decided to go with the CAVLI-C16QS but upon reading the following thread 

https://github.com/espressif/esp-protocols/issues/507

it seems that even the CAVLI modem is not really supported with the library. so it would be really helpful if you can help with something on what is happening over here. 

```
esp> set_mode CMUX
I (22711) modem_console: Switching to CMUX name...
I (22951) modem_console: OK
esp> I (23001) esp-netif_lwip-ppp: Connected
I (23011) esp-netif_lwip-ppp: Name Server1: 10.174.107.166
I (23011) esp-netif_lwip-ppp: Name Server2: 8.8.8.8
esp> get_operator_name
I (32481) modem_console: Reading operator name...
I (32491) modem_console: OK. Operator name: 40405
esp> httpget
I (36691) modem_console_httpget: HTTP GET Status = 200, content_length = 271
esp> set_mode CMD
I (47011) modem_console: Switching to CMD name...
I (47021) esp-netif_lwip-ppp: User interrupt
I (47021) esp_modem_netif: PPP state changed event 5
W (47051) CMUX: Restarting CMUX state machine (reason: 3)
I (47051) CMUX: Protocol recovered
E (48071) modem_console: Failed to set the desired mode
Command returned non-zero error code: 0x1 (ERROR)
esp> get_operator_name
I (71411) modem_console: Reading operator name...
W (71411) CMUX: Restarting CMUX state machine (reason: 3)
I (71411) CMUX: Protocol recovered
E (146411) modem_console: Failed with TIMEOUT
Command returned non-zero error code: 0x1 (ERROR)
esp> set_mode CMUX
I (158281) modem_console: Switching to CMUX name...
E (158281) modem_console: Failed to set the desired mode
Command returned non-zero error code: 0x1 (ERROR)
esp> reset
I (163031) modem_console: Resetting the module...
W (163041) CMUX: Restarting CMUX state machine (reason: 3)
E (223031) modem_console: Failed with TIMEOUT
Command returned non-zero error code: 0x1 (ERROR)
esp> 
```

thanks in advance.

david-cermak, 2024-06-10:
>  the modem is not exiting the CMUX mode correctly.
> i am using the SIMCOM-A7672S

https://github.com/espressif/esp-protocols/blob/7f248bd03f30d5054ac7a8faa0c2a42cd47413dc/docs/esp_modem/en/README.rst?plain=1#L212-L236

> it seems that even the CAVLI modem is not really supported with the library

https://github.com/espressif/esp-protocols/issues/507#issuecomment-2020854916

---

Hi @parthbhat13 

I have experienced couple of minor deviations in CMUX protocol implementation with some modems. 
For both of the issues above, the deviation seems to be on the modem side -- please see the references to the spec & linked explanations.

TBH, I'm not sure how to handle this in the library; so far, I'm publishing patches in the docs, feeling a bit reluctant to add those directly to the sources, as they break the CMUX spec.

Not a big problem for the SIM-A7672, as exiting CMUX mode is a marginal feature, with no actual usecase for embedded applications. The CAVLI device, however, doesn't even enter the CMUX mode properly and I don't see a good workaround, nor any add-on sources that would *alter* the initialization on the application level. I'd probably also add a patch and attach it to the `esp_modem` docs.

parthbhat13, 2024-06-10:
Hi @david-cermak,
Thanks a lot for the prompt reply, since the documentation specifically mentioned the patch for the A7670C I didn't spend much time trying to implement the patch. I will try to do it tomorrow and update you. 
As far for the 2nd question what modem shall I be selecting when starting the dce? Like mentioned before I've selected the SIM7600, would that be any problem? Since I noticed while running the esp-console example, for reset it sends AT+CRESET rather than the AT+RESET required by the A7672S. 

Does choosing the Sim7600 for dce makes any issues? Thanks.

david-cermak, 2024-06-10:
> Does choosing the Sim7600 for dce makes any issues?

haven't seen any protocol related issues with SIM7600. if there's a problem with specific commands, like missing AT commands, timeouts or the reset command, it could be easily updated on the application layer.

parthbhat13, 2024-06-10:
> > Does choosing the Sim7600 for dce makes any issues?
> 
> 
> 
> haven't seen any protocol related issues with SIM7600. if there's a problem with specific commands, like missing AT commands, timeouts or the reset command, it could be easily updated on the application layer.

Hey David,
I failed to convey the right question here 

```
esp_modem_dce_device
```

This does not have A7672S as an option, so is it okay if we have selected SIM7600 as device and use A7670S ?

sisodiakaran, 2024-07-14:
I have reviewed this thread and found several relevant points, as I am working on a similar project. Hi, I am Karan. I recently started working with ESP32 and Simcom modules, utilizing the ESP_Modem library for PPP connections. @parthbhat13, I completely understand the frustration with network disconnections, and it’s crucial to implement an effective runtime detection mechanism.

On the hardware side, achieving true 4G/5G speeds over a UART interface is unlikely, as the typical 115200 baud rate is insufficient to match the actual upload/download speeds. While HTTP requests are faster than HTTPS due to the additional layer of certificate authentication, compromising security, especially in public domains, is not an option.

In my current project, we are using an ESP32S3 and an A7672G module connected via UART. Uploading 200 MB of data (~1000 files) to a high-end server with a renowned PHP framework backend took approximately 4 hours.

Next week, I plan to test the USB interface to see if it can deliver true 4G speeds.

parthbhat13, 2024-07-14:
hi @sisodiakaran ,
im glad to have a 3rd person point of view for the whole thread, appreciate you reading through the long comments ive made. since we are into vending industry and having a real time communication is must for us, unlike other situations where we can have some amount of downtime lets say even a min or 2, in our case if that happens then the customers declare it as a faulty machine and not to purchase the product. 

i eventually had decided to go with the quectel ec200u as there are many plus points in that modem apart from simcom, but my team was firm to not change the modem as it would lead to multiple design changes and two stack able pcb due to the size issues.

about the baudrate and usb interface we cannot use esp32-s3 since it is not easily available at our vendor, and if we have to buy it then we have to purchase the whole reel it seems, and since the price is now declared to the client we cannot even change the prices. i was asked by many to go for the USB connection including the technical head of simcom india, but we could not go there since the esp32 do not have USB and rather esp32-s3 has. 

so far the connection drops and other automagically causing issues of the simcom modem is fixed and i would recommend the following procedures to others who are reading this so they do not face the issue. 

1. make sure to give a proper dedicated power supply to the gsm modem, i choose to go with, AP62301Z6 buck converter due to is form factor and very less supporting components. (earlier due to size constrains and cost cutting we used array of SS34 diodes to lower the 5V supply to 3.9V, there are people i know who use it and it works for them but not for me)

2. have a very good antenna design on the PCB, again price constrain so only 2 layer pcb for us, but people recommend to use 4 layer, irrespective of the 4 layer we did it on 2 layer and now it works like a charm to the point that in open sky without even connecting the antenna we get RSSI upto 25. 

3. Finally since everything depends on the quality and control of the PCB manufracturing we decided to print it in india from PCB Power, but even JLC works best for this. 

following the above 3 points so far have got the GSM absolutely very stable for esp32 to put any kind of load onto it. earlier it was as poor as if we ping the GSM with AT command after the power up, it would either freeze or shutdown. 

as of now, we are dealing with issues related to the xcrt bundle certificates with the esp and its HTTPS request as it seems the server is way too secure for the esp32 liking and we often get the socket open error, trying to fix that as im typing this. 

the last thing i would wanna try is to use CMUX, since cmux for simcom has patch in the esp-modem and at times it has been noticed that if the PPP fails out of the blue from the GSM side, the library cannot recover and we have to destroy and reinitialize the whole library to start from start. maybe @david-cermak can throw some light on this. i would still implement the CMUX in our production code rather than relying on the esp-modem-console example and see how well does that work for us. 

but please do update on how well things are working on your side and if you have any inputs for the backend server i would be more than happy to share personally, since cannot share it public. 

thanks for your comment and appreciate your time.

sisodiakaran, 2024-07-14:
@parthbhat13, thanks for the pointers. We also have our PCBs manufactured by PCB Power, as they offer many advanced capabilities, such as buried vias. Our PCB is a 4-layer design, and we are consistently receiving an RSSI of 24, indicating stable network and power performance. Although we are not using the specified buck converter (I need to verify this with our E. engineer).

I also struggled many times with the certificates, every time when I get new project it is fixed that my first 1 hr will be invested in getting correct certificate. However, as per me socket open error can be resolved by increasing ssl handshaking timeout.

parthbhat13, 2024-07-14:
>  However, as per me socket open error can be resolved by increasing ssl handshaking timeout.

@sisodiakaran can you shed more light on this? Or share any idf example? 

Thanks in advance

parthbhat13, 2024-08-23:
hi @david-cermak , 
Hope you are doing good, since the last update i have been working and developing on this for a while now, and during the stress testing i happen to bump into the following error crash. i have used esp-modem with A7672s below is the size's ive declared for the esp-modem and its buffers. and then after that ive posted the error. 

```
.espModemConfig.rxBuffSize = 2048,
.espModemConfig.txBuffSize = 1024,
.espModemConfig.uartEventQueueSize = 50,
.espModemConfig.gsmTaskStackSize = 1024 * 4,
.espModemConfig.gsmTaskPriority = configMAX_PRIORITIES - 4,
```

here is the error which i am facing, its a core-dump 

```
Mem alloc fail. size 0x000005f0 caps 0x00001800

Backtrace: 0x400821fe:0x3ffda310 0x4008b151:0x3ffda330 0x40082541:0x3ffda350 0x4008259d:0x3ffda3a0 0x400825b5:0x3ffda3c0 0x40095269:0x3ffda3e0 0x40107190:0x3ffda400 0x4010720f:0x3ffda420 0x401072a1:0x3ffda440 0x40107fed:0x3ffda460 0x40119d59:0x3ffda490 0x40100916:0x3ffda4b0 0x401916c9:0x3ffda4d0 0x400e43fe:0x3ffda4f0 0x400e440d:0x3ffda510 0x400e2cfd:0x3ffda530 0x400e5be6:0x3ffda560 0x400e2cfd:0x3ffda580 0x400e2d99:0x3ffda5b0 0x400e2ea6:0x3ffda5e0 0x40090b7a:0x3ffda600
0x400821fe: panic_abort at /home/parthasus/esp/esp-idf/components/esp_system/panic.c:408

0x4008b151: esp_system_abort at /home/parthasus/esp/esp-idf/components/esp_system/esp_system.c:137

0x40082541: heap_caps_alloc_failed at /home/parthasus/esp/esp-idf/components/heap/heap_caps.c:95

0x4008259d: heap_caps_malloc at /home/parthasus/esp/esp-idf/components/heap/heap_caps.c:198

0x400825b5: heap_caps_malloc_default at /home/parthasus/esp/esp-idf/components/heap/heap_caps.c:220

0x40095269: malloc at /home/parthasus/esp/esp-idf/components/newlib/heap.c:24

0x40107190: mem_malloc at /home/parthasus/esp/esp-idf/components/lwip/lwip/src/core/mem.c:237

0x4010720f: do_memp_malloc_pool at /home/parthasus/esp/esp-idf/components/lwip/lwip/src/core/memp.c:254

0x401072a1: memp_malloc at /home/parthasus/esp/esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x40107fed: pbuf_alloc at /home/parthasus/esp/esp-idf/components/lwip/lwip/src/core/pbuf.c:247

0x40119d59: pppos_input_tcpip at /home/parthasus/esp/esp-idf/components/lwip/lwip/src/netif/ppp/pppos.c:434

0x40100916: esp_netif_lwip_ppp_input at /home/parthasus/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.c:320

0x401916c9: esp_netif_receive at /home/parthasus/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:966

0x400e43fe: esp_modem::Netif::receive(unsigned char*, unsigned int) at /home/parthasus/ESP32/esp32Projects/localPopProjects/RAiMECH_Clients/PaymentDevices/CDV6_Files/CD_RaimechV6/managed_components/espressif__esp_modem/src/esp_modem_netif.cpp:68

0x400e440d: std::_Function_handler<bool (unsigned char*, unsigned int), esp_modem::Netif::start()::{lambda(unsigned char*, unsigned int)#1}>::_M_invoke(std::_Any_data const&, unsigned char*&&, unsigned int&&) at /home/parthasus/ESP32/esp32Projects/localPopProjects/RAiMECH_Clients/PaymentDevices/CDV6_Files/CD_RaimechV6/managed_components/espressif__esp_modem/src/esp_modem_netif.cpp:87
 (inlined by) _M_invoke at /home/parthasus/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:282

0x400e2cfd: std::function<bool (unsigned char*, unsigned int)>::operator()(unsigned char*, unsigned int) const at /home/parthasus/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687

0x400e5be6: std::_Function_handler<bool (unsigned char*, unsigned int), esp_modem::DTE::set_read_cb(std::function<bool (unsigned char*, unsigned int)>)::{lambda(unsigned char*, unsigned int)#1}>::_M_invoke(std::_Any_data const&, unsigned char*&&, unsigned int&&) at /home/parthasus/ESP32/esp32Projects/localPopProjects/RAiMECH_Clients/PaymentDevices/CDV6_Files/CD_RaimechV6/managed_components/espressif__esp_modem/src/esp_modem_dte.cpp:293
 (inlined by) _M_invoke at /home/parthasus/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:282

0x400e2cfd: std::function<bool (unsigned char*, unsigned int)>::operator()(unsigned char*, unsigned int) const at /home/parthasus/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687

0x400e2d99: esp_modem::UartTerminal::task() at /home/parthasus/ESP32/esp32Projects/localPopProjects/RAiMECH_Clients/PaymentDevices/CDV6_Files/CD_RaimechV6/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp:124

0x400e2ea6: esp_modem::UartTerminal::s_task(void*) at /home/parthasus/ESP32/esp32Projects/localPopProjects/RAiMECH_Clients/PaymentDevices/CDV6_Files/CD_RaimechV6/managed_components/espressif__esp_modem/src/esp_modem_uart.cpp:74

0x40090b7a: vPortTaskWrapper at /home/parthasus/esp/esp-idf/components/freertos/port/xtensa/port.c:142
```

i wonder where things are going wrong? do i need to put uart in the ISR? i would test that while i am making this comment. 
thanks in advance

diplfranzhoepfinger, 2024-08-23:
you have PSRAM ?

parthbhat13, 2024-08-23:
update, 
i even tried to enable the uart ISR, but still i am having the exact same error and issue.

parthbhat13, 2024-08-23:
Another update, 
Solved the issue, the gsm stack was technically telling us it had no more memory left.so the memory was being consumed by a json request where I failed to free the memory. 

I'll later close this thread with a comment which would help others as well. 

Thanks a lot.