espressif/esp-protocols #553: problems with modem_tcp_client and Telit LE910Cx modem

owen63, 2024-04-18:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am following the modem_tcp_client example but need to transcribe the SIM7600/BG96 commands to the Telit equivalents. I am using the 'localhost listener' mode as I want to use the modem for HTTP as well as MQTT requests. My start_connecting cmd leaves the modem in data mode rather than command mode (because I couldn't consume the OK response). So I see the MQTT CONNECT received by the localhost listener and I need to send that to the dte so that it gets to the server. But I'm not sure how to do that, and the socket closes. Also I see that remote data is handled by an event file descriptor but I don't see how that event is signaled.

```
' (48489) sock_commands: start_connecting 'AT#SD=1,0,1883,"test.mosquitto.org",0,0,0
D (48499) sock_dce: Socket accepted!
V (48499) modem_client: ...performing
D (48509) sock_dce: socket read: data available
V (48509) sock_dce: 0x3fca0828   10 18 00 04 4d 51 54 54  04 02 00 78 00 0c 45 53  |....MQTT...x..ES|
V (48519) sock_dce: 0x3fca0838   50 33 32 5f 65 37 44 34  31 43                    |P32_e7D41C|
V (48529) sock_commands: start_sending
V (48529) modem_client: ...performing
D (48539) sock_dce: socket read: data available
E (48539) sock_dce: EOF 128
V (48539) sock_commands: net_close
```

david-cermak, 2024-04-19:
> I am following the modem_tcp_client example but need to transcribe the SIM7600/BG96 commands to the Telit equivalents.

Good to hear, maybe you can also think of updating the example in future? (PRs welcome)

>  am using the 'localhost listener' mode as I want to use the modem for HTTP as well as MQTT requests.

Note that both the HTTP request and the MQTT client libraries in IDF use `tcp_transport`, so you could also use the other mode.

> My start_connecting cmd leaves the modem in data mode rather than command mode (because I couldn't consume the OK response).

You need to use the AT command that sends raw data to the opened TCP connection.

> (because I couldn't consume the OK response).

This is happening in a separate thread in that example.

> I see the MQTT CONNECT received by the localhost listener and I need to send that to the dte so that it gets to the server. But I'm not sure how to do that, and the socket closes.

Here's the code that handles the data from the localhost listener. from the log it looks like it reads your data correctly

https://github.com/espressif/esp-protocols/blob/bd6e12050928464b4357015c7c8c1b215432d580/components/esp_modem/examples/modem_tcp_client/main/sock_dce.cpp#L53

but it only initiate sending, the actual data are added asynchronously.

> Also I see that remote data is handled by an event file descriptor but I don't see how that event is signaled.

whenever you get an asynchronous data from the modem, e.g. here:

https://github.com/espressif/esp-protocols/blob/bd6e12050928464b4357015c7c8c1b215432d580/components/esp_modem/examples/modem_tcp_client/main/sock_commands_sim7600.cpp#L346-L350

(I think both modems in the example just send async notification about "data-available", and the acutual reading is done synchronously)

david-cermak, 2024-11-14:
Any update with this @owen63 ? Please reopen if you need more assistance.

closed 2024-11-14 by david-cermak