espressif/esp-protocols #562: LwIP PPP connection close and no authentication

dmitrij9992905, 2024-04-28:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

For now, I need to connect to Internet from ESP32 using SIM800C again, using LilyGO T-Call based on AXP192 PMIC.
I use ESP-IDF 5.2.1 and esp-modem v1.1.0 component, and use simple_cmux_client as a base. 

But all I see that AT commands are working, PDP context is setting up, but connection to *99# establishes and interrupts even without connection to Internet. Even with setting the PAP authentication.

Now I'm using MTS sim card and I see the same effect as with Beeline.

Here is my code:
```
/*
 * SPDX-FileCopyrightText: 2022-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */
/* PPPoS Client Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <cstring>
#include <iostream>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_netif.h"
#include "esp_log.h"
#include "esp_event.h"
#include "cxx_include/esp_modem_dte.hpp"
#include "esp_modem_config.h"
#include "cxx_include/esp_modem_api.hpp"
#include "simple_mqtt_client.hpp"
#include "esp_vfs_dev.h"        // For optional VFS support
#include "esp_https_ota.h"      // For potential OTA configuration
#include "vfs_resource/vfs_create.hpp"
#include "SIM7070_gnss.hpp"
#include "driver/gpio.h"
#include "driver/i2c.h"
#include "axp20x.h"
#include "esp_netif.h"
#include "esp_netif_ppp.h"

#if defined(CONFIG_EXAMPLE_FLOW_CONTROL_NONE)
#define EXAMPLE_FLOW_CONTROL ESP_MODEM_FLOW_CONTROL_NONE
#elif defined(CONFIG_EXAMPLE_FLOW_CONTROL_SW)
#define EXAMPLE_FLOW_CONTROL ESP_MODEM_FLOW_CONTROL_SW
#elif defined(CONFIG_EXAMPLE_FLOW_CONTROL_HW)
#define EXAMPLE_FLOW_CONTROL ESP_MODEM_FLOW_CONTROL_HW
#endif

#define _I2C_NUMBER(num)            I2C_NUM_##num
#define I2C_NUMBER(num)             _I2C_NUMBER(num)

#define AXP_ISR_GPIO                (gpio_num_t)CONFIG_AXPxxx_INTERRUPT_PIN  /*!< axp power chip interrupt Pin*/
#define GPIO_INPUT_PIN_SEL          (1ULL<<AXP_ISR_GPIO)

#define I2C_MASTER_SCL_IO           CONFIG_I2C_MASTER_SCL                   /*!< gpio number for I2C master clock */
#define I2C_MASTER_SDA_IO           CONFIG_I2C_MASTER_SDA                   /*!< gpio number for I2C master data  */
#define I2C_MASTER_NUM              I2C_NUMBER(CONFIG_I2C_MASTER_PORT_NUM)  /*!< I2C port number for master dev */
#define I2C_MASTER_FREQ_HZ          CONFIG_I2C_MASTER_FREQUENCY             /*!< I2C master clock frequency */
#define I2C_MASTER_TX_BUF_DISABLE   0                                       /*!< I2C master doesn't need buffer */
[... 982 lines cut ...]
D (12512) esp_netif_lwip: check: remote, if=0x3ffbf388 fn=0x400ec064
0x400ec064: esp_netif_stop_api at /home/espidf521/esp/esp-idf-v5.2.1/components/esp_netif/lwip/esp_netif_lwip.c:1167

D (12522) esp-netif_lwip-ppp: Phase Network
D (12522) esp-netif_lwip-ppp: esp_netif_stop_ppp: Stopped PPP connection: 0x3ffbf5b8
D (12532) esp-netif_lwip-ppp: Phase Terminate
D (12532) esp-netif_lwip-ppp: Phase Establish
D (12542) esp_netif_lwip: check: local, if=0x3ffbf388 fn=0x400ec594
0x400ec594: esp_netif_update_default_netif_lwip at /home/espidf521/esp/esp-idf-v5.2.1/components/esp_netif/lwip/esp_netif_lwip.c:316

D (12542) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbf388
V (12552) esp_netif_objects: esp_netif_next_unsafe 0x0
V (12562) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3ffbf388
V (12562) esp_netif_objects: esp_netif_next_unsafe 0x3ffbf388
D (12572) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (12572) esp-netif_lwip-ppp: Phase Disconnect
D (12582) esp-netif_lwip-ppp: Phase Dead
I (12582) esp-netif_lwip-ppp: User interrupt
D (12592) event: running post NETIF_PPP_STATUS:5 with handler 0x400dc994 and context 0x3ffc0930 on loop 0x3ffb9f44
0x400dc994: esp_modem::Netif::on_ppp_changed(void*, char const*, long, void*) at /home/espidf521/esp32-workspace/simple_cmux_client/managed_components/espressif__esp_modem/src/esp_modem_netif.cpp:22

I (12602) esp_modem_netif: PPP state changed event 5
V (12602) intr_alloc: esp_intr_free: Disabling int, killing handler
I (12612) main_task: Returned from app_main()
```

david-cermak, 2024-05-07:
Hi @dmitrij9992905 

It seems like the initial connection just cannot be established, are you sure that your SIM card is activated, you've got the correct APN and the GSM band is supported in your country (SIM800 is G2 only)?

Could you try the default pppos client example, just to rule out potential CMUX issues?

dmitrij9992905, 2024-05-07:
> Hi @dmitrij9992905
> 
> It seems like the initial connection just cannot be established, are you sure that your SIM card is activated, you've got the correct APN and the GSM band is supported in your country (SIM800 is G2 only)?

Yes, 2G works in my country, and I took credentials such as APN and user/password. It seems to me that APN is OK but user and password is not being transfered at all. So, the connection closes.

> Could you try the default pppos client example, just to rule out potential CMUX issues?
Although I build my project based on default example, I will try to build another one based on pppos client example and write you back

dmitrij9992905, 2024-05-07:
I've just check both pppos and cmux modes, and it appeared that after power on, I restarted SIM800C modem and waited 10 secs, and it was too small to initiate modem and its connection to network. I set it 30 secs and started working normally