espressif/esp-protocols #570: esp_wifi_remote: key-pair generated by generate_test_certs.py not work in example

leeebo, 2024-05-11:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to run esp_wifi_remote examples with ESP32-P4+ESP32-C6, follow the example readme step by step:

1. generate all necessary keys and CA using `generate_test_certs.py`
2. copy the dumped `CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_CA=xxx` `CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_CRT=xxx` `CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_KEY=xxx` `CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_CA=xxx` `CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_CRT=xxx`  `CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_KEY=xxx` to both `mqtt` and `server` example's `sdkconfig.defaults`
3. build and flash both chips, but encounter issues "E (3207) esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x3D00".

Generate keys and ca:

```
Server's CN: Espressif
Client's CN: client_cn
Certificate request self-signature ok
subject=CN = Espressif
Certificate request self-signature ok
subject=CN = client_cn
CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_CA="MIIDIzCCAgugAwIBAgIUWbKChtZhcGB0HfOEGQ49Xbs0a7cwDQYJKoZIhvcNAQELBQAwITELMAkGA1UEBhMCQ1oxEjAQBgNVBAMMCUVzcHJlc3NpZjAeFw0yNDA1MTEwOTU3MzhaFw0yNTA1MTEwOTU3MzhaMCExCzAJBgNVBAYTAkNaMRIwEAYDVQQDDAlFc3ByZXNzaWYwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCsKybskW5t56XJG9gp2Jy00mIYyXS4UdDOTE/nFfRIe1YzAXbh0WfR4RJ8VH2OYNmD4VBzqnYp5nY0CNfmvoVrltpSDTrb7bzMxoZLqAaUkYz8bcAtzs6S9j5axPOIkW96mTmLJRML9B01409Zz7Yef2K3h9soKLLndhsNMcRKGLTdVLJTmppPVM0PrI80CeFm7pZPOoqhAUzBSH9tiwz1Ydxd0bniBoWw4JWEoRgjeeqpH2ySwiveLbMaAsxR8qAuL9/cmLnaNs625sdXJzVfPQ7yHCOIDJMjxv8QPdKdTPKqssDeeNGhJnHPu6CEZb3Y2dVyqcHhvURHlXo6oKdJAgMBAAGjUzBRMB0GA1UdDgQWBBQjf59LZmxghhSDxEy1P8PGhT6SPTAfBgNVHSMEGDAWgBQjf59LZmxghhSDxEy1P8PGhT6SPTAPBgNVHRMBAf8EBTADAQH/MA0GCSqGSIb3DQEBCwUAA4IBAQAFP82xx0OIFbGIjcP3p8gFNz494ywcoeJSsJzaRPJ/dP/nigCfml0xvGMgaX74J/y3jJldbwdVKnhbxgB2DZdwx1xjAo8ERxpU2/zJRi5owrvFXQlN7OyYMfJ8YMQfy62aFMkOU8Hau8JiH/j6wQn4JjKS7zep3FjtGi7tMQXBLb9wKDrxBCb2mrk6phHcSUhBOg78mo/vjqrlBjD8cJxIbVtALpZcqyzKOvSJfGTFU9UBZ2Y7o7QACEXjoO8rljdLWXrBRPvsuRo/Z3rrt0yQjWX0CgFJYqOjQVF9kVV3Hle6QUeOTyLTzSBg06xNQssnPquhRfp7Cwn+tL3Ck93X"
CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_CRT="MIICvDCCAaQCFGA9TGBPAYghd7eolEdQyAyLFiESMA0GCSqGSIb3DQEBCwUAMCExCzAJBgNVBAYTAkNaMRIwEAYDVQQDDAlFc3ByZXNzaWYwHhcNMjQwNTExMDk1NzM4WhcNMjUwNTExMDk1NzM4WjAUMRIwEAYDVQQDDAljbGllbnRfY24wggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCjQLRO7KXlUFAC0CLACVo1+0PEnxDRyjBfzSlPUL2Le/6zBeZUoGEEJYb7ZMPdP4XafdAckIgXdnYFsU3vYqK9L57V6Uzhp2VgIBs3TghpshrvIW2q5T3hnPBU+DjTbQu8yCfLd9nUlvDX3fIHPYX8UXd6xJUNkTe3a17+GzL84t6fYiVu59Wg8fziX7EjCrUvUBQcwhDSvwNR//Eg8vFFoG+PHVetwIoNHvcEbPT70DPxhGXM6uHa/cHYNyrDIa6PtbuSFg+oC+QHb6N1l6Vvisv4H4IGDh928nYCBzD3I5jamX3iFqJHgrqOYiWfhz3PY6ausC0bBcvURmwKfVV3AgMBAAEwDQYJKoZIhvcNAQELBQADggEBAGHMwxJXozNJ10m4Hn5/4jCO81isbPGCJPn0VURxH58/41eI7AOLrNngay6uY3cAQbEovmtPKwgewND5XF+B7nPiIVGBWx6y+Bg5NCalcIJAUHxAnIFU6Rlaojbr1HgXTdeA/GXtg89e/GWwGP+hwkn065KsMJQ6sRG4Wb8S22XupOTZp3TvYVqYT38iwW8sAmN/xvOLPwqaraiFNXL6t4tx0QA8D/Q/kLv9/+siJ+Moz+lGjC6rWgfJ/if1kGRefHszAb2cGqba6+aLW/nzknpE4a7da1LsA8fKJLrY/kETrj9pVPfKEDAmQicg2ZZEzClX/ZQLz3LXlb6yg8q/ZGM="
CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_KEY="MIIEvAIBADANBgkqhkiG9w0BAQEFAASCBKYwggSiAgEAAoIBAQCjQLRO7KXlUFAC0CLACVo1+0PEnxDRyjBfzSlPUL2Le/6zBeZUoGEEJYb7ZMPdP4XafdAckIgXdnYFsU3vYqK9L57V6Uzhp2VgIBs3TghpshrvIW2q5T3hnPBU+DjTbQu8yCfLd9nUlvDX3fIHPYX8UXd6xJUNkTe3a17+GzL84t6fYiVu59Wg8fziX7EjCrUvUBQcwhDSvwNR//Eg8vFFoG+PHVetwIoNHvcEbPT70DPxhGXM6uHa/cHYNyrDIa6PtbuSFg+oC+QHb6N1l6Vvisv4H4IGDh928nYCBzD3I5jamX3iFqJHgrqOYiWfhz3PY6ausC0bBcvURmwKfVV3AgMBAAECggEAPU/wSBIFzkVm/193JaN9j3eSQJazBO+IUnqlkKXUwj2r70WAd3kSZrHHa8Qw8XfJiEEnN4sqqZEyiznHeOgVEbOAVnt8BeXOI5d7UT6L11JuXjAyVXQqVQERm//rq7AxL9IGbBrTo02bH5+6JfD70MMx+dg52LIY+FipgQrrD+F094063/T+Qas7GZ/UpP7pewzA7teKn6JzBo9rQbI6KoKH9tvd/bm67h4u5e12tDvUB01sVlj8JlN3AQJ//3JXRlLOs6ZCJK2vAvTUBNtiaXSj3fjxloyUIdhg6R6dnIOZCEfek/39pWy4GQIXXypj60YuD18VYATMbvOGTmRaVQKBgQDZ2K3OTQcIonW59meCUuIheGLXdg3QLNlTdAairu2wPMvZ6bjAhttgjc8TYZGvRLdKf84mgjQjhdzahyzdDnrzs0sk9QYgJQhJonX77XESq7Urs6XqWGzXgpKTw1h6j8TxJDkq9omQL6fVDEHGRR9gNi7HN2Fq18tK7KZQiLXWfQKBgQC/2EZI/D0wVeoknYcCTgHu32t2DrEIqfaSLYu0aeHPP1AcISdczXDZLxbsgxgMSlX6OXmtPwpxpOwYZe921g9xXgUTiGF0OK0aV/SARv2eC9qwGKiL15sbN2MDAeO5FKUUFF6JVAAWISud6tGjwv+kwE6rKkR4SoSGyKmlwqm6AwKBgHoRa1rlWKKi/hrFEr05GlMs9R3MnfLfnqufoTQ5lyChAI1L5OWk63qQG8+9ZXfzpk1NjkOPxA1/rgCZrC72NTknZ1z/G36Zoao5KamQJEMVCkDBup+1CtA770xzMroCo5WH+ay0VuL/gSjlZFnoU6uC0PV6d9D5uIV0U8FeyZ21AoGAO/fZEJt4bRGOHV/4genshMTmDINPJ4wpfUwrbXvjkCsUWZarfUDi06D6o3ffHohmE8+/jz4zf460FnRWxGjpbuCa6Z4qCHBm++MelLKcOLkv5PfwR7Zfz8QieSAwFPy1FW/AlWDinghEYBq06WHdE5zhenO973QrtQ4oqZ4N/A0CgYBhwRLzLJ3BtV9/6vtwXmVYOe844fJ3s0LrznwoxUHZG/uP14RaQwdwOthmZFe63HBuzWV6XmnpgHEblqYcr4pS6UOf4OadxFjBhEzDRW9XGAb4bAac3/NIfTF42BIDHE6yDgzzgBBiKK15NweButjTPEROSBooihpPKYNCJk0lYw=="
CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_CA="MIIDIzCCAgugAwIBAgIUWbKChtZhcGB0HfOEGQ49Xbs0a7cwDQYJKoZIhvcNAQELBQAwITELMAkGA1UEBhMCQ1oxEjAQBgNVBAMMCUVzcHJlc3NpZjAeFw0yNDA1MTEwOTU3MzhaFw0yNTA1MTEwOTU3MzhaMCExCzAJBgNVBAYTAkNaMRIwEAYDVQQDDAlFc3ByZXNzaWYwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCsKybskW5t56XJG9gp2Jy00mIYyXS4UdDOTE/nFfRIe1YzAXbh0WfR4RJ8VH2OYNmD4VBzqnYp5nY0CNfmvoVrltpSDTrb7bzMxoZLqAaUkYz8bcAtzs6S9j5axPOIkW96mTmLJRML9B01409Zz7Yef2K3h9soKLLndhsNMcRKGLTdVLJTmppPVM0PrI80CeFm7pZPOoqhAUzBSH9tiwz1Ydxd0bniBoWw4JWEoRgjeeqpH2ySwiveLbMaAsxR8qAuL9/cmLnaNs625sdXJzVfPQ7yHCOIDJMjxv8QPdKdTPKqssDeeNGhJnHPu6CEZb3Y2dVyqcHhvURHlXo6oKdJAgMBAAGjUzBRMB0GA1UdDgQWBBQjf59LZmxghhSDxEy1P8PGhT6SPTAfBgNVHSMEGDAWgBQjf59LZmxghhSDxEy1P8PGhT6SPTAPBgNVHRMBAf8EBTADAQH/MA0GCSqGSIb3DQEBCwUAA4IBAQAFP82xx0OIFbGIjcP3p8gFNz494ywcoeJSsJzaRPJ/dP/nigCfml0xvGMgaX74J/y3jJldbwdVKnhbxgB2DZdwx1xjAo8ERxpU2/zJRi5owrvFXQlN7OyYMfJ8YMQfy62aFMkOU8Hau8JiH/j6wQn4JjKS7zep3FjtGi7tMQXBLb9wKDrxBCb2mrk6phHcSUhBOg78mo/vjqrlBjD8cJxIbVtALpZcqyzKOvSJfGTFU9UBZ2Y7o7QACEXjoO8rljdLWXrBRPvsuRo/Z3rrt0yQjWX0CgFJYqOjQVF9kVV3Hle6QUeOTyLTzSBg06xNQssnPquhRfp7Cwn+tL3Ck93X"
CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_CRT="MIICvDCCAaQCFDSDK0LfVk9svsGxFt2Gzfdk8P9FMA0GCSqGSIb3DQEBCwUAMCExCzAJBgNVBAYTAkNaMRIwEAYDVQQDDAlFc3ByZXNzaWYwHhcNMjQwNTExMDk1NzM4WhcNMjUwNTExMDk1NzM4WjAUMRIwEAYDVQQDDAlFc3ByZXNzaWYwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDrJFZVJtMBzXlVTwqNdM2mMsX7+KPk3HidPfOXAKre4VnFxHcMtrWbkwbgN5jY81fTO1nOBNpQ3qXZSTvCWg+No2mj6EFQEGsRxTuAEbdGrlhWhAYXVXa7n5xLFvwrluXlfgkfmPqfCRgAp9wv0nELigrgKHK01k2kzXE1l2J/Boo+JScsFJDlrMfCxqnN/H4ZC3Ln6FVhEyUnUfEZKTCv2q0ebZT+wjQQDYgKAbbgk8vZE5mqZ3w5qrb8IwFcLlE4xHMhzSvQEi1mITpFQ4HNs24tsHKrPRLzYWZoAXX44AE4GQnWn76XsHlGWIZpeUKFvMukCdX8HJ3vXMOPDc95AgMBAAEwDQYJKoZIhvcNAQELBQADggEBAA8zvKDeVtVHXkrUSquRKm2Qh8QueDb/MxxBBso0agGLt7hfWy2t7X+Hf52TNSb0Xs9QsgJm0qf849zNqX/gt5nOzhjg+hg4S2DmOy0uu/ZADymroXHWWSeW+ZqC1KcyCt3iJoarELCWACOclGk+GmPvIezv8K4ux1G51hv3waIbV4/qTi+xrDXWTUlW+4PNlf14z8N3DDwQQEgH8T/rQk9IqyzKyHHUe+wIMP1Rk9F+N7OJCgA1E8Jdcif/qVBlpzyVEokdOR2plqYHUhik6fzwWAk4N5HHhcjo3Eof7XpAhOnvKz1wUCsX4gga9IYpXMgyhA5X+HzzWCjgLXJpOQM="
CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_KEY="MIIEvQIBADANBgkqhkiG9w0BAQEFAASCBKcwggSjAgEAAoIBAQDrJFZVJtMBzXlVTwqNdM2mMsX7+KPk3HidPfOXAKre4VnFxHcMtrWbkwbgN5jY81fTO1nOBNpQ3qXZSTvCWg+No2mj6EFQEGsRxTuAEbdGrlhWhAYXVXa7n5xLFvwrluXlfgkfmPqfCRgAp9wv0nELigrgKHK01k2kzXE1l2J/Boo+JScsFJDlrMfCxqnN/H4ZC3Ln6FVhEyUnUfEZKTCv2q0ebZT+wjQQDYgKAbbgk8vZE5mqZ3w5qrb8IwFcLlE4xHMhzSvQEi1mITpFQ4HNs24tsHKrPRLzYWZoAXX44AE4GQnWn76XsHlGWIZpeUKFvMukCdX8HJ3vXMOPDc95AgMBAAECggEAZnp1rbK372cvZACIptGq9q3WH1RjiVCUuGLRkJPX5D7uMwTGpaHTUDQpoPpaAOM8+XA6WVFAHFKMkUAbbT523NHFLowAOvaM0zfUEKKn74P4N+xeAzlZ/7c2FHu19wFSvEvLihvPzhPWhCKkEatzXQYSa1L1et6TBCoNFsAdSyImrcsQkjlEfbroZEWx1Egcv4OmXhI2PIJprUMvNQE86eEByKW78W17KLL/w/S1Z7jTOGnJqFKOzn5k9/cjeuYnKmxtcpEokWfnyquEacafqOozyPM2sZ8Elnn66EG2q2s3ehuJX1kaEQEZOAo7J5GR2R2OVSs5WiDeH57HwZFLiQKBgQD2UBI0lp84OQW2O1sdkbYY0RDz9s38eFAzWGWUtSzCCrvW8ZZmqy8yrvUfAC1JRrdBlpX6JoWFCCtbW8ojffociZSPqYSCV3NajtWuiexGt/QWK8jAZbyHmL0lYW1qdwj90tM9eLVXAcJjx4TCqwSZPMlyly+pniwO3m6CDjMBIwKBgQD0Y8u5g5sJxgXua2IZlN2Jp/xzav5zqojqrTuW2GSZtgHjqJZkM4u44qBo+d8ZiuQrBrLhX736aRxgNlB7eOhqGS/IO0Iy0IHF24ON+peN1cn8ScLYCK6csWXgmZ54MyVpxsxonn+bdlofF+/4JKmsZd7T30Cv2/4k+efcpiYsswKBgQCiNTL4DOVH6k13p/kmLOpFx5CpiTQ3+qeeWK101CdolhcDw09vx3Yqo4trYftw0TAn8lpMRYgShsG+xWEGJJVMq9H5FgLcr2iZD41rBwthuwz0eNsY0SbqCU0uy815mnCCb4+LYO3TIaZ5fVBRvrwliiGD5EqQIkkTdapC3cT7DwKBgGNdJ/m3bpPctZCojGxGQIS1jkkHGYnBroVsBaxhMxXkEI [... cut ...]
E (12497) esp-tls-mbedtls: Failed to set server pki context
E (12507) esp-tls-mbedtls: Failed to set server configurations, returned [0x8019] (ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED)
E (12517) esp-tls-mbedtls: create_ssl_handle failed, returned [0x8019] (ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED)
E (12527) rpc_server: init_server(229): Failed to create TLS session
E (12527) rpc_server: init(44): Failed to init RPC engine
I (12537) main_task: Returned from app_main()
```

Would you please point out the problems?

david-cermak, 2024-05-13:
There's one issue with your configuration (as the server's CN is hardcoded to `espressif.local`)

https://github.com/espressif/esp-protocols/blob/8f19c60c1998fb6234f881fd101c0064e37cbd8b/components/esp_wifi_remote/eppp/wifi_remote_rpc_client.cpp#L220C24-L220C37

so you should generate the keys calling:
```bash
./managed_components/espressif__esp_wifi_remote/examples/test_certs/generate_test_certs.sh espressif.local
```

Which doesn't explain the error above, though. This `ESP_ERR_MBEDTLS_PK_PARSE_KEY_FAILED` means that mbedTLS just doesn't understand the key format. Did you copy the keys correctly, no whitespaces, properly quoted, do they appear in the generated `sdkconfig.h`? Maybe try `cat sdkcofnig.h | grep 'CONFIG_ESP_WIFI_REMOTE_EPPP_'` and compare with the generated output?

I'm also working on IDF example in `!26469`, here's a draft of the README from it:
[README.md](https://github.com/espressif/esp-protocols/files/15289772/README.md)

leeebo, 2024-05-17:
@david-cermak  The strings in `sdkconfig.h` are same with generated output.

```sh
❯ cat ./build/config/sdkconfig.h | grep 'CONFIG_ESP_WIFI_REMOTE_EPPP_'
#define CONFIG_ESP_WIFI_REMOTE_EPPP_UART_TX_PIN 16
#define CONFIG_ESP_WIFI_REMOTE_EPPP_UART_RX_PIN 50
#define CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_CA "MIIDIzCCAgugAwIBAgIUWbKChtZhcGB0HfOEGQ49Xbs0a7cwDQYJKoZIhvcNAQELBQAwITELMAkGA1UEBhMCQ1oxEjAQBgNVBAMMCUVzcHJlc3NpZjAeFw0yNDA1MTEwOTU3MzhaFw0yNTA1MTEwOTU3MzhaMCExCzAJBgNVBAYTAkNaMRIwEAYDVQQDDAlFc3ByZXNzaWYwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCsKybskW5t56XJG9gp2Jy00mIYyXS4UdDOTE/nFfRIe1YzAXbh0WfR4RJ8VH2OYNmD4VBzqnYp5nY0CNfmvoVrltpSDTrb7bzMxoZLqAaUkYz8bcAtzs6S9j5axPOIkW96mTmLJRML9B01409Zz7Yef2K3h9soKLLndhsNMcRKGLTdVLJTmppPVM0PrI80CeFm7pZPOoqhAUzBSH9tiwz1Ydxd0bniBoWw4JWEoRgjeeqpH2ySwiveLbMaAsxR8qAuL9/cmLnaNs625sdXJzVfPQ7yHCOIDJMjxv8QPdKdTPKqssDeeNGhJnHPu6CEZb3Y2dVyqcHhvURHlXo6oKdJAgMBAAGjUzBRMB0GA1UdDgQWBBQjf59LZmxghhSDxEy1P8PGhT6SPTAfBgNVHSMEGDAWgBQjf59LZmxghhSDxEy1P8PGhT6SPTAPBgNVHRMBAf8EBTADAQH/MA0GCSqGSIb3DQEBCwUAA4IBAQAFP82xx0OIFbGIjcP3p8gFNz494ywcoeJSsJzaRPJ/dP/nigCfml0xvGMgaX74J/y3jJldbwdVKnhbxgB2DZdwx1xjAo8ERxpU2/zJRi5owrvFXQlN7OyYMfJ8YMQfy62aFMkOU8Hau8JiH/j6wQn4JjKS7zep3FjtGi7tMQXBLb9wKDrxBCb2mrk6phHcSUhBOg78mo/vjqrlBjD8cJxIbVtALpZcqyzKOvSJfGTFU9UBZ2Y7o7QACEXjoO8rljdLWXrBRPvsuRo/Z3rrt0yQjWX0CgFJYqOjQVF9kVV3Hle6QUeOTyLTzSBg06xNQssnPquhRfp7Cwn+tL3Ck93X"
#define CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_CRT "MIICvDCCAaQCFGA9TGBPAYghd7eolEdQyAyLFiESMA0GCSqGSIb3DQEBCwUAMCExCzAJBgNVBAYTAkNaMRIwEAYDVQQDDAlFc3ByZXNzaWYwHhcNMjQwNTExMDk1NzM4WhcNMjUwNTExMDk1NzM4WjAUMRIwEAYDVQQDDAljbGllbnRfY24wggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCjQLRO7KXlUFAC0CLACVo1+0PEnxDRyjBfzSlPUL2Le/6zBeZUoGEEJYb7ZMPdP4XafdAckIgXdnYFsU3vYqK9L57V6Uzhp2VgIBs3TghpshrvIW2q5T3hnPBU+DjTbQu8yCfLd9nUlvDX3fIHPYX8UXd6xJUNkTe3a17+GzL84t6fYiVu59Wg8fziX7EjCrUvUBQcwhDSvwNR//Eg8vFFoG+PHVetwIoNHvcEbPT70DPxhGXM6uHa/cHYNyrDIa6PtbuSFg+oC+QHb6N1l6Vvisv4H4IGDh928nYCBzD3I5jamX3iFqJHgrqOYiWfhz3PY6ausC0bBcvURmwKfVV3AgMBAAEwDQYJKoZIhvcNAQELBQADggEBAGHMwxJXozNJ10m4Hn5/4jCO81isbPGCJPn0VURxH58/41eI7AOLrNngay6uY3cAQbEovmtPKwgewND5XF+B7nPiIVGBWx6y+Bg5NCalcIJAUHxAnIFU6Rlaojbr1HgXTdeA/GXtg89e/GWwGP+hwkn065KsMJQ6sRG4Wb8S22XupOTZp3TvYVqYT38iwW8sAmN/xvOLPwqaraiFNXL6t4tx0QA8D/Q/kLv9/+siJ+Moz+lGjC6rWgfJ/if1kGRefHszAb2cGqba6+aLW/nzknpE4a7da1LsA8fKJLrY/kETrj9pVPfKEDAmQicg2ZZEzClX/ZQLz3LXlb6yg8q/ZGM="
#define CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_KEY "MIIEvAIBADANBgkqhkiG9w0BAQEFAASCBKYwggSiAgEAAoIBAQCjQLRO7KXlUFAC0CLACVo1+0PEnxDRyjBfzSlPUL2Le/6zBeZUoGEEJYb7ZMPdP4XafdAckIgXdnYFsU3vYqK9L57V6Uzhp2VgIBs3TghpshrvIW2q5T3hnPBU+DjTbQu8yCfLd9nUlvDX3fIHPYX8UXd6xJUNkTe3a17+GzL84t6fYiVu59Wg8fziX7EjCrUvUBQcwhDSvwNR//Eg8vFFoG+PHVetwIoNHvcEbPT70DPxhGXM6uHa/cHYNyrDIa6PtbuSFg+oC+QHb6N1l6Vvisv4H4IGDh928nYCBzD3I5jamX3iFqJHgrqOYiWfhz3PY6ausC0bBcvURmwKfVV3AgMBAAECggEAPU/wSBIFzkVm/193JaN9j3eSQJazBO+IUnqlkKXUwj2r70WAd3kSZrHHa8Qw8XfJiEEnN4sqqZEyiznHeOgVEbOAVnt8BeXOI5d7UT6L11JuXjAyVXQqVQERm//rq7AxL9IGbBrTo02bH5+6JfD70MMx+dg52LIY+FipgQrrD+F094063/T+Qas7GZ/UpP7pewzA7teKn6JzBo9rQbI6KoKH9tvd/bm67h4u5e12tDvUB01sVlj8JlN3AQJ//3JXRlLOs6ZCJK2vAvTUBNtiaXSj3fjxloyUIdhg6R6dnIOZCEfek/39pWy4GQIXXypj60YuD18VYATMbvOGTmRaVQKBgQDZ2K3OTQcIonW59meCUuIheGLXdg3QLNlTdAairu2wPMvZ6bjAhttgjc8TYZGvRLdKf84mgjQjhdzahyzdDnrzs0sk9QYgJQhJonX77XESq7Urs6XqWGzXgpKTw1h6j8TxJDkq9omQL6fVDEHGRR9gNi7HN2Fq18tK7KZQiLXWfQKBgQC/2EZI/D0wVeoknYcCTgHu32t2DrEIqfaSLYu0aeHPP1AcISdczXDZLxbsgxgMSlX6OXmtPwpxpOwYZe921g9xXgUTiGF0OK0aV/SARv2eC9qwGKiL15sbN2MDAeO5FKUUFF6JVAAWISud6tGjwv+kwE6rKkR4SoSGyKmlwqm6AwKBgHoRa1rlWKKi/hrFEr05GlMs9R3MnfLfnqufoTQ5lyChAI1L5OWk63qQG8+9ZXfzpk1NjkOPxA1/rgCZrC72NTknZ1z/G36Zoao5KamQJEMVCkDBup+1CtA770xzMroCo5WH+ay0VuL/gSjlZFnoU6uC0PV6d9D5uIV0U8FeyZ21AoGAO/fZEJt4bRGOHV/4genshMTmDINPJ4wpfUwrbXvjkCsUWZarfUDi06D6o3ffHohmE8+/jz4zf460FnRWxGjpbuCa6Z4qCHBm++MelLKcOLkv5PfwR7Zfz8QieSAwFPy1FW/AlWDinghEYBq06WHdE5zhenO973QrtQ4oqZ4N/A0CgYBhwRLzLJ3BtV9/6vtwXmVYOe844fJ3s0LrznwoxUHZG/uP14RaQwdwOthmZFe63HBuzWV6XmnpgHEblqYcr4pS6UOf4OadxFjBhEzDRW9XGAb4bAac3/NIfTF42BIDHE6yDgzzgBBiKK15NweButjTPEROSBooihpPKYNCJk0lYw=="
#define CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_CA "MIIDIzCCAgugAwIBAgIUWbKChtZhcGB0HfOEGQ49Xbs0a7cwDQYJKoZIhvcNAQELBQAwITELMAkGA1UEBhMCQ1oxEjAQBgNVBAMMCUVzcHJlc3NpZjAeFw0yNDA1MTEwOTU3MzhaFw0yNTA1MTEwOTU3MzhaMCExCzAJBgNVBAYTAkNaMRIwEAYDVQQDDAlFc3ByZXNzaWYwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCsKybskW5t56XJG9gp2Jy00mIYyXS4UdDOTE/nFfRIe1YzAXbh0WfR4RJ8VH2OYNmD4VBzqnYp5nY0CNfmvoVrltpSDTrb7bzMxoZLqAaUkYz8bcAtzs6S9j5axPOIkW96mTmLJRML9B01409Zz7Yef2K3h9soKLLndhsNMcRKGLTdVLJTmppPVM0PrI80CeFm7pZPOoqhAUzBSH9tiwz1Ydxd0bniBoWw4JWEoRgjeeqpH2ySwiveLbMaAsxR8qAuL9/cmLnaNs625sdXJzVfPQ7yHCOIDJMjxv8QPdKdTPKqssDeeNGhJnHPu6CEZb3Y2dVyqcHhvURHlXo6oKdJAgMBAAGjUzBRMB0GA1UdDgQWBBQjf59LZmxghhSDxEy1P8PGhT6SPTAfBgNVHSMEGDAWgBQjf59LZmxghhSDxEy1P8PGhT6SPTAPBgNVHRMBAf8EBTADAQH/MA0GCSqGSIb3DQEBCwUAA4IBAQAFP82xx0OIFbGIjcP3p8gFNz494ywcoeJSsJzaRPJ/dP/nigCfml0xvGMgaX74J/y3jJldbwdVKnhbxgB2DZdwx1xjAo8ERxpU2/zJRi5owrvFXQlN7OyYMfJ8YMQfy62aFMkOU8Hau8JiH/j6wQn4JjKS7zep3FjtGi7tMQXBLb9wKDrxBCb2mrk6phHcSUhBOg78mo/vjqrlBjD8cJxIbVtALpZcqyzKOvSJfGTFU9UBZ2Y7o7QACEXjoO8rljdLWXrBRPvsuRo/Z3rrt0yQjWX0CgFJYqOjQVF9kVV3Hle6QUeOTyLTzSBg06xNQssnPquhRfp7Cwn+tL3Ck93X"
#define CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_CRT "MIICvDCCAaQCFDSDK0LfVk9svsGxFt2Gzfdk8P9FMA0GCSqGSIb3DQEBCwUAMCExCzAJBgNVBAYTAkNaMRIwEAYDVQQDDAlFc3ByZXNzaWYwHhcNMjQwNTExMDk1NzM4WhcNMjUwNTExMDk1NzM4WjAUMRIwEAYDVQQDDAlFc3ByZXNzaWYwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDrJFZVJtMBzXlVTwqNdM2mMsX7+KPk3HidPfOXAKre4VnFxHcMtrWbkwbgN5 [... cut ...] TQ3+qeeWK101CdolhcDw09vx3Yqo4trYftw0TAn8lpMRYgShsG+xWEGJJVMq9H5FgLcr2iZD41rBwthuwz0eNsY0SbqCU0uy815mnCCb4+LYO3TIaZ5fVBRvrwliiGD5EqQIkkTdapC3cT7DwKBgGNdJ/m3bpPctZCojGxGQIS1jkkHGYnBroVsBaxhMxXkEISViQAYjRiQ3BOSdIkYFYkA4j4MBVwI3IapzrNFB4oLz4ejwFiUug7OH4DcgD4kzEkNXHElpokC/58AFpRchzuonndy6gM1bZLy0pibn3j+ukh+nvhhg8VY7lKiIydNAoGAHqqBoXKE3hUVQ4olFzae2NPdj0/DquQ+aGKKcpbbfIZM2pkgT3raVInf/MHpqDIRka6lW6AG9CzLZgltcwHDTt3ddRsJ6euKh4enve7uCt0DS1ezVksczCTe4ot7VOFHrwteCithnGNMAftGC7vABYIO+JVocbwrKuiqf2W+l2Q="
```

leeebo, 2024-05-17:
Using `espressif.local` can pass now, from the slave's log(esp32c6) the Wi-Fi is connected successfully, but from the master side (esp32p4), it block in `I (448) eppp_link: Waiting for IP address 0`

ESP32-P4 (client) log:

```
I (418) main_task: Calling app_main()
I (418) esp_mqtt_demo: [APP] Startup..
I (418) esp_mqtt_demo: [APP] Free memory: 597128 bytes
I (428) esp_mqtt_demo: [APP] IDF version: v5.3-dev-3225-g5a40bb8746-dirty
I (438) esp_mqtt_demo: ESP_WIFI_MODE_STA
I (448) uart: queue free spaces: 16
I (448) eppp_link: Waiting for IP address 0
I (60448) esp-netif_lwip-ppp: Connection lost
I (60448) eppp_link: Disconnected
I (60448) eppp_link: PPP Connection failed 1 times, try to reconnect.
```

ESP32-C6 (server) log:

```
I (455) main_task: Calling app_main()
I (455) esp_mqtt_demo: [APP] Startup..
I (455) esp_mqtt_demo: [APP] Free memory: 392012 bytes
I (465) esp_mqtt_demo: [APP] IDF version: v5.3-dev-3225-g5a40bb8746-dirty
I (485) esp_mqtt_demo: ESP_WIFI_MODE_STA
I (485) pp: pp rom version: 5b8dcfa
I (485) net80211: net80211 rom version: 5b8dcfa
I (495) wifi:wifi driver task: 40827f60, prio:23, stack:6656, core=0
I (495) wifi:wifi firmware version: a209922
I (495) wifi:wifi certification version: v7.0
I (505) wifi:config NVS flash: enabled
I (505) wifi:config nano formating: disabled
I (505) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (515) wifi:Init data frame dynamic rx buffer num: 32
I (515) wifi:Init static rx mgmt buffer num: 5
I (525) wifi:Init management short buffer num: 32
I (525) wifi:Init dynamic tx buffer num: 32
I (535) wifi:Init static tx FG buffer num: 2
I (535) wifi:Init static rx buffer size: 1700
I (535) wifi:Init static rx buffer num: 10
I (545) wifi:Init dynamic rx buffer num: 32
I (545) wifi_init: rx ba win: 6
I (555) wifi_init: accept mbox: 6
I (555) wifi_init: tcpip mbox: 32
I (555) wifi_init: udp mbox: 6
I (565) wifi_init: tcp mbox: 6
I (565) wifi_init: tcp tx win: 5760
I (575) wifi_init: tcp rx win: 5760
I (575) wifi_init: tcp mss: 1440
I (575) wifi_init: WiFi IRAM OP enabled
I (585) wifi_init: WiFi RX IRAM OP enabled
I (605) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
W (645) wifi:(bf)761:0x600a7cac:0x01b4b4b0
W (645) wifi:(agc)0x600a7128:0xd21d1800, min.avgNF:0xce->0xd2(dB), RCalCount:0x1d1, min.RRssi:0x800(-128.00)
W (645) wifi:(TB)WDEV_PWR_TB_MCS0:19
W (645) wifi:(TB)WDEV_PWR_TB_MCS1:19
W (655) wifi:(TB)WDEV_PWR_TB_MCS2:19
W (655) wifi:(TB)WDEV_PWR_TB_MCS3:19
W (655) wifi:(TB)WDEV_PWR_TB_MCS4:19
W (665) wifi:(TB)WDEV_PWR_TB_MCS5:19
W (665) wifi:(TB)WDEV_PWR_TB_MCS6:18
W (665) wifi:(TB)WDEV_PWR_TB_MCS7:18
W (675) wifi:(TB)WDEV_PWR_TB_MCS8:17
W (675) wifi:(TB)WDEV_PWR_TB_MCS9:15
W (675) wifi:(TB)WDEV_PWR_TB_MCS10:15
W (685) wifi:(TB)WDEV_PWR_TB_MCS11:15
I (685) wifi:11ax coex: WDEVAX_PTI0(0x55777555), WDEVAX_PTI1(0x00003377).

I (695) wifi:mode : sta (54:32:04:08:49:d8)
I (695) wifi:enable tsf
[... 28 lines cut ...]
I (2225) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (2235) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (2425) wifi:dp: 2, bi: 102400, li: 4, scale listen interval from 307200 us to 409600 us
I (2425) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (3215) esp_netif_handlers: sta ip: 192.168.1.167, mask: 255.255.255.0, gw: 192.168.1.1
I (3215) esp_mqtt_demo: EVENT type IP_EVENT id 0
I (3215) esp_mqtt_demo: got ip:192.168.1.167
I (3215) esp_mqtt_demo: connected to ap SSID:HAPPYBOY password:43539100
I (3225) esp_mqtt_demo: Other event id:7
I (3235) main_task: Returned from app_main()
W (3485) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x1004547, TALO:0x201841ec, (ssn:1, win:64, cur_ssn:1), CONF:0xc0000005
I (4485) esp_mqtt_demo: MQTT_EVENT_CONNECTED
I (4485) esp_mqtt_demo: sent publish successful, msg_id=52726
I (4485) esp_mqtt_demo: sent subscribe successful, msg_id=62380
I (4485) esp_mqtt_demo: sent subscribe successful, msg_id=50762
I (4495) esp_mqtt_demo: sent unsubscribe successful, msg_id=38462
I (4985) esp_mqtt_demo: MQTT_EVENT_PUBLISHED, msg_id=52726
I (5305) esp_mqtt_demo: MQTT_EVENT_SUBSCRIBED, msg_id=62380
I (5305) esp_mqtt_demo: sent publish successful, msg_id=0
I (5705) esp_mqtt_demo: MQTT_EVENT_SUBSCRIBED, msg_id=50762
I (5705) esp_mqtt_demo: sent publish successful, msg_id=0
I (5715) esp_mqtt_demo: MQTT_EVENT_DATA
TOPIC=/topic/qos1
DATA=data_3
```

leeebo, 2024-05-17:
I got the same result after changing the name to "espressif.local"
`esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x3D00`

david-cermak, 2024-05-17:
I wonder why the `ESP32-C6 (server) log:` uses the tag `esp_mqtt_demo` and tries to connect to MQTT ? should be just this project: https://github.com/espressif/esp-protocols/tree/master/components/esp_wifi_remote/examples/server

david-cermak, 2024-05-17:
> I got the same result after changing the name to "espressif.local" esp-tls-mbedtls: mbedtls_pk_parse_keyfile returned -0x3D00

must be some copy-paste error. I'll try to repeat the steps from above line by line, to check if we're not missing something.

PS: You can also generate the keys and certs manually.

leeebo, 2024-05-17:
> I wonder why the `ESP32-C6 (server) log:` uses the tag `esp_mqtt_demo` and tries to connect to MQTT ? should be just this project: https://github.com/espressif/esp-protocols/tree/master/components/esp_wifi_remote/examples/server

Please ignore the previous comment (deleted now), I burned the wrong bin on Slave.

david-cermak, 2024-05-17:
Just checked with freshly generated keys:

1) `idf.py create-project-from-example "espressif/esp_wifi_remote:server"`
2) `idf.py set-target esp32c6`
3 `./managed_components/espressif__esp_wifi_remote/examples/test_certs/generate_test_certs.sh espressif.local`
4) Open local `sdkconfig` and Edit:
   - `CONFIG_ESP_WIFI_REMOTE_EPPP_UART_*` Rx, Tx from `sdkconfig.ci.c6`
   - `CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_*`  from step `3)
   - `CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_*`  from step `3)
5) `idf.py build flash monitor`
6) `idf.py create-project-from-example "espressif/esp_wifi_remote:server"`
7) `idf.py set-target esp32p4`
8) Open local `sdkconfig` and Edit:
   - `CONFIG_ESP_WIFI_REMOTE_EPPP_UART_*` Rx, Tx from `sdkconfig.ci.p4`
   - `CONFIG_ESP_WIFI_REMOTE_EPPP_SERVER_*`  from step `3)
   - `CONFIG_ESP_WIFI_REMOTE_EPPP_CLIENT_*`  from step `3)
9) `idf.py menuconfig` -- Setup WiFi ssid and password
10) `idf.py build flash monitor`

leeebo, 2024-05-20:
@david-cermak it is `-----BEGIN PRIVATE KEY-----`

david-cermak, 2024-05-20:
> it is -----BEGIN PRIVATE KEY-----

Oh, understand now why it doesnt' understand the format, thanks.
Will fix!

closed 2024-05-28 by david-cermak