espressif/esp-protocols #576: modem pppos_client idf4.4.7 Not compatible with smart_config process

chen-laodi, 2024-05-23:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

- ESP_IDF: 4.4.7
- esp_modem:1.0.5
- development kit: esp32-wrover-e

### Problem Description
When using idf4.4.7 ppp to access the Internet, there is an incompatibility problem when starting wifi smart_config at the same time. When switching to idf5.2.1, it can run normally. Is it probably caused by freeRTOS?

After transplanting the pppos_client routine to my project, mqtt sending data and voip receiving calls will also cause this exception. The cost of migrating my project based on idf4.4.7 to idf5.x is very high.

In the attached test code I roughly identified the code that caused the problem `uxBits = xEventGroupWaitBits(s_wifi_event_group, CONNECTED_BIT | ESPTOUCH_DONE_BIT, true, false, portMAX_DELAY);`, but I did not have the ability to solve it

### Debug Logs
```
ELF file SHA256: bc4b279a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7004
ho 0 tail 12 room 4
load:0x40078000,len:15784
ho 0 tail 12 room 4
load:0x40080400,len:4536
0x40080400: _init at ??:?

entry 0x400806bc
I (30) boot: ESP-IDF v4.4.7-dirty 2nd stage bootloader
I (31) boot: compile time 17:11:47
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (38) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (48) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 16MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00020000
I (81) boot:  1 otadata          OTA data         01 00 00029000 00002000
I (89) boot:  2 phy_init         RF data          01 01 0002b000 00001000
I (96) boot:  3 factory          factory app      00 00 00030000 00200000
I (104) boot:  4 ota_0            OTA app          00 10 00230000 00200000
I (111) boot:  5 ota_1            OTA app          00 11 00430000 00200000
I (119) boot:  6 flash_tone       Unknown data     01 ff 00630000 00100000
I (126) boot: End of partition table
I (131) boot: Defaulting to factory image
I (135) esp_image: segment 0: paddr=00030020 vaddr=3f400020 size=27430h (160816) map
I (188) esp_image: segment 1: paddr=00057458 vaddr=3ffb0000 size=03558h ( 13656) load
I (193) esp_image: segment 2: paddr=0005a9b8 vaddr=40080000 size=05660h ( 22112) load
I (201) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=c331ch (799516) map
I (423) esp_image: segment 4: paddr=00123344 vaddr=40085660 size=0df90h ( 57232) load
I (451) boot: Loaded app from partition at offset 0x30000
I (451) boot: Disabling RNG early entropy source...
I (462) cpu_start: Multicore app
I (463) psram: This chip is ESP32-D0WD
I (463) spiram: Found 64MBit SPI RAM device
I (465) spiram: SPI RAM mode: flash 80m sram 80m
I (470) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (477) cpu_start: Pro cpu up.
I (481) cpu_start: Starting app cpu, entry point is 0x400814cc
0x400814cc: call_start_cpu1 at D:/esp_runtime/4.4.7/esp-idf-v4.4.7/components/esp_system/port/cpu_start.c:151

I (472) cpu_start: App cpu up.
I (983) spiram: SPI SRAM memory test OK
I (991) cpu_start: Pro cpu start user code
I (991) cpu_start: cpu freq: 240000000
I (991) cpu_start: Application information:
I (994) cpu_start: Project name:     pppos_client
I (1000) cpu_start: App version:      mdns-v1.3.1-2-ged021a9-dirty
I (1006) cpu_start: Compile time:     May 23 2024 17:11:29
I (1013) cpu_start: ELF file SHA256:  bc4b279a...
I (1018) cpu_start: ESP-IDF:          v4.4.7-dirty
I (1024) cpu_start: Min chip rev:     v0.0
I (1028) cpu_start: Max chip rev:     v3.99
I (1033) cpu_start: Chip rev:         v3.1
I (1038) heap_init: Initializing. RAM available for dynamic allocation:
I (1045) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1051) heap_init: At 3FFB46A0 len 0002B960 (174 KiB): DRAM
I (1058) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1064) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1071) heap_init: At 400935F0 len 0000CA10 (50 KiB): IRAM
I (1077) spiram: Adding pool of 4082K of external SPI memory to heap allocator
I (1086) spi_flash: detected chip: gd
I (1089) spi_flash: flash io: qio
I (1094) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1109) spiram: Reserving pool of 18K of internal memory for DMA/internal allocations
I (1146) wifi:wifi driver task: 3ffc20b4, prio:23, stack:6656, core=0
I (1146) system_api: Base MAC address is not set
I (1146) system_api: read default base MAC address from EFUSE
I (1155) wifi:wifi firmware version: 1fd20f4
I (1157) wifi:wifi certification version: v7.0
I (1161) wifi:config NVS flash: enabled
I (1164) wifi:config nano formating: disabled
I (1169) wifi:Init data frame dynamic rx buffer num: 128
I (1174) wifi:Init static rx mgmt buffer num: 5
I (1178) wifi:Init management short buffer num: 32
I (1182) wifi:Init static tx buffer num: 9
I (1187) wifi:Init tx cache buffer num: 32
I (1190) wifi:Init static rx buffer size: 1600
I (1194) wifi:Init static rx buffer num: 9
I (1198) wifi:Init dynamic rx buffer num: 128
I (1203) wifi_init: rx ba win: 16
I (1206) wifi_init: tcpip mbox: 64
I (1210) wifi_init: udp mbox: 64
I (1214) wifi_init: tcp mbox: 64
I (1218) wifi_init: tcp tx win: 65535
I (1222) wifi_init: tcp rx win: 65535
I (1226) wifi_init: tcp mss: 1440
I (1230) wifi_init: WiFi/LWIP prefer SPIRAM
I (1236) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (1307) wifi:mode : sta (30:c9:22:f1:61:4c)
I (1307) wifi:enable tsf
I (1309) pppos_example: Initializing esp_modem for the SIM7600 module...
I (1310) uart: queue free spaces: 30
I (1315) esp-netif_lwip-ppp: User interrupt
I (1319) pppos_example: PPP state changed event 5
I (1324) pppos_example: User interrupted event from netif:0x3ffc9bbc
I (1331) esp_modem_netif: PPP state changed event 5
I (1359) smartconfig: SC version: V3.0.1
Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3ffb5ae4  PS      : 0x00050033  A0      : 0x3ffb5ae4  A1      : 0x3ffb5890
A2      : 0x3ffb5bc0  A3      : 0x3f40c264  A4      : 0x3ffb6ac0  A5      : 0x00060f23
A6      : 0x007b1228  A7      : 0x003fffff  A8      : 0x800878e4  A9      : 0x0000051e
A10     : 0x3ffb3800  A11     : 0x3f40bc88  A12     : 0x00000000  A13     : 0x0000001e  
A14     : 0x00060f23  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000002
EXCVADDR: 0x3ffb5ae4  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x3ffb5ae1:0x3ffb5890 |<-CORRUPTED
```
[pppos_client.zip](https://github.com/espressif/esp-protocols/files/15414774/pppos_client.zip)

david-cermak, 2024-05-28:
Hi @chen-laodi 

I was able to reproduce the issue and it went away after increasing stack size of the main task.
It seems that if exceptions are disabled, the reported fatal error isn't very readable. After enabling exceptions it just reported stack overflow.

chen-laodi, 2024-05-29:
Yes, after turning the main task stack size up it seems to be working fine.
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8192

chen-laodi, 2024-05-29:
Yes, after turning the main task stack size up it seems to be working fine.Thank you@david-cermak
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8192

chen-laodi, 2024-05-29:
@david-cermak Yes, after turning the main task stack size up it seems to be working fine.Thank you
CONFIG_ESP_MAIN_TASK_STACK_SIZE=8192

chen-laodi, 2024-06-20:
Hi @david-cermak 
I encountered another stack overflow issue occurring at esp_modem_new_dev. From examining the source code, I discovered that the exception occurs at this point.

Adjusting the dte_config.task_stack_size to 8192 did not resolve the issue, despite having plenty of PSRAM space.

If you need the complete code, please provide me with your email, and I will send it to you.

```c
std::unique_ptr<Terminal> create_uart_terminal(const esp_modem_dte_config *config)
{
    TRY_CATCH_RET_NULL(
        auto term = std::make_unique<UartTerminal>(config); // this code stack overflow
        term->start();
        return term;
    )
}
```

log
```log
entry 0x400806bc
I (31) boot: ESP-IDF v4.4.7-dirty 2nd stage bootloader
I (31) boot: compile time 15:36:01
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 16MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00020000
I (81) boot:  1 otadata          OTA data         01 00 00029000 00002000
I (89) boot:  2 phy_init         RF data          01 01 0002b000 00001000
I (96) boot:  3 factory          factory app      00 00 00030000 00200000
I (104) boot:  4 ota_0            OTA app          00 10 00230000 00200000
I (111) boot:  5 ota_1            OTA app          00 11 00430000 00200000
I (119) boot:  6 flash_tone       Unknown data     01 ff 00630000 00100000
I (127) boot: End of partition table
I (131) boot: Defaulting to factory image
I (135) esp_image: segment 0: paddr=00030020 vaddr=3f400020 size=3e07ch (254076) map
I (214) esp_image: segment 1: paddr=0006e0a4 vaddr=3ffb0000 size=01f74h (  8052) load
I (217) esp_image: segment 2: paddr=00070020 vaddr=400d0020 size=116c30h (1141808) map
I (534) esp_image: segment 3: paddr=00186c58 vaddr=3ffb1f74 size=01a3ch (  6716) load
I (537) esp_image: segment 4: paddr=0018869c vaddr=40080000 size=175b4h ( 95668) load
I (583) boot: Loaded app from partition at offset 0x30000
I (583) boot: Disabling RNG early entropy source...
I (594) cpu_start: Multicore app
I (594) psram: This chip is ESP32-D0WD
I (595) spiram: Found 64MBit SPI RAM device
I (596) spiram: SPI RAM mode: flash 80m sram 80m
I (601) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (609) cpu_start: Pro cpu up.
I (612) cpu_start: Starting app cpu, entry point is 0x4008168c
0x4008168c: call_start_cpu1 at D:/esp_runtime/4.4.7/esp-idf-v4.4.7/components/esp_system/port/cpu_start.c:151

I (0) cpu_start: App cpu up.
I (1115) spiram: SPI SRAM memory test OK
I (1123) cpu_start: Pro cpu start user code
I (1123) cpu_start: cpu freq: 240000000
I (1123) cpu_start: Application information:
I (1126) cpu_start: Project name:     sound_control
I (1132) cpu_start: App version:      09169b8-dirty
I (1137) cpu_start: Compile time:     Jun 20 2024 10:20:38
I (1143) cpu_start: ELF file SHA256:  dab428d07a726e5b...
I (1149) cpu_start: ESP-IDF:          v4.4.7-dirty
I (1155) cpu_start: Min chip rev:     v0.0
I (1160) cpu_start: Max chip rev:     v3.99
I (1165) cpu_start: Chip rev:         v3.1
[... 66 lines cut ...]
I (5833) AUDIO_ELEMENT: [record-0x3f9e977c] Element task created
I (5839) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:2331947 Bytes, Inter:210595 Bytes, Dram:175995 Bytes

I (5851) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (5876) AUDIO_ELEMENT: [record] AEL_MSG_CMD_RESUME,state:1
I (5877) AUDIO_PIPELINE: Pipeline started
I (6369) wifi:ic_enable_sniffer
I (6370) smartconfig: Start to find channel...
I (17533) uart: queue free spaces: 30

***ERROR*** A stack overflow in task esp_timer has been detected.

Backtrace: 0x40082355:0x3ffb58a0 0x40091545:0x3ffb58c0 0x40091c41:0x3ffb58e0 0x40092d75:0x3ffb5960 0x40091640:0x3ffb5980 0x400915f2:0x3ffb59a0
0x40082355: panic_abort at D:/esp_runtime/4.4.7/esp-idf-v4.4.7/components/esp_system/panic.c:408

0x40091545: esp_system_abort at D:/esp_runtime/4.4.7/esp-idf-v4.4.7/components/esp_system/esp_system.c:137

0x40091c41: vApplicationStackOverflowHook at D:/esp_runtime/4.4.7/esp-idf-v4.4.7/components/freertos/port/xtensa/port.c:407

0x40092d75: vTaskSwitchContext at D:/esp_runtime/4.4.7/esp-idf-v4.4.7/components/freertos/tasks.c:3664

0x40091640: _frxt_dispatch at D:/esp_runtime/4.4.7/esp-idf-v4.4.7/components/freertos/port/xtensa/portasm.S:436

0x400915f2: _frxt_int_exit at D:/esp_runtime/4.4.7/esp-idf-v4.4.7/components/freertos/port/xtensa/portasm.S:231
```

david-cermak, 2024-06-20:
> ***ERROR*** A stack overflow in task esp_timer has been detected.

The issue is happening in **`esp_timer` task**, not the main task, nor the DTE task.

chen-laodi, 2024-06-20:
Hi @david-cermak 

I am indeed using esp_timer, but the exception is not generated by esp_timer itself. I created a timer synchronously during the initialization of PHY. If it detects that the Ethernet cable has not been inserted within 10 seconds, it then shuts down the PHY and starts the MODEM.

Due to the insufficient detail in the output logs, I am unable to fully pinpoint the problem. However, I am quite certain that the stack overflow occurs when executing this line: dce = esp_modem_new_dev(ESP_MODEM_DCE_SIM7600, &dte_config, &dce_config, ppp_netif);

phy init code
```c
void initialize_eth(bool start_timer)
{
    // 注册eth插入监听计时器
    if (start_timer)
    {
        const esp_timer_create_args_t timer_args = {
            .callback = &no_eth_timer_callback,
            .name = "no_eth_timer"};
        ESP_ERROR_CHECK(esp_timer_create(&timer_args, &no_eth_timer));
    }

    // 为以太网创建新的 esp-netif 默认实例
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    eth_netif = esp_netif_new(&cfg);

    // 初始化 MAC 配置
    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    mac_config.smi_mdc_gpio_num = ETH_MDC_GPIO;
    mac_config.smi_mdio_gpio_num = ETH_MDIO_GPIO;
    eth_mac_t = esp_eth_mac_new_esp32(&mac_config);

    // 初始化 PHY 配置
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = ETH_PHY_ADDR;
    phy_config.reset_gpio_num = ETH_PHY_RST_GPIO;
    eth_phy_t = esp_eth_phy_new_ip101(&phy_config);

    // 安装 ETH 驱动
    esp_eth_config_t config = ETH_DEFAULT_CONFIG(eth_mac_t, eth_phy_t);
    esp_err_t ret = esp_eth_driver_install(&config, &eth_handle);
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "Ethernet driver installation failed with error: %s", esp_err_to_name(ret));
        return;
    }

    // 将以太网驱动程序附加到 TCP/IP 堆栈
    s_eth_glue = esp_eth_new_netif_glue(eth_handle);
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif, s_eth_glue));

    // 注册用户定义的事件处理程序
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &event_handler, NULL));

    // 启动以太网驱动程序状态机
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
}
```

no_eth_timer_callback
```c
/**
 * @brief 网线插入定时器回调函数
 *        若10s内没有插入网线，则关闭eth开启调制解调器
 */
static void no_eth_timer_callback(void *arg)
{
    // 取消注册事件处理程序
    ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_ETH_GOT_IP, &event_handler));
    ESP_ERROR_CHECK(esp_event_handler_unregister(ETH_EVENT, ESP_EVENT_ANY_ID, &event_handler));

    // 卸载ETH
    ESP_ERROR_CHECK(esp_eth_stop(eth_handle));
    ESP_ERROR_CHECK(esp_eth_del_netif_glue(s_eth_glue));
    ESP_ERROR_CHECK(esp_eth_driver_uninstall(eth_handle));
    eth_phy_t->del(eth_phy_t);
    eth_mac_t->del(eth_mac_t);
    esp_netif_destroy(eth_netif);
    no_eth_timer = NULL;

    // 失能PHY
    gpio_set_level(ETH_PHY_RST_GPIO, 0);
    vTaskDelay(pdMS_TO_TICKS(2000));

    // 开启modem
    initialize_cell();
}
```

laodi-chen, 2024-06-21:
Hi @david-cermak 
I have adjusted the task stack size in several places, but this problem still occurs.
help me🙏🙏🙏🙏