espressif/esp-protocols #581: Websocket - Allow to choose which memory takes (internal, external) for buffer

filzek, 2024-05-28:
### Is your feature request related to a problem?

Memory allocartion always internal, need to have an option to choose it to delegates internal or external.

filzek, 2024-05-28:
could be setup in the esp_websocket_client_config_t  or with a custom handler like in the CJSON to override:

memoryHook.malloc_fn = spiram_calloc;
memoryHook.free_fn = spiram_free;
cJSON_InitHooks(&memoryHook);

// Wrapper calloc SPI RAM.
void* spiram_calloc(size_t size) {
    void* ptr = TRACEABLE_HEAP_CAPS_CALLOC(1, size * sizeof(unsigned char), MALLOC_CAP_SPIRAM);
    if(ptr == NULL) {
        printf("Fail alloc SPIRAM\n");
    }
    return ptr; // return NULL if TRACEABLE_HEAP_CAPS_CALLOC fail
}

void spiram_free(void* ptr) {
    heap_caps_free(ptr);
}

so in the hooks the way to handle the memory could be lead to user to choose, or just like a flag in the esp_websocket_client_config_t   as:
.memorytouse = 0 (0, internal, 1, external)

filzek, 2024-06-03:
I have sent an pull request proposal to the maintainers.

filzek, 2024-06-03:
https://github.com/espressif/esp-protocols/pull/586

send, so, if you need to use external memory in websocket, now it is full working like a charm.,

bryghtlabs-richard, 2025-10-03:
I also had this problem, but was able to solve it by adjusting `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL`. It's not as fine-grained as per-module allocator options, but worked well enough for us.