espressif/esp-protocols #589: esp32 s3 fails to connect to mqtt ssl broker with A7670C lte modem

VaishnaviSPadiyappanavar, 2024-06-06:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32 S3 is connected to the A7670C lte modem via usb and the pppos_client is being used from esp_modem external component .
ESP32 S3 succesfully connects to the Non-SSL broker but fails to connect to the SSL Mqtt broker 

these are logs when it fails to connect to mqtt broker
I (7263) APP_MQTT: Client started
I (7263) HEAP_CHECK: Free heap size DURING MQTT SSL CONNECTION: 214548 bytes
I (7273) APP_MQTT: Waiting for bits
I (7793) mbedtls: ssl_tls.c:3939 => handshake

I (7803) mbedtls: ssl_msg.c:2124 => flush output

I (7803) mbedtls: ssl_msg.c:2133 <= flush output

I (7803) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_HELLO_REQUEST

I (7813) mbedtls: ssl_msg.c:2124 => flush output

I (7813) mbedtls: ssl_msg.c:2133 <= flush output

I (7823) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_CLIENT_HELLO

I (7833) mbedtls: ssl_client.c:906 => write client hello

W (7833) mbedtls: ssl_client.c:258 got supported group(001d)

W (7843) mbedtls: ssl_client.c:258 got supported group(0017)

W (7853) mbedtls: ssl_client.c:258 got supported group(0018)

W (7853) mbedtls: ssl_client.c:258 got supported group(0019)

W (7863) mbedtls: ssl_client.c:258 got supported group(001a)

W (7863) mbedtls: ssl_client.c:258 got supported group(001b)

W (7873) mbedtls: ssl_client.c:258 got supported group(001c)

I (7883) mbedtls: ssl_msg.c:2554 => write handshake message

I (7883) mbedtls: ssl_msg.c:2714 => write record

I (7893) mbedtls: ssl_msg.c:2851 <= write record

I (7903) mbedtls: ssl_msg.c:2675 <= write handshake message

I (7903) mbedtls: ssl_client.c:994 <= write client hello

I (7913) mbedtls: ssl_msg.c:2124 => flush output

I (7913) mbedtls: ssl_msg.c:2138 message length: 246, out_left: 246

I (7923) mbedtls: ssl_msg.c:2145 ssl->f_send() returned 246 (-0xffffff0a)

I (7933) mbedtls: ssl_msg.c:2172 <= flush output

I (7933) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_SERVER_HELLO

I (7943) mbedtls: ssl_tls12_client.c:1195 => parse server hello

I (7953) mbedtls: ssl_msg.c:3887 => read record

I (7953) mbedtls: ssl_msg.c:1926 => fetch input
[... 313 lines cut ...]
I (10063) mbedtls: ssl_msg.c:2124 => flush output

I (10063) mbedtls: ssl_msg.c:2133 <= flush output

I (10073) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_SERVER_CHANGE_CIPHER_SPEC

I (10083) mbedtls: ssl_tls12_client.c:3379 => parse new session ticket

I (10083) mbedtls: ssl_msg.c:3887 => read record

I (10093) mbedtls: ssl_msg.c:1926 => fetch input

I (10103) mbedtls: ssl_msg.c:2066 in_left: 0, nb_want: 5

I (20103) mbedtls: ssl_msg.c:2086 in_left: 0, nb_want: 5

I (20103) mbedtls: ssl_tls.c:3950 <= handshake

W (20103) esp-tls: Failed to open new connection in specified timeout
E (20103) transport_base: Failed to open a new connection
E (20113) mqtt_client: Error transport connect
I (20113) APP_MQTT: INFO:MQTT EVENT_0
E (20123) APP_MQTT: INFO:MQTT_EVENT_ERROR
I (20123) APP_MQTT: INFO:MQTT EVENT_2
I (20133) APP_MQTT: INFO:Client disconnected

VaishnaviSPadiyappanavar, 2024-06-07:
yes we have increased the timeout till 60 seconds but still the issue is not solved with this.

VaishnaviSPadiyappanavar, 2024-06-11:
Just a quick follow-up on my previous message . I'm still facing the same issue and would greatly appreciate any suggestions you might have. Do you think the issue could be related to USB?.

david-cermak, 2024-06-11:
I wasn't able to reproduce the issue, just checked with SIMCOM A7670 over USB and this public broker `mqtts://mqtt.eclipseprojects.io:8883` and everything works as expected.

Are you able to connect to your broker using WiFi or Ethernet? Could you check with some public service, so I'd be able to replicate the problem?

VaishnaviSPadiyappanavar, 2024-06-12:
Yes, we can connect to our MQTT SSL broker via WiFi and Ethernet. We also connected to the public broker mqtts://mqtt.eclipseprojects.io:8883 without issues. However, our MQTT SSL broker requires certificates for validation, and it isn't working in that case

david-cermak, 2024-06-12:
I was able to reproduce the issue when trying to connect to another public service `mqtts://test.mosquitto.org:8884` which requires client side verification.

The issue seems to be related to the USB host transmit size.
At the moment I can offer the following workaround:

```patch
diff --git a/components/esp_modem/src/esp_modem_netif.cpp b/components/esp_modem/src/esp_modem_netif.cpp
index d808db07..5a550d52 100644
--- a/components/esp_modem/src/esp_modem_netif.cpp
+++ b/components/esp_modem/src/esp_modem_netif.cpp
@@ -32,9 +32,23 @@ esp_err_t Netif::esp_modem_dte_transmit(void *h, void *buffer, size_t len)
 {
     auto *ppp = static_cast<Netif *>(h);
     if (ppp->signal.is_any(PPP_STARTED)) {
-        if (ppp->ppp_dte && ppp->ppp_dte->write((uint8_t *) buffer, len) > 0) {
-            return ESP_OK;
+        ESP_LOGD("Modem Netif", "Tx size(%d)", len);
+        if (!ppp->ppp_dte) {
+            return ESP_FAIL;
         }
+
+        uint8_t *ptr = (uint8_t*)buffer;
+        const size_t max_len = 256;
+        size_t remain = len;
+        while (remain > 0) {
+            int batch = std::min(max_len, remain);
+            if (ppp->ppp_dte->write(ptr, batch) <= 0) {
+                return ESP_FAIL;
+            }
+            remain -= batch;
+            ptr += batch;
+        }
+        return ESP_OK;
     }
     return ESP_FAIL;
 }
@@ -65,6 +79,7 @@ esp_err_t Netif::esp_modem_post_attach(esp_netif_t *esp_netif, void *args)
 
 void Netif::receive(uint8_t *data, size_t len)
 {
+    ESP_LOGD("Modem Netif", "Rx size(%d)", len);
     esp_netif_receive(driver.base.netif, data, len, nullptr);
 }
```

Thanks for reporting! We'll look into it and fix it.

VaishnaviSPadiyappanavar, 2024-06-13:
@david-cermak Thank you for the workaround!. Just to confirm, do you foresee any potential issues or drawbacks with this workaround in the long run?

closed 2024-06-26 by tore-espressif