espressif/esp-protocols #614: ASIO doesn't compile on Mac OS

moreana, 2024-07-13:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp-idf: v5.2.2
asio: v1.28.2

I am trying to compile tcp_echo_server example from ASIO and I get such linking errors:
```
/Users/attila/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/espressif__asio/libespressif__asio.a(asio.cpp.obj):(.literal._ZN4asio6detail10socket_ops9inet_ntopEiPKvPcjmRSt10error_code+0x4): undefined reference to `if_indextoname'
/Users/attila/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/espressif__asio/libespressif__asio.a(asio.cpp.obj):(.literal._ZN4asio6detail18signal_set_service16open_descriptorsEv+0x4): undefined reference to `pipe'
/Users/attila/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/espressif__asio/libespressif__asio.a(asio.cpp.obj): in function `_ZN4asio6detail10socket_ops9inet_ntopEiPKvPcjmRSt10error_code':
/Users/attila/projects/esp/esp-idf/examples/wifi/scan/managed_components/espressif__asio/asio/asio/include/asio/detail/impl/socket_ops.ipp:2519:(.text._ZN4asio6detail10socket_ops9inet_ntopEiPKvPcjmRSt10error_code+0xa2): undefined reference to `if_indextoname'
/Users/attila/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/espressif__asio/libespressif__asio.a(asio.cpp.obj): in function `_ZN4asio6detail18signal_set_service16open_descriptorsEv':
/Users/attila/projects/esp/esp-idf/examples/wifi/scan/managed_components/espressif__asio/asio/asio/include/asio/execution_context.hpp:400:(.text._ZN4asio6detail18signal_set_service16open_descriptorsEv+0xa): undefined reference to `pipe'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/attila/projects/esp/esp-idf/examples/wifi/scan/managed_components/espressif__asio/examples/tcp_echo_server/build/log/idf_py_stderr_output_14993 and /Users/attila/projects/esp/esp-idf/examples/wifi/scan/managed_components/espressif__asio/examples/tcp_echo_server/build/log/idf_py_stdout_output_14993
```
There is no definition for pipe and if_indextoname and I am not sure where these functions should come from.

Could you please help me to solve this issue?

euripedesrocha, 2024-07-19:
Hi @moreana thanks for reporting. 
This is an issue that happens with IDF version 5.2.2, it is already solved in IDF 5.2 release branch. 
I'm trying to find a solution on ASIO build side to fix it for the released tag, in the meantime I would suggest you to use the HEAD of that IDF release branch.

ZantsuRocks, 2024-07-22:
Same issue happens in 5.2.1 when using `(asio::ip::udp::endpoint).address()` in your code

david-cermak, 2024-12-18:
It's possible to workaround this issue by adding a dependency to `sock_utils` library (which actually adds a simple `pipe()` implementation).
There might also be a link issue with `if_indextoname()` -> to fix this, you need switch to the `v5.2.3` or cherry-pick https://github.com/espressif/esp-idf/commit/6567f0b9b8c and set `CONFIG_LWIP_NETIF_API=y`.

**PS**: if you really need to stick with `v5.2.2`, you can put this to your main component
```
#include "lwip/if_api.h"

char *if_indextoname(unsigned int ifindex, char *ifname)
{
    return lwip_if_indextoname(ifindex, ifname);
}
```

**PPS**: This all will be resolved when switching to the upcoming asio version `v1.32`, adding in https://github.com/espressif/esp-protocols/pull/717

closed 2025-02-03 by euripedesrocha