espressif/esp-protocols #624: Request for Additional Method in ESP Modem: Handling Larger AT Command Responses

srikanthpalvai, 2024-07-26:
### Is your feature request related to a problem?

When issuing the commands AT+CMGF=0 and AT+CMGL=4, the response results in more data than usual. To handle this, we need an additional method that takes max_len as an argument.

### Describe the solution you'd like.

This method allows the user to specify the maximum length of the output buffer, ensuring that larger responses can be handled appropriately.

extern "C" esp_err_t esp_modem_at_raw_ext(esp_modem_dce_t *dce_wrap, const char *cmd, char *p_out, const char *pass, const char *fail, int timeout, **size_t max_len**)
{
    if (dce_wrap == nullptr || dce_wrap->dce == nullptr) {
        return ESP_ERR_INVALID_ARG;
    }
    std::string out;
    auto ret = command_response_to_esp_err(dce_wrap->dce->at_raw(cmd, out, pass, fail, timeout));
    if ((p_out != NULL) && (!out.empty())) {
        strlcpy(p_out, out.c_str(), **max_len**);
    }
    return ret;
}

### Describe alternatives you've considered.

#define STR_MAX 1024

esp_modem_dce_t *dce_wrap;
char response[ESP_MODEM_C_API_STR_MAX];
esp_err_t ret = esp_modem_at_raw_ext(dce_wrap, "AT+CMGL=4", response, "OK", "ERROR", 1000, STR_MAX );
if (ret == ESP_OK) {
    // Process the response
}

srikanthpalvai, 2024-07-26:
@david-cermak 
Could you please have a look on it.

david-cermak, 2024-07-26:
Yes, I'll look into it, planning on some maintenance work and the v1.2 release in about two weeks.
Meantime, we're open for ideas/suggestions and PRs are welcome.

srikanthpalvai, 2024-08-01:
Hi @david-cermak

I was not able to create a pull request for esp-modem
but here is the patch file it will help you.
[0001-Add-max_len-parameter-to-esp_modem_at_raw_ext-for-bu.patch](https://github.com/user-attachments/files/16462228/0001-Add-max_len-parameter-to-esp_modem_at_raw_ext-for-bu.patch)

srikanthpalvai, 2024-08-07:
Hi @david-cermak 
I have fixed it with below logic. Added it in the project root CMakeLists.txt

set(ESP_MODEM_C_API_STR_MAX_DEFAULT 512 CACHE STRING "Default value for ESP_MODEM_C_API_STR_MAX")
target_compile_definitions(__idf_espressif__esp_modem PUBLIC ESP_MODEM_C_API_STR_MAX=${ESP_MODEM_C_API_STR_MAX_DEFAULT})

closed 2024-10-30 by david-cermak