espressif/esp-protocols #628: using MQTT with certificate bundle results in tls timeout

txf-, 2024-08-13:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm on IDF version 4.4. and I am using a simcom 7000 modem  on the LilyGO-T-SIM7000G board.

I'm using a task that is heavily based on the pppos_client example. The difference Is that I am using mqtt with ssl via the esp_crt_bundle_attach function. 

Unfortunately I am having various intermittent issue, mostly around tls handshake failures. Very rarely it succeeds and then I can send a few messages briefly (though often only a few before the connection is dropped).

attached is a log with lwip ppp debug messages enabled.[mqtt_ssl_timeout.txt](https://github.com/user-attachments/files/16598137/mqtt_ssl_timeout.txt)

david-cermak, 2024-08-19:
Hi @txf- 

Are you using `esp_modem` library and your project is based on the `pppos_client` example from this repo?
Or just based on the IDF example from here 
or this [pppos_client](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/protocols/pppos_client) example from IDF v4.4 ?

Just quickly checked the `esp_modem` example on `v4.4` and the connection to `"mqtts://mqtt.eclipseprojects.io"` with
```
mqtt_config.crt_bundle_attach = esp_crt_bundle_attach;
```
worked without any issue on my side.

txf-, 2024-08-19:
> Are you using esp_modem library and your project is based on the pppos_client example from this repo?

This repo.

> worked without any issue on my side.

For me the code from this repo does work sometimes, but it is very erratic. Connecting without SSL improves reliability, but not fully. 

So I should use the 4.4 code from the esp-idf repo? Does it support the Sim7000? I notice in the example only the sim800, sim7600 and the bg96 are supported.

david-cermak, 2024-08-19:
> For me it does work sometimes, but it is very erratic. Connecting without SSL improves reliability, but not fully.

If the MQTT connection over TCP is unstable, then there must be something seriously wrong (means that you're not able to reliably send/receive ~6 bytes!). Do you have an antenna attached to your board? 

> It is consistently reporting good RSSI readings, or is that not a good indicator of network quality?

Which RSSI numbers are you getting exactly?

> So I should use the 4.4 code from the esp-idf repo?

Nono, `esp_modem` && this repos is the preferred way.

> Is there a reason for the PPP connection to fail, if it fails to connect to the broker? Shouldn't it maintain the tunnel and retry the TCP connection?

I think what you're seeing here is that the TCP connection reports error faster than PPP layer even realizes we're disconnected. MQTT client retries to connect (automatically), but the PPP does not.

txf-, 2024-08-19:
> Do you have an antenna attached to your board?

Yes. It is the flex antenna that came with the board. 

> Which RSSI numbers are you getting exactly?

between 19 and 23. Sometimes I get 99, but that is because for some reason resetting the device via PWRKEY is finicky. 

Is there a way to reestablish PPP automatically? In a safe way? I'll do some more testing, but I noticed that doing it without resetting the modem did not give me good results.