espressif/esp-protocols #634: BUG MDNS LoadProhibited in SDK 5.5.2 (IDFGH-13451) #14357

filzek, 2024-08-17:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

_mdns_execute_action(a);

Guru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.

Core 0 register dump:
PC : 0x40113552 PS : 0x00060330 A0 : 0x00000000 A1 : 0x3fff9440
0x40113552: _mdns_execute_action at D:/Dropbox/Dev/Producao5/managed_components/espressif__mdns/mdns.c:5220
(inlined by) _mdns_service_task at D:/Dropbox/Dev/Producao5/managed_components/espressif__mdns/mdns.c:5537

A2 : 0x00000000 A3 : 0x3ffeac9c A4 : 0xfefefefe A5 : 0x3fff7c80
A6 : 0x3fff7c7c A7 : 0x3fff9440 A8 : 0x3ffe353c A9 : 0x00000008
A10 : 0x0000000b A11 : 0xffffffff A12 : 0xffffffff A13 : 0x00000000
A14 : 0x00000000 A15 : 0x3ffeaf34 SAR : 0x00000018 EXCCAUSE: 0x0000001c
EXCVADDR: 0xfefeff16 LBEG : 0x400911fc LEND : 0x40091218 LCOUNT : 0xffffffff
0x400911fc: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:162
0x40091218: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197

Backtrace: 0x4011354f:0x3fff9440
0x4011354f: _mdns_execute_action at D:/Dropbox/Dev/Producao5/managed_components/espressif__mdns/mdns.c:5217
(inlined by) _mdns_service_task at D:/Dropbox/Dev/Producao5/managed_components/espressif__mdns/mdns.c:5537

david-cermak, 2024-08-19:
Hi @filzek 

Is it possible that you're adding/removing services in one thread, while updating TXT entries in some other thread?
If so, you might be experiencing issues with some APIs. In this case, this PR would fix the issue: https://github.com/espressif/esp-protocols/pull/615
If you only add a constant number of services on startup, from one thread, then it would just look like a memory corruption issue.

david-cermak, 2024-10-30:
Most likely fixed in https://github.com/espressif/esp-protocols/pull/615 
Please reopen if you see similar issues again.

closed 2024-10-30 by david-cermak