espressif/esp-protocols #635: SIM7000G connect with user name password and chap auto.

Aviarbiser, 2024-08-19:
### Answers checklist.

- [X] I have read the documentation for esp-protocols [components](https://github.com/espressif/esp-protocols#readme) and the issue is not addressed there.
- [X] I have updated my esp-protocols branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I am using sim7000 with pppos, it working ok if I don't use chap authentication .  
I try to find some help or example to do this but I don't find something that work.
Is there any working example or any idea how to do this.
In the TinyGSM project on Arduino I fixed the AT command in gprsConnectImpl to: sendAT(GF("+CNCFG=1,\""), apn, "\",\"", "\",\"", user, pwd,authentication, '" ');
And it works well, maybe it will help solve the problem here too.

david-cermak, 2024-08-19:
Hi @Aviarbiser 

I'm sorry, we currently don't have an example that demonstrates PPP connection using CHAP authentication method.
you have to add a call to `esp_netif_ppp_set_auth()` before  switching to data mode.

```cpp
#include "esp_netif_ppp.h"
 ...

//
{
    ...
    // before PPPoS connection, but after initializing netif and modem
    esp_netif_ppp_set_auth(esp_netif, NETIF_PPP_AUTHTYPE_CHAP, USERNAME, PASSWORD);
```

Aviarbiser, 2024-08-20:
Hi david,

Thanks for your fast replay.
I will test it today and update.

Regards

Avihai Arbiser
+972508942777
***@***.***

Aviarbiser, 2024-08-20:
Hi, thanks for your fast replay.

I add this to my code and it not connecting to GRPS (LTE) .
I try to change to other SIM card that not use the same APN (using the default "internet") but I don't changed the connection settings like APN ,user , password and chap auth. and the modem is connected and I got IP .
as much as I know if you try to connect with wrong APN it will not connect .
I try to set the APN with "esp_modem_set_apn(dce, MODEM_PPP_APN);" and it still connect with the wrong APN.
I am also not remove "esp_netif_ppp_set_auth(esp_netif, NETIF_PPP_AUTHTYPE_CHAP, "Admin", "1234");" and it was connected .
Maybe that's why it doesn't connect with the SIM that uses APN and CHAP.
how can I check it ?

david-cermak, 2024-08-20:
Note that `esp_netif_ppp_set_auth()` configure the username and password for authentication over PPPoS, while the `AT+CGAUTH` sets APN credentials via AT commands (not sure if they are the same or which one you should use)

Aviarbiser, 2024-08-20:
Ok , but if I see that the modem is connected "ATD*99# return CONNECT 150000000 " why I don't get IP and get timeout?

What I do to check (debug) this problem?

Aviarbiser, 2024-08-21:
Hello again, 
I d more tests , and I see that the modem is connected to the gprs , and the pppos program try to send something (I fink this is a query to check the IP or the connection status). 
But in the end I get the msg "NO CARRIER" see attached log :
CONNECT 150000000<CR><LF>
~?}#?!}!} } }9}"}&} } } } }#}%?#}%}%}&?‘??}'}"}(}"&}3~~?}#?!}"}?} }4}"}&} } } } }%}&?¨\8}'}"}(}"} },~~?}#?!}!}!} }4}"}&} } } } }%}&?‘??}'}"}(}"?},~~?}#?!}+}"} }(?‘??¶?~~€!<ETX><CR>
<NUL><FS><ACK><LF>
<VT><FF>
<CR>
ƒ<ACK><LF>
<VT><FF>
<SO>‚<ACK><LF>
<VT><FF>
<CR>
„<ACK><LF>
<VT><FF>
<SO><DC2>?~<CR><LF>
NO CARRIER<CR><LF>

any suggestion?

david-cermak, 2024-08-23:
And here's how it looks like if the authentication succeeds: https://github.com/espressif/esp-protocols/actions/runs/10528192870/job/29173244199?pr=640#step:5:217 (I'll have to still update it to run both methods)
(you can run the test locally: https://github.com/espressif/esp-protocols/tree/master/components/esp_modem/test/target, just need a usb2serial (e.g. esp_prog) and  running `pytest`)

Aviarbiser, 2024-08-24:
Thank I will check and update

Aviarbiser, 2024-08-27:
After further investigation, I discovered that the issue was related to the SIM7000 module. I have since replaced it with a SIM7600 module, and I am pleased to report that everything is now functioning perfectly.

Thank you for your support and assistance.

Best regards,

david-cermak, 2024-10-30:
CHAP authentication tests are merged in https://github.com/espressif/esp-protocols/pull/640

closed 2024-10-30 by david-cermak